turtlebot/turtlebot4 #509: /odom can't echo from PC

pvg6, 2024-12-07:
### Describe the the feature you would like

/odom topic can't echo from PC

Ubuntu: 22 
Distro: humble
Simple Discovery

Navigation and Slam things are not working...
topics in my PC

/battery_state
/cliff_intensity
/cmd_audio
/cmd_lightring
/cmd_vel
/color/preview/camera_info
/color/preview/image
/diagnostics
/diagnostics_agg
/diagnostics_toplevel_state
/dock
/hazard_detection
/hmi/buttons
/hmi/display
/hmi/display/message
/hmi/led
/imu
/interface_buttons
/ip
/ir_intensity
/ir_opcode
/joint_states
/joy
/joy/set_feedback
/kidnap_status
/mobility_monitor/transition_event
/mouse
/odom
/parameter_events
/robot_description
/robot_state/transition_event
/rosout
/scan
/slip_status
/static_transform/transition_event
/stereo/camera_info
/stereo/depth
/stop_status
/tf
/tf_static
/wheel_status
/wheel_ticks
/wheel_vels

Exported the CYCLONEDDS_URI
topics same as the PI and create 3 is connected to same wifi,..
PI , create3 and PC are of same ROS_DOMIAN_ID

### Motivation and impact

I bought two turtlebot4 lite but both of them are facing same issues,..please help me to resolve this issue

smatarCPR, 2024-12-31:
Hello @pvg6 ,
Thank you for reaching out on the Turtlebot 4 Github. Apologies for the late reply on this matter. I have a couple of questions/recommendations regarding this matter:

- Are you able to echo the odom topic when SSH'ed in the Turtlebot's Raspberry Pi?
- Are you able to echo non-Create3 topics from your remote computer, for example /scan topic?
- If you list the topics on the remote computer and when SSH'ed into the Raspberry Pi are the lists the same?
- If you operate with only one Turtlebot on, does the behaviour change?
- Are you able to test out a Discovery server setup temporarily? Does the behaviour change with this new setup?

Looking forward to hearing from you soon.
Best Regards,
Saif

smatarCPR, 2025-01-27:
Hello @pvg6 
This issue is being closed due to inactivity. If you are still experiencing the issue, feel free to reopen this ticket when you are ready to continue the troubleshooting process.
Best Regards,
Saif

closed 2025-01-27 by smatarCPR