turtlebot/turtlebot4 #510: Not able to connect the Controller with TB4

UdayWaveMark, 2024-12-17:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Discovery Server

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

Jazzy

### Type of issue

Navigation (SLAM, Nav2 etc.)

### Expected behaviour

Should be able to connect the Controller without any issues

### Actual behaviour

Short version: Not able to connect the Controller with TB4. (Jazzy, Ubuntu 24.04)

Long version:
Since this form doesn't have an option to select the following I have selected the existing options for ROS Distro and OS.

ROS Distro: Jazzy ISO version 2.0.1
OS: Ubuntu 24.04

First I would like to say that I have experimented with ROS2 Humble and setup my Turtlebot4 Standard and was able to connect the Controller which came with the TB4 Standard via Bluetooth and use RViz, create a map of a location and managed to do autonomous driving of that map.

My issue is with ROS2 Jazzy ISO version 2.0.1. Created Jazzy image, replaced the old SD card (that had Humble) with the Jazzy image, followed the Basic Setup instructions + Discovery Server setup and finished setting up the Turtlebot4 with Jazzy.

Then as it is mentioned in the 'Supplemental Configuration', to connect my controller, I SSH into the Turtlebot4 and follow the instructions.  

When I issue the command 'sudo ssh bluetoothctl' the 'bluetoothd' CLI starts.  Then I go on and issue 'scan on' command.  That lists all the bluetooth devices available in vicinity.  Then I turn ON the Controller by pressing the 'Home' and 'Share' buttons as specified in the document.  I see the controller's MAC address displayed in the list of Bluetooth devices.

Then I call 'trust MAC_ADDRESS'.  This is successful.
Then I call 'pair MAC_ADDRESS'.  This is successful.
Then I call 'connect MAC_ADDRESS'.  This command FAILS saying the device NOT AVAILABLE.  I've tried this over 10 times and in different intervals.  

Here I've added an image of my SSH prompt.

[image]

### To Reproduce

1. ssh sudo bluetoothctl
2. scan on
3. trust MAC_ADDRESS
4. pair MAC_ADDRESS
5. connect MAC_ADDRESS

RustyCPR, 2024-12-20:
Hi @UdayWaveMark 
Sorry to see your having controller pairing concerns.

1. Can you confirm the Saitake STK-4006L Bluetooth controller is charging? 
Have you tried another charger and or micro USB cable? You may want to keep it on the charger for over night. 
The controller - when it charges you should see it pulsing orange - then after some time it will turn off. 
When you power it on - it will flash white - then turns solid blue when it connects with the TB4. 
2. Can you confirm you are seeing these LED status colors when powering on and charging? 

From there assuming its charging. 

Run `sudo bluetoothctl` 
Run the command: `paired-devices `
Then untrust and remove all devices: 
`untrust XX:XX:XX:XX:XX:XX`
`remove XX:XX:XX:XX:XX:XX`

Each of these will confirm that the device has been untrusted or removed - if not you may need to run the command again. 

With the Saitake STK-4006L Bluetooth controller in pairing mode - Press and hold both the home and share buttons on the controller until the light starts blinking.
[image]

Run `scan on`
Find the Bluetooth device listed as Controller and controller address.
Then run the below commands with the listed controller address: 
`trust XX:XX:XX:XX:XX:XX `
`pair XX:XX:XX:XX:XX:XX ​`
`connect XX:XX:XX:XX:XX:XX ​`
You may need to enter the above trust/pair or connect commands two or three times for it to succeed.
https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html#turtlebot-4-controller-manual-setup

Let me know,
Rusty

UdayWaveMark, 2025-01-24:
Hi @RustyCPR ,

I just wanted to know that I resolved this issue and close it.  Basically, I was trying to issue all those commands in the command line and it failed consistently.  Somewhere I came across a script that I could run for the same thing.  I tried the script and it worked.  My controller connected with the TB4 successfully.  So, clearly it was a timing issue when I was executing the commands in the command line.  It doesn't matter how fast I run these commands in the command line, it always failed to connect.  The script did the trick.

Posting the script here (for anyone who's dealing with the same/similar issue):

#!/usr/bin/expect -f

set device [lindex $argv 0];
send_user "Device $device\n"
set timeout 60

spawn bluetoothctl
expect "Agent registered"
send -- "remove $device\r"
expect "Device has been removed"
send -- "scan on\r"
expect "$device"
send -- "pair $device\r"
expect "Pairing successful"
send -- "connect $device\r"
expect "Connection successful"
send -- "trust $device\r"
expect "trust succeeded"
send -- "exit\r"