turtlebot/turtlebot4 #512: Humble movement command doesn't work

bchen-dev, 2024-12-30:
### Robot Model

Turtlebot4 Lite

### ROS distro

Humble

### Networking Configuration

Simple Discovery

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

N/A

### Type of issue

Other

### Expected behaviour

I followed the documentation, but the humble command didn't work for driving it and the jazzy command worked.
```
Humble command (didn't work):
ros2 topic pub /cmd_vel geometry_msgs/msg/Twist \
"twist:
  linear:
    x: 0.1
    y: 0.0
    z: 0.0
  angular:
    x: 0.0
    y: 0.0
    z: 0.0"

Jazzy command (worked):
ros2 topic pub /cmd_vel geometry_msgs/msg/TwistStamped \
"twist:
  linear:
    x: 0.1
    y: 0.0
    z: 0.0
  angular:
    x: 0.0
    y: 0.0
    z: 0.0"
```

### Actual behaviour

Both should work

### To Reproduce

I tried both commands

bchen-dev, 2024-12-30:
The Create3 is running I.0.0.FastDDS version. Should I downgrade to the Humble release?

smatarCPR, 2024-12-31:
Hello @bchen-dev ,
Thank you for reaching out on the Turtlebot 4 Github page. I am very sorry to hear that you are having issues with your Turtlebot unit. I have a few questions/recommendations regarding your issues:

- Is the Turtlebot able to move if you use other commands, such as teleop twist - https://turtlebot.github.io/turtlebot4-user-manual/tutorials/driving.html#keyboard-teleoperation
- What version of the Create3 firmware is currently installed? This can be checked by logging into the Create3 webserver and seeing what is written at the top of the welcome page
[image]
- You mention that the Jazzy version of the command seems to work with your turtlebot 4, what is the output of `echo $ROS_DISTRO` if you run this command on the Turtlebot? What is the output if you run the same command on your remote computer
- Have you tried running the command whilst SSH'ed into the Turtlebot 4's Raspberry Pi are the results the same (Jazzy version works, but Humble version does not)?

Looking forward to hearing from you soon.
Best Regards,
Saif

bchen-dev, 2024-12-31:
> Hello @bchen-dev , Thank you for reaching out on the Turtlebot 4 Github page. I am very sorry to hear that you are having issues with your Turtlebot unit. I have a few questions/recommendations [...]

Hey @smatarCPR , thanks for your reply. 
* The standard `ros2 run teleop_twist_keyboard teleop_twist_keyboard` doesn't work and the `ros2 run teleop_twist_keyboard teleop_twist_keyboard --ros-args -p stamped:=true` works well.
* I am running `VERSION: I.0.0.FastDDS`.
* I am running the humble on my remote computer (ubuntu 22) and the Turtlebot is running the jazzy (ubuntu 24). 
* The Turtlebot only has the jazzy version. Should I install the humble one?

smatarCPR, 2025-01-02:
Hello @bchen-dev ,
Thank you for providing these details. Based on these details I have the following recommendations:

- Cross distro communication (Humble to Jazzy) is not supported and can lead to various issues, particularly the one you are currently encountering. This would explain why the Jazzy version of commands works on the Turtlebot whereas running Humble versions of the command would not work. Other issues that may occur with the setup you have described is data loss when subscribing to topics from the Turtlebot.
- If you are unable to have the remote PC run ROS Jazzy, then the recommended action is to modify the Turtlebot to run ROS Humble. Instructions on how to upgrade to Jazzy can be found on our online documentation here - https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html#install-latest-raspberry-pi-image

Let me know if you have any questions or if I can help with anything else.
Best Regards,
Saif

smatarCPR, 2025-01-13:
Hello @bchen-dev 
I will go ahead and close this issue. Should you encounter any follow up issues, do not hesitate to reach out by simply replying to this thread or opening up a new issue.
Best Regards,
Saif

closed 2025-01-13 by smatarCPR