turtlebot/turtlebot4 #513: moti- turtlebot  can not move , although I publish the command

orangelee89, 2025-01-05:
### Robot Model

Turtlebot4 Lite

### ROS distro

Humble

### Networking Configuration

Simple Discovery

### OS

Other Linux

### Built from source or installed?

Built from Source

### Package version

ROS :Jazzy 
OS :  Ubuntu 24
Icreate 3 setting:
[image]

name space and domain ID are same as my turtle’s setting
[image]

### Type of issue

Navigation (SLAM, Nav2 etc.)

### Expected behaviour

in mult-robot case ,type moving command ,or use: 
**ros2 run teleop_twist_keyboard teleop_twist_keyboard --ros-args -p stamped:=true**
to move robot1.

### Actual behaviour

I can’t let robot move, I tried use command , but is still not moving. other sensors are working properly ,I’ve tried commend below : 
[image]

the topic has info ,but the robot is not moving 

### To Reproduce

ttbot1:
ros2 launch turtlebot4_bringup lite.launch.py

PC:
**ros2 run teleop_twist_keyboard teleop_twist_keyboard --ros-args -p stamped:=true**
or
**ros2 topic pub /ttbot1/cmd_vel geometry_msgs/msg/TwistStamped "twist:
  linear:
    x: 0.1
    y: 0.0
    z: 0.0
  angular:
    x: 0.0
    y: 0.0
    z: 0.0"**
    
    **ros2 topic echo /ttbot1/cmd_vel geometry_msgs/msg/TwistStamped can receive info**

soeroesg, 2025-01-06:
I have a similar issue, the my TB4 standard is not moving after upgrade to Jazzy (and also upgrading the Create3 to I.0.0). I believe there is some inconsistency with stamped/non-stamped cmd_vel commands.

I did the following 4 tests:

Test1: Bluetooth remote controller: 
left/right/forward do not work, the robot is not moving. Wall follow left/right work well (as side note, I hate to have those commands on buttons that often can be pushed accidentally)

Test 2:
RPi: `ros2 topic echo cmd_vel`
Error: `Cannot echo topic 'cmd_vel', as it contains more than one type: [geometry_msgs/msg/Twist, geometry_msgs/msg/TwistStamped]`

Test 3:
PC: `ros2 run teleop_twist_keyboard teleop_twist_keyboard --ros-args -p stamped:=true`
RPi: `ros2 topic echo cmd_vel_stamped`  --> no output
The robot does not move

Test 4:
PC: `ros2 run teleop_twist_keyboard teleop_twist_keyboard --ros-args -p stamped:=false`
RPi: `ros2 topic echo cmd_vel_unstamped`  --> no output
OK, the robot can be controlled with IJKL buttons

orangelee89, 2025-01-06:
-  I tried it in Raspberry Pi also, but still not moving
-  all the setting are same in my case : 
    **turtlebot:**
   
[image]
[image]

   **PC:**
[image]

-  I only modify the camera type in turtlebot_lite as below : 
[turtlebot4_bringup/config/oakd_pro.yaml](https://github.com/turtlebot/turtlebot4_robot/blob/humble/turtlebot4_bringup/config/oakd_pro.yaml)

i_pipeline_type: RGB     ——>i_pipeline_type: RGBD
i_interleaved: true          ——> false
 i_low_bandwidth:true   ——>false 
[image]

and as the link, modify the urdf file and lunch file to change oakd_lite camera to oakd_ pro_w
[https://github.com/turtlebot/turtlebot4/issues/102#issuecomment-1416704740](https://github.com/turtlebot/turtlebot4/issues/102#issuecomment-1416704740)

- all the ros versions are Jazzy , just in the submission there is not jazzy option ,as the image below
[image]

orangelee89, 2025-01-06:
[turtlebot4_log.txt](https://github.com/user-attachments/files/18325302/turtlebot4_log.txt)

icreate3:
[messages.txt](https://github.com/user-attachments/files/18325414/messages.txt)

orangelee89, 2025-01-10:
how can I solve the problem that the robot need to be "activate", and If I wanna use ros2 moving commend that is stamped ,how I do it

> @smatarCPR hi. as he said [#513 (comment)](https://github.com/turtlebot/turtlebot4/issues/513#issuecomment-2573146013)
> 
> I can use command **ros2 run teleop_twist_keyboard teleop_twist_keyboard --ros-args --remap cmd_vel:=ttbot1/cmd_vel** to control robot to move, but I have to hold the forward” I ” key for long time ,then the robot start to move ,it looks like I have to publish lot of moving topic for “activating” turtlebot . I think that’s hard to control by code .
> 
> and also when I add parameter **-p stamped:=true** the robot is not moving

orangelee89, 2025-01-15:
@smatarCPR 
hi, 
- it seems like 30 seconds, however it’s less than 1 min
- Create3-I.0.0.FastDDS
- no any traffic, I though that’s due to the camera , but when I take off the camera , I got same delay.

- 1. Yes, definitely correct ,if isn’t correct, the robot shouldn’t move.
- 2. Yes, the Create3 connected to 2.4 GHz network
- 3. will try later

orangelee89, 2025-01-16:
@smatarCPR 
hi ,for  **To confirm, you are running a simple discovery setup correct? Is the Create3 connected to 2.4 GHz network? If you remove the namespacing, (setting it to /_do_not_use on the create3) do you still encounter the issue?**

I did try it, but still got some delay, but is batter than adding namespace, if I use command
 “**ros2 topic pub /cmd_vel geometry_msgs/msg/Twist "{linear: {x: 0.1, y: 0.0, z: 0.0}, angular: {x: 0.0, y: 0.0, z: 0.0}}" -r 1**”

since 12th topic published ,the robot start to move ,and one more question is the discovery server necessary? 

[image]

orangelee89, 2025-01-17:
HI I’ve tried without namespace ,It’s working properly ,but since I added name space :ttbot1, the turtlebot doesn’t move at all ,even use unstamped command…

**ros2 run teleop_twist_keyboard teleop_twist_keyboard --ros-args -p stamped:=true  --remap cmd_vel:=ttbot1/cmd_vel
ros2 run teleop_twist_keyboard teleop_twist_keyboard --ros-args -p stamped:=false  --remap cmd_vel:=ttbot1/cmd_vel
ros2 run teleop_twist_keyboard teleop_twist_keyboard --remap cmd_vel:=ttbot1/cmd_vel**

all keyboard controls are not working

orangelee89, 2025-01-17:
@smatarCPR HI I’ve solve the issue , thanks for the help

orangelee89, 2025-01-18:
Hi , the when I was doing slam ,I have some problem as below :
ttbot1:
**ros2 launch turtlebot4_bringup lite.launch.py** 
PC:
****ros2 launch turtlebot4_viz view_navigation.launch.py namespace:=/ttbot1**

[rviz2-1] [INFO] [1737168098.357343239] [ttbot1.rviz2]: Message Filter dropping message: frame 'rplidar_link' at time 1737168097.688 for reason 'discarding message because the queue is full'
[rviz2-1] [ERROR] [1737168099.465242890] [rviz2]: Lookup would require extrapolation into the future.  Requested time 1737168099.430232 but the latest data is at time 1737168099.401779, when looking up transform from frame [rplidar_link] to frame [map]
[rviz2-1] [INFO] [1737168099.583000841] [ttbot1.rviz2]: Message Filter dropping message: frame 'rplidar_link' at time 1737168099.285 for reason 'unknown'

**ros2 launch turtlebot4_navigation slam.launch.py namespace:=/ttbot1****

[sync_slam_toolbox_node-1] [WARN] [1737167908.079997837] [ttbot1.slam_toolbox]: Failed to compute odom pose
[sync_slam_toolbox_node-1] [INFO] [1737167908.117332208] [ttbot1.slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1737167908.101 for reason 'discarding message because the queue is full'

soeroesg, 2025-01-20:
@orangelee89 what was your solution at the end?

orangelee89, 2025-01-21:
@soeroesg  hi, So basically I add **/_do_not_use** after my namespace in the icreate3 ‘s namespace , and set ntp server address as my RPI’s. then I can use the stamped velocity command .

smatarCPR, 2025-01-27:
Hello @orangelee89 ,
Glad to hear that you were able to resolve the issue. I will go ahead and close this issue. Should you encounter any follow up issues, do not hesitate to reach out by simply replying to this thread or opening up a new issue.
Best Regards,
Saif

closed 2025-01-27 by smatarCPR

soeroesg, 2025-01-29:
@smatarCPR 
my problem also got solved by changing the namespace of the create3 from empty to /_do_not_use 

My symptoms were that controlling by the joystick or by Nav2 commands did not move the robot. I identified that the problem is due to incompatible types on the /cmd_vel topic, so I looked deeper into the code.
I figured out by digging the source code of the create3_republisher that it is expecting the create3 namespace to be /_do_not_use.

I caused this problem myself because I did not perform a clean jazzy install but upgraded the robot from the previous version where I had no namespaces at all and I never ran the namespace change command in the turtlebot4-setup (since I do not want to use namespace now either)

Since then, I also found the documentation about this here: https://turtlebot.github.io/turtlebot4-user-manual/software/create3.html?highlight=republisher

Can you please add this link and a warning to the main TB4 Jazzy software page so that we can spare this headache for other users? Thanks for your help.