turtlebot/turtlebot4 #514: Unable to ssh using ethernet cable static IP 192.168.185.3 after updating wifi settings

rarenicks, 2025-01-05:
### Robot Model

Turtlebot4 Lite

### ROS distro

Humble

### Networking Configuration

Simple Discovery

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

TB4

### Type of issue

Networking

### Expected behaviour

**Steps to Reproduce:**

Connect TurtleBot4 to an Ethernet cable.
Access the TurtleBot4 through 192.168.185.3.
Run turtlebot4-setup and navigate to WiFi settings.
Enter the following:
WiFi SSID: <Your_SSID>
WiFi Password: <Your_Password>
Static IP: 172.20.10.10
Apply the settings.
Issue:
After applying the WiFi settings, the system becomes inaccessible.

Unable to log in through the previous IP 192.168.185.3, with ethernet connected. 

### Actual behaviour

No access to the TurtleBot4 via either of the IPs.

### Error messages

```bash
After applying setting (changing wifi - name, password and IP in turtlebot-setup), got the following error.

Permissions for /etc/netplan/40-ethernets.yaml are too open. Netplan configuration should NOT be accessible by others.

** (generate:2671): WARNING **: 21:14:39.860: Permissions for /etc/netplan/50-wifis.yaml are too open. Netplan configuration should NOT be accessible by others.
/etc/netplan/50-wifis.yaml:12:13: Error in network definition: address '172.20.10.10' is missing /prefixlength
            - 172.20.10.10
            ^
Connection to 192.168.185.3 closed by remote host.
Connection to 192.168.185.3 closed.

After this I am unable to login using static IP 192.168.185.3 as well. How can I reset this setting to be atleast login through 192.168.185.3 so that I can set my wifi again.
```

### To Reproduce

- connect to ethernet cable by steps mentioned at : "https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html#recovering-the-raspberry-pi"
- login through 192.168.185.3 to TB4
- Do turtlebot4-setup, wifi settings 
- Enter some wifi ssid, password and some static IP (172.20.10.10),
- Apply settings 
- Now, not able to login through 192.168.185.3 as well.   

### Other notes

However was able to connect to create3 to wifi from create3 webserver, following is the log file. Now create3 has a IP and able to access its webserver but not raspberry, i think because of the network file issue, Please check as its very urgent. 

[messages.txt](https://github.com/user-attachments/files/18311815/messages.txt)

rarenicks, 2025-01-05:
@RustyCPR , @smatarCPR 
Please check and help here.

pvg6, 2025-01-06:
@RustyCPR @smatarCPR 

Same Issue i am facing too... any progress...
please help me with this....

how to connect to raspberrypi

closed 2025-01-06 by rarenicks