turtlebot/turtlebot4 #518: Unable to perform Mapping

manu042k, 2025-01-15:
### Robot Model

Turtlebot4 Lite

### ROS distro

Humble

### Networking Configuration

Simple Discovery

### OS

Ubuntu 22.04

### Built from source or installed?

Built from Source

### Package version

ROS :Jazzy
OS : Ubuntu 24

### Type of issue

Navigation (SLAM, Nav2 etc.)

### Expected behaviour

perform Mapping 

### Actual behaviour

no map topic and tf data is getting published 
[image]

### To Reproduce

1) ros2 launch turtlebot4_bringup lite.launch.py 
2) ros2 launch turtlebot4_navigation slam.launch.py 
3) ros2 launch turtlebot4_viz view_navigation.launch.py 

I have attacked logs also.

[lite.txt](https://github.com/user-attachments/files/18430183/lite.txt)
[messages.txt](https://github.com/user-attachments/files/18430184/messages.txt)
[rviz.txt](https://github.com/user-attachments/files/18430185/rviz.txt)
[slam.txt](https://github.com/user-attachments/files/18430186/slam.txt)

manu042k, 2025-01-17:
@RustyCPR and @smatarCPR 
can you look into this ?

hilary-luo, 2025-03-07:
@manu042k Are you still experiencing this issue? I noticed that you have a namespace listed in the screenshot but you did not launch slam with the namespace, so that is likely what was happening.

RustyCPR, 2025-04-14:
Hi @manu042k We have not seen any updates on this ticket in some time. 
Are you still experiencing a issue? 
Thanks,
Rusty

closed 2025-04-20 by manu042k