turtlebot/turtlebot4 #519: Turtlebot4 COMM light and create_path errors

rosss22, 2025-01-16:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Simple Discovery

### OS

Ubuntu 20.04

### Built from source or installed?

Installed

### Package version

turtlebot4_python_tutorials

### Type of issue

Navigation (SLAM, Nav2 etc.)

### Expected behaviour

1) The COMM light stays on and
2) the turtlebot4 python tutorials run where the create_path python file works: you open Rviz, use the 2D estimate tool to move the turtlebot and the turtlebot will navigate there

### Actual behaviour

1) the COMM light keeps going out, and then when we try to turn off the turtlebot, it takes forever to turn off
2) the create_path python file, when we open localization and nav2 stack the following errors may show up: 
o	Aborting lifecycle bringup
o	Failed to bring up all requested nodes
o	Failed to transform initial pose in time

### Error messages

```bash
I have written them in actual behaviour, but will add them as well
```

### To Reproduce

ros2 launch turtlebot4_navigation localization.launch.py map:=office.yaml
ros2 launch turtlebot4_navigation nav2.launch.py
ros2 run turtlebot4_python_tutorials create_path
ros2 launch turtlebot4_viz view_robot.launch.py

rosss22, 2025-01-17:
We are using Ubuntu 22.04

RustyCPR, 2025-01-20:
Hi there @rosss22 
Thanks for writing in. 
Can you please ensure you use a 5GHz WiFi network with [Discovery Server.](https://turtlebot.github.io/turtlebot4-user-manual/setup/discovery_server.html#discovery-server) 

If issues continue please provide all terminals outputs with commands you are running. 
Thank you,
Rusty

rosss22, 2025-01-21:
Hi Rusty,
We do not plan on changing our network setup with Discovery Server as we only have a few months left to get this project done. I will send output tomorrow.
Thank you,
Sonja

rosss22, 2025-01-21:
cmd line: ros2 launch turtlebot4_navigation localization.launch.py map:=office.yaml
output:
failed to send response to /amcl/change_state (timeout): client will not receive response, at /home/arco-1/ros2_humble/src/ros2/rmw_fastrtps/rmw_fastrtps_shared_cpp/src/rmw_response.cpp:154, at /home/arco-1/ros2_humble/src/ros2/rcl/rcl/src/rcl/service.c:314

we are able to run the create_path file once, and then once we run it again this happens

rosss22, 2025-01-21:
also we are connected to Discovery Server, that was my fault in checking

RustyCPR, 2025-01-27:
HI there @rosss22 
Thanks for the details. 
If using Discovery Server please make sure you have all the needed software updates, and make sure the Create3  is not connected to the internet or WiFI.
Run it all again. 
If you are still seeing issues please provide full terminal text outputs of all terminal windows open.
Best regards,
Rusty

rosss22, 2025-01-27:
source ~/.bashrc
source /opt/ros/humble/setup.bash
arco-1@arco1-ThinkPad:~$ ros2 launch turtlebot4_navigation nav2.launch.py
[INFO] [launch]: All log files can be found below /home/arco-1/.ros/log/2025-01-27-16-20-36-113544-arco1-ThinkPad-22341
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [controller_server-1]: process started with pid [22342]
[INFO] [smoother_server-2]: process started with pid [22344]
[INFO] [planner_server-3]: process started with pid [22346]
[INFO] [behavior_server-4]: process started with pid [22348]
[INFO] [bt_navigator-5]: process started with pid [22350]
[INFO] [waypoint_follower-6]: process started with pid [22352]
[INFO] [velocity_smoother-7]: process started with pid [22354]
[INFO] [lifecycle_manager-8]: process started with pid [22356]
[lifecycle_manager-8] [INFO] [1738012836.390675765] [lifecycle_manager_navigation]: Creating
[waypoint_follower-6] [INFO] [1738012836.391865456] [waypoint_follower]: 
[waypoint_follower-6] 	waypoint_follower lifecycle node launched. 
[waypoint_follower-6] 	Waiting on external lifecycle transitions to activate
[waypoint_follower-6] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[velocity_smoother-7] [INFO] [1738012836.391870299] [velocity_smoother]: 
[velocity_smoother-7] 	velocity_smoother lifecycle node launched. 
[velocity_smoother-7] 	Waiting on external lifecycle transitions to activate
[velocity_smoother-7] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[controller_server-1] [INFO] [1738012836.392049396] [controller_server]: 
[controller_server-1] 	controller_server lifecycle node launched. 
[controller_server-1] 	Waiting on external lifecycle transitions to activate
[controller_server-1] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[lifecycle_manager-8] [INFO] [1738012836.393135068] [lifecycle_manager_navigation]: Creating and initializing lifecycle service clients
[waypoint_follower-6] [INFO] [1738012836.395930203] [waypoint_follower]: Creating
[smoother_server-2] [INFO] [1738012836.397869834] [smoother_server]: 
[smoother_server-2] 	smoother_server lifecycle node launched. 
[smoother_server-2] 	Waiting on external lifecycle transitions to activate
[smoother_server-2] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[planner_server-3] [INFO] [1738012836.399188898] [planner_server]: 
[planner_server-3] 	planner_server lifecycle node launched. 
[planner_server-3] 	Waiting on external lifecycle transitions to activate
[planner_server-3] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[bt_navigator-5] [INFO] [1738012836.405558174] [bt_navigator]: 
[bt_navigator-5] 	bt_navigator lifecycle node launched. 
[bt_navigator-5] 	Waiting on external lifecycle transitions to activate
[bt_navigator-5] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[bt_navigator-5] [INFO] [1738012836.405791524] [bt_navigator]: Creating
[smoother_server-2] [INFO] [1738012836.409986624] [smoother_server]: Creating smoother server
[planner_server-3] [INFO] [1738012836.411149151] [planner_server]: Creating
[planner_server-3] [INFO] [1738012836.415333530] [global_costmap.global_costmap]: 
[planner_server-3] 	global_costmap lifecycle node launched. 
[planner_server-3] 	Waiting on external lifecycle transitions to activate
[planner_server-3] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[behavior_server-4] [INFO] [1738012836.416756158] [behavior_server]: 
[behavior_server-4] 	behavior_server lifecycle node launched. 
[behavior_server-4] 	Waiting on external lifecycle transitions to activate
[behavior_server-4] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[planner_server-3] [INFO] [1738012836.418032308] [global_costmap.global_costmap]: Creating Costmap
[controller_server-1] [INFO] [1738012836.426392084] [controller_server]: Creating controller server
[controller_server-1] [INFO] [1738012836.430507095] [local_costmap.local_costmap]: 
[controller_server-1] 	local_costmap lifecycle node launched. 
[controller_server-1] 	Waiting on external lifecycle transitions to activate
[controller_server-1] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[controller_server-1] [INFO] [1738012836.433343876] [local_costmap.local_costmap]: Creating Costmap
[lifecycle_manager-8] [INFO] [1738012838.394420797] [lifecycle_manager_navigation]: Waiting for service controller_server/get_state...
[lifecycle_manager-8] [INFO] [1738012840.359652893] [lifecycle_manager_navigation]: Starting managed nodes bringup...
[lifecycle_manager-8] [INFO] [1738012840.359810104] [lifecycle_manager_navigation]: Configuring controller_server
[controller_server-1] [INFO] [1738012840.361553716] [controller_server]: Configuring controller interface
[controller_server-1] [INFO] [1738012840.362340584] [controller_server]: getting goal checker plugins..
[controller_server-1] [INFO] [1738012840.362912344] [controller_server]: Controller frequency set to 20.0000Hz
[controller_server-1] [INFO] [1738012840.363043517] [local_costmap.local_costmap]: Configuring
[controller_server-1] [INFO] [1738012840.386216325] [local_costmap.local_costmap]: Using plugin "static_layer"
[... cut ...]
[lifecycle_manager-8] [INFO] [1738012845.610608066] [lifecycle_manager_navigation]: Configuring bt_navigator
[bt_navigator-5] [INFO] [1738012845.612561755] [bt_navigator]: Configuring
[lifecycle_manager-8] [ERROR] [1738012853.039876817] [lifecycle_manager_navigation]: Failed to change state for node: bt_navigator. Exception: bt_navigator/get_state service client: async_send_request failed.
[lifecycle_manager-8] [ERROR] [1738012853.040017985] [lifecycle_manager_navigation]: Failed to bring up all requested nodes. Aborting bringup.
this is exactly what happens with navigation

rosss22, 2025-01-27:
ros2 launch turtlebot4_navigation localization.launch.py map:=CRL_map_01_14_25.yaml
[INFO] [launch]: All log files can be found below /home/arco-1/.ros/log/2025-01-27-16-20-14-657943-arco1-ThinkPad-22297
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [map_server-1]: process started with pid [22298]
[INFO] [amcl-2]: process started with pid [22300]
[INFO] [lifecycle_manager-3]: process started with pid [22302]
[lifecycle_manager-3] [INFO] [1738012814.864472563] [lifecycle_manager_localization]: Creating
[amcl-2] [INFO] [1738012814.865267004] [amcl]: 
[amcl-2] 	amcl lifecycle node launched. 
[amcl-2] 	Waiting on external lifecycle transitions to activate
[amcl-2] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[map_server-1] [INFO] [1738012814.865282045] [map_server]: 
[map_server-1] 	map_server lifecycle node launched. 
[map_server-1] 	Waiting on external lifecycle transitions to activate
[map_server-1] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[map_server-1] [INFO] [1738012814.865426138] [map_server]: Creating
[lifecycle_manager-3] [INFO] [1738012814.866888975] [lifecycle_manager_localization]: Creating and initializing lifecycle service clients
[amcl-2] [INFO] [1738012814.868533188] [amcl]: Creating
[lifecycle_manager-3] [INFO] [1738012816.867922271] [lifecycle_manager_localization]: Waiting for service map_server/get_state...
[lifecycle_manager-3] [INFO] [1738012817.365980456] [lifecycle_manager_localization]: Starting managed nodes bringup...
[lifecycle_manager-3] [INFO] [1738012817.366146131] [lifecycle_manager_localization]: Configuring map_server
[map_server-1] [INFO] [1738012817.368502308] [map_server]: Configuring
[map_server-1] [INFO] [1738012817.368773931] [map_io]: Loading yaml file: CRL_map_01_14_25.yaml
[map_server-1] [INFO] [1738012817.369385064] [map_io]: resolution: 0.05
[map_server-1] [INFO] [1738012817.369411518] [map_io]: origin[0]: -12.3
[map_server-1] [INFO] [1738012817.369426709] [map_io]: origin[1]: -6.3
[map_server-1] [INFO] [1738012817.369440919] [map_io]: origin[2]: 0
[map_server-1] [INFO] [1738012817.369455098] [map_io]: free_thresh: 0.25
[map_server-1] [INFO] [1738012817.369468903] [map_io]: occupied_thresh: 0.65
[map_server-1] [INFO] [1738012817.369482931] [map_io]: mode: trinary
[map_server-1] [INFO] [1738012817.369498447] [map_io]: negate: 0
[map_server-1] [INFO] [1738012817.369932013] [map_io]: Loading image_file: ./CRL_map_01_14_25.pgm
[map_server-1] [INFO] [1738012817.381234776] [map_io]: Read map ./CRL_map_01_14_25.pgm: 259 X 180 map @ 0.05 m/cell
[lifecycle_manager-3] [INFO] [1738012817.401861436] [lifecycle_manager_localization]: Configuring amcl
[amcl-2] [INFO] [1738012818.404946722] [amcl]: Configuring
[amcl-2] [INFO] [1738012818.405753503] [amcl]: initTransforms
[amcl-2] [INFO] [1738012818.424854148] [amcl]: initPubSub
[amcl-2] [INFO] [1738012818.434292969] [amcl]: Subscribed to map topic.
[lifecycle_manager-3] [INFO] [1738012818.454821372] [lifecycle_manager_localization]: Activating map_server
[map_server-1] [INFO] [1738012818.455392507] [map_server]: Activating
[map_server-1] [INFO] [1738012818.455638106] [map_server]: Creating bond (map_server) to lifecycle manager.
[amcl-2] [INFO] [1738012818.456067278] [amcl]: Received a 259 X 180 map @ 0.050 m/pix
[lifecycle_manager-3] [INFO] [1738012818.563212013] [lifecycle_manager_localization]: Server map_server connected with bond.
[lifecycle_manager-3] [INFO] [1738012818.563340552] [lifecycle_manager_localization]: Activating amcl
[amcl-2] [INFO] [1738012818.564416217] [amcl]: Activating
[amcl-2] [INFO] [1738012818.564674711] [amcl]: Creating bond (amcl) to lifecycle manager.
[amcl-2] [INFO] [1738012818.645817315] [amcl]: createLaserObject
[lifecycle_manager-3] [INFO] [1738012818.672897303] [lifecycle_manager_localization]: Server amcl connected with bond.
[lifecycle_manager-3] [INFO] [1738012818.673037299] [lifecycle_manager_localization]: Managed nodes are active
[lifecycle_manager-3] [INFO] [1738012818.673090798] [lifecycle_manager_localization]: Creating bond timer...
[amcl-2] [INFO] [1738012819.524701233] [amcl]: Message Filter dropping message: frame 'rplidar_link' at time 1738012818.387 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[amcl-2] [WARN] [1738012820.475026988] [amcl]: AMCL cannot publish a pose or update the transform. Please set the initial pose...
[amcl-2] [WARN] [1738012822.550480756] [amcl]: AMCL cannot publish a pose or update the transform. Please set the initial pose...

RustyCPR, 2025-02-03:
Hi @rosss22 
Thanks for the details. 
Please confirm if the Create3 is not connected to the Internet? 
Can you please provide the[ Create 3 logs](https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html#accessing-the-create-3-webserver)? 
Thanks,
Rusty

rosss22, 2025-02-05:
Feb  5 16:58:08 iRobot-D588C845B3CF499484DF780829B92FF8 user.notice ntpd: ntpd: timed out waiting for 192.168.186.1, reach 0x00, next query in 32s
Feb  5 16:58:38 iRobot-D588C845B3CF499484DF780829B92FF8 user.notice ntpd: ntpd: reply from 192.168.186.3: delay 0.003044 is too high, ignoring
Feb  5 16:58:41 iRobot-D588C845B3CF499484DF780829B92FF8 user.notice create-platform: [INFO] [1738774721.721897340] [system_health]: CPU usage: max 100 [%] mean 96 [%] RAM usage: 31/59 [MB]
Feb  5 16:58:57 iRobot-D588C845B3CF499484DF780829B92FF8 user.notice ntpd: ntpd: timed out waiting for 192.168.186.1, reach 0x00, next query in 33s
Feb  5 16:59:41 iRobot-D588C845B3CF499484DF780829B92FF8 user.notice create-platform: [INFO] [1738774781.722665608] [system_health]: CPU usage: max 93 [%] mean 91 [%] RAM usage: 31/59 [MB]
Feb  5 16:59:46 iRobot-D588C845B3CF499484DF780829B92FF8 user.notice ntpd: ntpd: timed out waiting for 192.168.186.1, reach 0x00, next query in 65s
Feb  5 17:00:01 iRobot-D588C845B3CF499484DF780829B92FF8 cron.info crond[1093]: USER root pid 2822 cmd /bin/date -u +%4Y%2m%2d%2H%2M%2S > /etc/timestamp
Feb  5 17:00:15 iRobot-D588C845B3CF499484DF780829B92FF8 user.notice ntpd: ntpd: reply from 192.168.186.3: delay 0.003716 is too high, ignoring
Feb  5 17:00:41 iRobot-D588C845B3CF499484DF780829B92FF8 user.notice create-platform: [INFO] [1738774841.723656887] [system_health]: CPU usage: max 94 [%] mean 91 [%] RAM usage: 31/59 [MB]
Feb  5 17:01:07 iRobot-D588C845B3CF499484DF780829B92FF8 user.notice ntpd: ntpd: timed out waiting for 192.168.186.1, reach 0x00, next query in 64s
Feb  5 17:01:41 iRobot-D588C845B3CF499484DF780829B92FF8 user.notice create-platform: [INFO] [1738774901.724969388] [system_health]: CPU usage: max 92 [%] mean 89 [%] RAM usage: 31/59 [MB]
Feb  5 17:02:26 iRobot-D588C845B3CF499484DF780829B92FF8 user.notice ntpd: ntpd: reply from 192.168.186.3: delay 0.002563 is too high, ignoring
Feb  5 17:02:27 iRobot-D588C845B3CF499484DF780829B92FF8 user.notice ntpd: ntpd: timed out waiting for 192.168.186.1, reach 0x00, next query in 66s
Feb  5 17:02:41 iRobot-D588C845B3CF499484DF780829B92FF8 user.notice create-platform: [INFO] [1738774961.721328354] [system_health]: CPU usage: max 100 [%] mean 98 [%] RAM usage: 33/59 [MB]
Feb  5 17:03:41 iRobot-D588C845B3CF499484DF780829B92FF8 user.notice create-platform: [INFO] [1738775021.723715989] [system_health]: CPU usage: max 100 [%] mean 100 [%] RAM usage: 33/59 [MB]
Feb  5 17:03:50 iRobot-D588C845B3CF499484DF780829B92FF8 user.notice ntpd: ntpd: timed out waiting for 192.168.186.1, reach 0x00, next query in 67s

these are the create3 logs since running this... we are connected to internet

rosss22, 2025-02-05:
Feb  5 17:00:08 iRobot-D588C845B3CF499484DF780829B92FF8 syslog.info syslogd started: BusyBox v1.30.1
Feb  5 17:00:08 iRobot-D588C845B3CF499484DF780829B92FF8 kern.notice kernel: klogd started: BusyBox v1.30.1 (2024-02-29 18:59:06 UTC)
Feb  5 17:00:08 iRobot-D588C845B3CF499484DF780829B92FF8 kern.notice kernel: [    0.000000] Linux version 4.14.115-yocto-standard-custom (oe-user@oe-host) (gcc version 8.3.0 (GCC)) #1 Thu Feb 29 19:09:17 UTC 2024
Feb  5 17:00:08 iRobot-D588C845B3CF499484DF780829B92FF8 kern.info kernel: [    0.000000] Board has DDR2
Feb  5 17:00:08 iRobot-D588C845B3CF499484DF780829B92FF8 kern.info kernel: [    0.000000] Analog PMU set to hw control
Feb  5 17:00:08 iRobot-D588C845B3CF499484DF780829B92FF8 kern.info kernel: [    0.000000] Digital PMU set to hw control
Feb  5 17:00:08 iRobot-D588C845B3CF499484DF780829B92FF8 kern.info kernel: [    0.000000] SoC Type: MediaTek MT7688 ver:1 eco:2
Feb  5 17:00:08 iRobot-D588C845B3CF499484DF780829B92FF8 kern.info kernel: [    0.000000] bootconsole [early0] enabled
Feb  5 17:00:08 iRobot-D588C845B3CF499484DF780829B92FF8 kern.info kernel: [    0.000000] CPU0 revision is: 00019655 (MIPS 24KEc)
Feb  5 17:00:08 iRobot-D588C845B3CF499484DF780829B92FF8 kern.info kernel: [    0.000000] MIPS: machine is iRobot Sundial
Feb  5 17:00:08 iRobot-D588C845B3CF499484DF780829B92FF8 kern.info kernel: [    0.000000] Determined physical RAM map:
Feb  5 17:00:08 iRobot-D588C845B3CF499484DF780829B92FF8 kern.info kernel: [    0.000000]  memory: 04000000 @ 00000000 (usable)
Feb  5 17:00:08 iRobot-D588C845B3CF499484DF780829B92FF8 kern.info kernel: [    0.000000] Initial ramdisk at: 0x82da0000 (2057901 bytes)
Feb  5 17:00:08 iRobot-D588C845B3CF499484DF780829B92FF8 kern.warn kernel: [    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
Feb  5 17:00:08 iRobot-D588C845B3CF499484DF780829B92FF8 kern.warn kernel: [    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Feb  5 17:00:08 iRobot-D588C845B3CF499484DF780829B92FF8 kern.info kernel: [    0.000000] Zone ranges:
Feb  5 17:00:08 iRobot-D588C845B3CF499484DF780829B92FF8 kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x0000000003ffffff]
Feb  5 17:00:08 iRobot-D588C845B3CF499484DF780829B92FF8 kern.info kernel: [    0.000000] Movable zone start for each node
Feb  5 17:00:08 iRobot-D588C845B3CF499484DF780829B92FF8 kern.info kernel: [    0.000000] Early memory node ranges
Feb  5 17:00:08 iRobot-D588C845B3CF499484DF780829B92FF8 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x0000000003ffffff]
Feb  5 17:00:08 iRobot-D588C845B3CF499484DF780829B92FF8 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000003ffffff]
Feb  5 17:00:08 iRobot-D588C845B3CF499484DF780829B92FF8 kern.debug kernel: [    0.000000] On node 0 totalpages: 16384
Feb  5 17:00:08 iRobot-D588C845B3CF499484DF780829B92FF8 kern.debug kernel: [    0.000000] free_area_init_node: node 0, pgdat 80560820, node_mem_map 81000060
Feb  5 17:00:08 iRobot-D588C845B3CF499484DF780829B92FF8 kern.debug kernel: [    0.000000]   Normal zone: 128 pages used for memmap
Feb  5 17:00:08 iRobot-D588C845B3CF499484DF780829B92FF8 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Feb  5 17:00:08 iRobot-D588C845B3CF499484DF780829B92FF8 kern.debug kernel: [    0.000000]   Normal zone: 16384 pages, LIFO batch:3
Feb  5 17:00:08 iRobot-D588C845B3CF499484DF780829B92FF8 kern.notice kernel: [    0.000000] random: get_random_bytes called from start_kernel+0x8c/0x47c with crng_init=0
Feb  5 17:00:08 iRobot-D588C845B3CF499484DF780829B92FF8 kern.debug kernel: [    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
Feb  5 17:00:08 iRobot-D588C845B3CF499484DF780829B92FF8 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 
Feb  5 17:00:08 iRobot-D588C845B3CF499484DF780829B92FF8 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 16256
Feb  5 17:00:08 iRobot-D588C845B3CF499484DF780829B92FF8 kern.notice kernel: [    0.000000] Kernel command line: bl_version=4.0.3 console=ttyS1,115200 earlyprintk=serial,ttyS1,115200,keep loglevel=3 ubi.mtd=5 ubi.block=0,rootfs root=/dev/ram rw
Feb  5 17:00:08 iRobot-D588C845B3CF499484DF780829B92FF8 kern.info kernel: [    0.000000] PID hash table entries: 256 (order: -2, 1024 bytes)
Feb  5 17:00:08 iRobot-D588C845B3CF499484DF780829B92FF8 kern.info kernel: [    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
Feb  5 17:00:08 iRobot-D588C845B3CF499484DF780829B92FF8 kern.info kernel: [    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
Feb  5 17:00:08 iRobot-D588C845B3CF499484DF780829B92FF8 kern.info kernel: [    0.000000] Writing ErrCtl register=0007ed55
Feb  5 17:00:08 iRobot-D588C845B3CF499484DF780829B92FF8 kern.info kernel: [    0.000000] Readback ErrCtl register=0007ed55
Feb  5 17:00:08 iRobot-D588C845B3CF499484DF780829B92FF8 kern.info kernel: [    0.000000] Memory: 56900K/65536K available (4201K kernel code, 194K rwdata, 1112K rodata, 180K init, 218K bss, 8636K reserved, 0K cma-reserved)
Feb  5 17:00:08 iRobot-D588C845B3CF499484DF780829B92FF8 kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
Feb  5 17:00:08 iRobot-D588C845B3CF499484DF780829B92FF8 kern.info kernel: [    0.000000] NR_IRQS: 256
[... cut ...]
Feb  5 17:51:46 iRobot-D588C845B3CF499484DF780829B92FF8 user.notice create-platform: [INFO] [1738777906.039062014] [system_health]: CPU usage: max 100 [%] mean 99 [%] RAM usage: 34/59 [MB]
Feb  5 17:52:46 iRobot-D588C845B3CF499484DF780829B92FF8 user.notice create-platform: [INFO] [1738777966.042997386] [system_health]: CPU usage: max 100 [%] mean 100 [%] RAM usage: 34/59 [MB]
Feb  5 17:52:52 iRobot-D588C845B3CF499484DF780829B92FF8 user.notice ntpd: ntpd: timed out waiting for 192.168.186.1, reach 0x00, next query in 134s

when I run it again this is the output from create3 logs

RustyCPR, 2025-02-13:
Hi @rosss22 
Thanks for the details. 
Having the Create3 connected to WiFi causes issues because the Create tries to communicate over WiFI and USB - you are maxing out your CPU usage. 
We recommended to update to the latest packages and use TB4 setup tool to apply a Namespace. 
Let me know if you have any questions,
Rusty

rosss22, 2025-02-17:
Hi, can you please explain how to do this? how do we use the TB4 setup tool to apply a namespace