turtlebot/turtlebot4 #521: Provide non-WiFi Access to Create3 Configuration WebServer

slowrunner, 2025-01-18:
### Describe the the feature you would like

As I understand, for best performance my TurtleBot4 Create3 should have the "WiFi/BlueTooth" switch in the BlueTooth position to prevent the Create3 from "seeing" (allocating Create3 resources and processing) unrelated ROS2 RMW traffic.   

## Issue: 
Turning off Create3 WiFi Prevents Browser Access to the Create3 Configuration Web Server.

## Issue Detail:  
Without a desktop on the Raspberry Pi to launch a browser to point to the Create3 Configuration Web Server at the USB0 IP address, there does not appear to be a way to view the web server content from a third computer set up as the "TurtleBot4 Desktop".  (Perhaps this is a safety feature to prevent unintentional alteration of the TurtleBot4's Create3 configuration parameters..)

## Feature Requested:  Provide non-WiFi access to Create3 Configuration Web Server  

## Possible Implementation:
Provide an HTTP republisher (or other mechanism) to make the Create3 Web Server request pages available, and pass along the needed Create3 responses - in particular:
- \<TurtleBot4_IP\>/about  (and the About:System Info responses)
- \<TurtleBot4_IP\>/logs

Probably should not "republish" the requests for other pages such as /ros-config, etc

## Note: I tried remote shell browsing using elinks - not successfull
```
sudo apt install -y elinks
elinks http://192.168.186.3:8080/about
```
But it does not display the "About:System Info":
[image]

nor logs  ``` elinks http://192.168.186.3:8080/logs```:

[image]

### Motivation and impact

Many people customize their TurtleBot4 robots, so guidance would seem to help many more users than solely me.

smatarCPR, 2025-01-28:
Hello @slowrunner ,
Thank you for reaching out on the Turtlebot 4 Github. In terms of this feature request, we do in fact allow users to connect to the Create3 Webserver using the Raspberry Pi's IP address - https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html#accessing-the-create-3-webserver

Is this in line with the feature you requested or is there something more you wish to be added?

Looking forward to hearing from you soon.
Best Regards,
Saif

slowrunner, 2025-01-28:
@smatarCPR  - yes exactly the feature needed, but it is not working for me:

[image]

Do I need to have the "Wifi/Bluetooth" switch in the Bluetooth position? 
- (Currently in Wifi position so I can access the webserver over the 2.4GHz WiFi)

closed 2025-01-31 by slowrunner