turtlebot/turtlebot4 #524: How to configure turtlebot4_node power_saver param to be false at turtlebot4 service start?

slowrunner, 2025-01-22:
### Robot Model

Turtlebot4 Lite

### ROS distro

Humble

### Networking Configuration

Discovery Server

### OS

Other Linux

### Built from source or installed?

Installed

### Package version

Package: ros-jazzy-turtlebot4-robot
Status: install ok installed
Priority: optional
Section: misc
Installed-Size: 41
Maintainer: rkreinin <rkreinin@clearpathrobotics.com>
Architecture: arm64
Version: 2.0.1-2noble.20241228.082418

### Type of issue

Other

### Expected behaviour

How do I set turtlebot4_node power_saver parameter to false when turtlebot4 service starts?  
(since turtlebot4_node.turtlebot4.cpp does not have a dynamic parameter callback 
  and turtlebot4.yaml does not have the parameter  
  and that file should not be directly edited as I understand.)

Expected behavior when power_saver is false is **not to stop oakd** and rplidar when dockstatus.is_docked changes from false to true.

(I need to test camera without stop_camera service being called to see if oakd container / camera will stay alive - currently dies sometimes when stop_camera service is called.)

### Actual behaviour

turtlebot4 service starts the turtlebot4_node and the power_saver parameter defaults to true.  (no entry in turtlebot4.yaml - assume that is where it would be set - additionally that file comes with the package and should not be edited, correct?)

### Error messages

```bash
There are no error messages.
```

### To Reproduce

```
ubuntu@TB5WaLI:~/TB5-WaLI/wali_ws$ turtlebot4-service-restart
ubuntu@TB5WaLI:~/TB5-WaLI/wali_ws$ ros2 param get turtlebot4_node power_saver
Boolean value is: True
```

smatarCPR, 2025-01-28:
Hello @slowrunner ,
Thank you for reaching out on the Turtlebot 4 Github page. If you wish to make modifications to the configuration of the base turtlebot yaml file you will need to:

1. Install the Turtlebot packages from source
2. Make modifications to the turtlebot4.yaml file in the workspace
3. Modify the ROS settings on the Turtlebot to use the source setup.bash for the source installed workspace.

Following this method you should be able to update the power_save param to false and it should stay that way.

Let me know if you need help with this process or if I can help with anything else.
Best Regards,
Saif

slowrunner, 2025-01-28:
Thank you @smatarCPR 

> 1. Install the Turtlebot packages from source

(I need to know which exactly are "the Turtlebot packages" needed.)

I have elaborated on this question to @hilary-luo [response here](https://github.com/turtlebot/turtlebot4/discussions/517#discussioncomment-11987842), so I will close this issue.

I really appreciate your reply.  Thank you.