turtlebot/turtlebot4 #526: Jazzy Turtlebot4 power_save feature triggers Oak-D-Lite depthai_ros_driver issue - fix promised next Jazzy sync 1 month

slowrunner, 2025-01-23:
### Robot Model

Turtlebot4 Lite

### ROS distro

Humble

### Networking Configuration

Discovery Server

### OS

Other Linux

### Built from source or installed?

Installed

### Package version

ros-jazzy-depthai-ros-driver                     2.10.3-1noble.20241228.074725           arm64

ubuntu@TB5WaLI:~/TB5-WaLI/wali_ws$ dpkg -l | grep turtlebot4*
ii  ros-jazzy-turtlebot4-base                        2.0.1-2noble.20241228.002030            arm64        Turtlebot4 Base Node
ii  ros-jazzy-turtlebot4-bringup                     2.0.1-2noble.20241228.080750            arm64        Turtlebot4 Robot Bringup
ii  ros-jazzy-turtlebot4-description                 2.0.1-1noble.20241228.043050            arm64        Turtlebot4 Description package
ii  ros-jazzy-turtlebot4-diagnostics                 2.0.1-2noble.20241228.015204            arm64        Turtlebot4 Diagnostics
ii  ros-jazzy-turtlebot4-msgs                        2.0.1-1noble.20241227.231937            arm64        Turtlebot4 Messages
ii  ros-jazzy-turtlebot4-navigation                  2.0.1-1noble.20241229.075007            arm64        Turtlebot4 Navigation
ii  ros-jazzy-turtlebot4-node                        2.0.1-1noble.20241228.001437            arm64        Turtlebot4 Node
ii  ros-jazzy-turtlebot4-robot                       2.0.1-2noble.20241228.082418            arm64        Turtlebot4 Robot Metapackage
ii  ros-jazzy-turtlebot4-setup                       2.0.2-1noble.20241227.234705            arm64        Turtlebot4 setup scripts
ii  ros-jazzy-turtlebot4-tests                       2.0.1-2noble.20241227.232435            arm64        Turtlebot4 System Tests

### Type of issue

Camera

### Expected behaviour

With default power_saver: true,  undocking will cause the turtlebot4_node to call the oakd container start_camera, which will resume /oakd/rgb/preview/image_raw publishing

### Actual behaviour

Due to an issue only in the Jazzy depthai_ros_driver package, the stop_camera / start_camera sequence will cause the driver to spin at 100% CPU and never issue a response and never resumes publishing.

### Error messages

```bash
No error message.
- no response to /start_camera service call
- CPU for oakd container jumps from 3% to 100%
- /oakd/rgb/preview/image_raw does not resume being published when turtlebot4 undocks
```

### To Reproduce

See https://github.com/luxonis/depthai-ros/issues/657#issuecomment-2610140994

### Other notes

workarounds: 
- kill the oakd container and launch the camera manually while turtlebot4 is off the dock, kill camera when done using before next docking
- set power_save: false in /opt/ros/jazzy/share/turtlebot_bringup/config/turtlebot4.yaml and ```turtlebot4-service-restart```
   (this will extend charging by about an extra hour but the camera will stay alive)
- build depthai overlay from branch mentioned in Luxonis issue, and configure turtlebot to use overlay

hilary-luo, 2025-01-27:
@slowrunner Thanks for sharing the details here!

slowrunner, 2025-02-24:
Checked on this - new merge 5 days ago - not in released package yet.

slowrunner, 2025-03-10:
Latest Jazzy depthai updates were released yesterday - issue remains - Turtlebot4 still cannot use power_save feature:

https://github.com/luxonis/depthai-ros/issues/657#issuecomment-2710855759

slowrunner, 2025-03-10:
Luxonis says this issue is non-deterministic and has created an issue with rclcpp:

https://github.com/ros2/rclcpp/issues/2763