turtlebot/turtlebot4 #527: VirtualBox extremely slow. Can't do anything meaningful

UdayWaveMark, 2025-01-24:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Discovery Server

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

Jazzy

### Type of issue

Navigation (SLAM, Nav2 etc.)

### Expected behaviour

When VirtualBox is installed and used in a powerful MacBook Pro with ample memory, processing power etc., should be able to run ROS tools (nav2, Rviz, etc.,) fast.  But not happening.

### Actual behaviour

Since this form doesn't have an option to select the following I have selected the existing options for ROS Distro and OS.

ROS Distro: Jazzy
OS: Ubuntu 24.04

I'm using VirtualBox 7.0.0 on my MacBook Pro.  My MacBook Pro has,

macOS Solana 14.7, 2.4 GHz 8-Core Intel Core i9 processor, 32 GB 2667 MHz DDR4 memory, Over 100 GB free space in SSD HD

This configuration works fantastic for my iOS App development using XCode.

For my TB4 development I've installed VirtualBox 7.0.0.  I'm using Ubuntu 24.04 with ROS2 Jazzy.  Using TB4 with Rviz I've generated a map of a location.  When I try the 'Nav to Pose' from Rviz by marking the Initial Pose and Nav to Pose locations 

(OR) 

Running the node 'ros2 run turtlebot4_python_tutorials nav_to_pose'

It rarely works.  Meaning sometimes it does and most of the times it fails.  In the terminal I get so much logs.  Also when Rviz is running or the python tutorial node is launched, my MacBook is making lotta noise and the MacBook Activity Monitor shows the '% CPU' used by the VirtualBox is over 350.  Also the MacBook battery depletes very fast even while it is connected with the charger.  Clicking on any screen in the VirtualBox either takes so much time to load or it never loads at all.

Just to be clear, I'm doing everything "as is" (from Generating the map to Navigation etc) per the instruction / commands documented in the Turtlebot 4 User Manual in the turtlebot4 GitHub.

Here I've attached few screenshots of my VirtualBox Settings:

[image]

[image]

[image]

[image]

### Error messages

```bash

```

### To Reproduce

There is nothing specific.

UdayWaveMark, 2025-01-24:
Also, I thought the following info is useful:

I load the saved map as it is mentioned in the Navigation section of the TB4 User Manual, then launch 'nav2'.  Then I launch Rviz.  Then in Rviz when I set the '2D Pose Estimate' and 'Nav 2 Goal'.

In the terminal where I launched 'nav2' I get the following warning over and over...

[controller_server]: Control loop missed its desired rate of 20.0000 Hz. Current loop rate is inf Hz.

[image]

slowrunner, 2025-01-24:
I run VMWare Fusion on my MacMini2018 3.2GHz 6-Core i7 which is less powerful than your config - BUT here is what I found will get me 31FPS on Rviz2:
- 4 cores
- 8GB RAM
- 8GB Graphics Memory - (Did you only give it 128MB  MegaBytes?)

UdayWaveMark, 2025-01-27:
@slowrunner , 

VirtualBox doesn't allow more than 256MB.  I've tried increasing it from 128MB to 256MB and for some reason it was even more slow (or at least it didn't improve the performance).

Thanks for the suggestion.  I'll install and try 'VMWare Fusion' and use the same configuration as you and see how things go.

Thank you.

slowrunner, 2025-01-27:
I reread the OP, you are doing everything on one machine?  Is this with a physical bot or simulator?

If sim you need to give more processors to VB,  if physical bot you need to move one function at a time off your mac and onto the Raspberry Pi.

UdayWaveMark, 2025-02-07:
@slowrunner, 

Thank you very much for your feedback related to this issue.  After seeing your first comment I decided to try VMWare Fusion with the same configuration as yours.  It is much faster and very responsive.  

Thank you very much.

closed 2025-02-07 by UdayWaveMark

slowrunner, 2025-02-07:
Our next challenge would appear to be configuring a non-RPi time server.  Our choices are in the VM or directly on the Mac.  I'm thinking I would like "on the Mac" so that the Create3 and the Raspberry Pi have a time server regardless of the visualization desktop (the virtual machine) up or not.

I'm starting the research - seems to be the clue to getting good SLAM and navigation per 
https://github.com/turtlebot/turtlebot4/issues/530

Running chrony on the Mac supposedly needs us to turn off the Mac setting to set time automatically.

When I figure it all out, I'll post on my blog: https://github.com/turtlebot/turtlebot4/discussions/517#top

slowrunner, 2025-02-09:
> I'm starting the research - seems to be the clue to getting good SLAM and navigation per
> [#530](https://github.com/turtlebot/turtlebot4/issues/530)

Very interesting - I set up my Raspberry Pi (5) from scratch and installed TB4 software using the GitHub setup command line.  
This may not have set up chronyd, I cannot be sure.  When I was playing around to query the time sync I installed some things and later found that turtlebot4-setup was no longer installed?  A mystery.

I installed turtlebot4-setup using ```sudo apt install ros-jazzy-turtlebot4-setup``` and voila! not only did I again have turtlebot4-setup working, but it had installed chrony and started the chronyd daemon.  I didn't see any "out of sync sensor" messages and my mapping worked much better than before.  

I submitted an issue to ask [why they are recommending to people to setup an external time server ](https://github.com/turtlebot/turtlebot4/issues/542)when installing the turtlebot4-setup package configures a time server on the Raspberry Pi.