turtlebot/turtlebot4 #528: Turtlebot4 No LCD and No LED's - Logs show I2C Error

Simba185, 2025-01-28:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Simple Discovery

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

n/a

### Type of issue

Select One

### Expected behaviour

Turtlebot4 should boot and show ip and information on LCD. 

### Actual behaviour

I have recently tried to configure multiple new Turtlebot4 units. 

I have some devices that are not able to boot up correctly and will show I2C errors throughout the logs. These devices will have no LCD and No LED's and it seems the Turtlebot.service is not available. Like the pi is prevented from booting fully without the i2c.

### Error messages

```bash
I2c errors throughout the logs.
```

### To Reproduce

power on the robot. No LCD

RustyCPR, 2025-01-28:
Hi there Tristan, 
Sorry to see you have this issue. 

1. Can you please detail what steps you followed or what you did to configure the TB4's? 
2. You stated "To have a failure after just opening the devices" Did you remove the PCBA? 
3. Can you please check the LCD ribbon cable is securely connected/inserted under the PCBA? 

[image]

4. Can you remove the rear bottom door on the TB4 and verify the Power Distriution board is fully inserted and both USB and power connected is connected. 

[image]

Thanks,
Rusty

RustyCPR, 2025-01-28:
HI Tristan - did reseating the LCD ribbon cable resolve the issue? 
Regards,
Rusty

RustyCPR, 2025-01-28:
Thanks Tristan. 
Follow up questions: 

3.  _lcd ribbon cable is well seated and reseating it and Pi cable did not resolve the issue._
> you checked the large gray ribbon cable from PCBA and RPi? See photo
I would suggest remove and reseat from both sides. 

 _It seems like turtlebot4.service is completely missing (but I don't know why i2c errors would cause that.)_
> What do you mean by this?  the service is not installed in the software or that it isn't running? 

What is the outputs of: 
`systemctl status turtlebot4.service` 
`i2cdetect -y 3`

[image]

Simba185, 2025-01-28:
Hi there Rusty, 

Yes the gray cable was removed and reseated.

ubuntu@ubuntu:~$ systemctl status turtlebot4.service
Unit turtlebot4.service could not be found.

ubuntu@ubuntu:~$ i2cdetect -y 3
Command 'i2cdetect' not found, but can be installed with:
sudo apt install i2c-tools

ubuntu@ubuntu:~$ printenv | grep ros
AMENT_PREFIX_PATH=/opt/ros/humble
PYTHONPATH=/opt/ros/humble/lib/python3.10/site-packages:/opt/ros/humble/local/lib/python3.10/dist-packages
WORKSPACE_SETUP=/opt/ros/humble/setup.bash
LD_LIBRARY_PATH=/opt/ros/humble/opt/rviz_ogre_vendor/lib:/opt/ros/humble/lib/aarch64-linux-gnu:/opt/ros/humble/lib
PATH=/opt/ros/humble/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin

Seems like there is an issues with the software, but then why the I2c Errors?

RustyCPR, 2025-01-28:
Hi @Simba185 
Thanks for the details. 
Reviewing with my team, they suggest reimaging the Pi:  https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html#install-latest-raspberry-pi-image
When downloading the PRi image make sure its the Standard TB4 image and you are selecting the correct ROS version (Humble or Jazzy).

Sorry for this encounter. We don't see a record of RMC purchasing these from us. 
Where did you buy these TB4s from? 
Can you share the serial numbers on the bottom of the units? 

We do have thorough product testing in our product build facility - this issue is not common. 
Regards,
Rusty

RustyCPR, 2025-01-29:
Thanks for confirming the issue was resolved @Simba185

closed 2025-01-29 by RustyCPR