turtlebot/turtlebot4 #529: Controller connection lost after flashing card

parvsytl, 2025-01-28:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Simple Discovery

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

Controller Connection Issue

After flashing the SD card using the recommended image and following the setup instructions, I have been unable to establish a connection with the controller.

• I carefully followed the flashing process and verified the hardware connections.

• Despite repeating the setup, the controller fails to connect consistently.

### Type of issue

Other

### Expected behaviour

The controller should successfully connect with the TurtleBot4.

### Actual behaviour

The controller gets paired, trusted and connected but immediately got disconnected.

[image]

### Error messages

```bash

```

### To Reproduce

1. Flash the SD card using the recommended image.
2. Follow the setup instructions for connecting the controller.
3. Attempt to pair the controller with the robot.

Simba185, 2025-01-31:
So they are not kidding when they say in the instructions that the Trust, Pair and connect commands need to extremely fast. I think less than 10 seconds.

Just keep trying to Trust, Pair and connect. if it fails just untrust and remove and try again.  the controller light will change to solid blue when working. Its a pain in the butt

parvsytl, 2025-02-02:
> So they are not kidding when they say in the instructions that the Trust, Pair and connect commands need to extremely fast. I think less than 10 seconds.
>
> Just keep trying to Trust, Pair and [...]

Hello,
Thank you for the reply. As mentioned in the error message, the controller connects successfully but immediately gets disconnected. This issue occurs only after flashing the SD card. Previously, everything worked perfectly-I was able to  pair it manually without any issues.

Simba185, 2025-02-04:
I have seen the connect/disconnect thing, I'm not sure of the reason but I solved it by just removing and re adding until it worked correctly.

smatarCPR, 2025-02-04:
Hello @parvsytl ,
Thank you for reaching out on the Turtlebot 4 Github page. I am very sorry to hear that you are having issues with one of our products. Would you be able to test the following pairing script? To run this script, extract the file and run the following set of commands:
```
sudo apt install expect
chmod 755 p4pair
./p4pair <MAC of P4 Wireless Controller>
```
Ensure that the controller is in pairing mode when the script is run. Let me know how this recommendation works for you. Looking forward to hearing from you soon.
Best Regards,
Saif

[p4pair.zip](https://github.com/turtlebot/turtlebot4/files/14654150/p4pair.zip)

parvsytl, 2025-02-04:
> Hello [@parvsytl](https://github.com/parvsytl) , Thank you for reaching out on the Turtlebot 4 Github page. I am very sorry to hear that you are having issues with one of our products. Would you be [...]

Hello Saif,

Thank you for your prompt response and for providing the p4pair script. I followed the steps you shared, and the controller is now successfully paired and working as expected. I appreciate your assistance in resolving this issue.

However, I am still encountering inconsistencies with the navigation and mapping functionality. The robot sometimes fails to map or navigate correctly, and I have been unable to determine the exact cause. Could you please assist me in diagnosing and resolving this issue?

Let me know if you need any logs, outputs, or further details to help troubleshoot.

Thank you again for your support, and I look forward to hearing from you.

Best regards,
Parvathy Santhosh

smatarCPR, 2025-02-05:
Hello @parvsytl 
I have sent a reply with some recommendations in the other thread you have opened regarding the navigation #530 I will go ahead and close this issue. Should you encounter any follow up issues, do not hesitate to reach out by simply replying to this thread or opening up a new issue.
Best Regards,
Saif