turtlebot/turtlebot4 #531: Create 3 unable to connect to Pi

Simba185, 2025-01-30:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Simple Discovery

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

default

### Type of issue

Select One

### Expected behaviour

robot should turn on and all light go green and drive from controller

### Actual behaviour

only 3 of the 5 leds activate and even after the chime is heard from the create there is no communication to the motors.

### Error messages

```bash

```

### To Reproduce

turn robot on

### Other notes

applications settings on create 3 match ros setup.

Simba185, 2025-01-31:
robot required 3x reboots to finally bot correctly.

smatarCPR, 2025-02-04:
Hello @Simba185 ,
Thank you for reaching out on the Turtlebot 4 Github page. I am glad to hear you were able to resolve the matter. A quick suggestion I have should this issue occur again: the Pi has a static IP address on its ethernet interface. Details on how to access the Pi over ethernet can be found on our online user manual - https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html#recovering-the-raspberry-pi

Let me know if you have any follow up questions.
Best Regards,
Saif

Simba185, 2025-02-05:
Yes but this doesn't solve the issue.

I was connected to the pi via wired connection and could see that the create was not publishing topics. 

there is no reason to have to reboot the device 3x times to solve the issues.

When are we getting a fix that makes the device reliable on every boot?

 Im not sure I would trust a "Clearpath Robot" (or Rockwell) for any critical or safety related operation.

closed 2025-02-14 by smatarCPR