turtlebot/turtlebot4 #533: Create Base will not turn off

Simba185, 2025-01-31:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Simple Discovery

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

default

### Type of issue

Select One

### Expected behaviour

robot turns on without connecting to a base and will not turn off unless battery is removed.

### Actual behaviour

robot will not boot correctly and will not turn off.

### Error messages

```bash

```

### To Reproduce

install battery, robot turns on.

slowrunner, 2025-02-01:
Please add complete steps to reproduce “Create3 base will not turn off”

Simba185, 2025-02-04:
> Please add complete steps to reproduce “Create3 base will not turn off”
> 
> What happens when you remove the bot from the dock, and press-hold the center Create3 button for 7-8 seconds (not longer, not shorter)?
 
Nothing happens. there is no sound and the device doesn't not respond to the power button.

The user then must remove the battery to turn off the device.   When the battery is reinstalled the device starts automatically.

smatarCPR, 2025-02-04:
Hello @Simba185 ,
Thank you for reaching out on the Turtlebot 4 Github page. I have a few questions regarding the current state of the Create3:

- What version of firmware are you currently running on the Create3?
- If you have the Turtlebot off the dock and use the screen and buttons to select the power off option does this work?
- If you factory reset the Create3 and reflash the Raspberry Pi SD card does this issue remain or is it cleared?

Looking forward to hearing from you soon.
Best Regards,
Saif

Simba185, 2025-02-05:
Using the centre button or the menu doesn't not work to turn off the device.

The battery is removed as there is no other way to power off. (bad design)

When the battery is installed the device powers on automatically, is this expected?

smatarCPR, 2025-02-14:
Hello @Simba185 ,
The behaviour you are expecting is not to be expected. The Create3 should receive the signal to enter storage mode when either the center button is held for roughly 7s or the option is selected using the buttons and screen on the Turtlebot 4 standard.

We have encountered a similar issue where the Create3 was unresponsive to power on/off commands, and it was discovered that there was a misconfigured namespace preventing the ROS power on/off signal from reaching the Create3.

I have the following suggestion regarding this current issue:

- Are you able to access the Create3's webserver when the it is turned on? If so double check the namespacing settings https://iroboteducation.github.io/create3_docs/webserver/application/ Would it also be possible to share what the namespace is set to?
- Have you attempted to factory reset the Create3? https://iroboteducation.github.io/create3_docs/webserver/about/#factory-reset
- Should the above fail, please provide the Create3 logs and we will review them and see what we can find. - https://iroboteducation.github.io/create3_docs/webserver/logs/

Looking forward to hearing from you soon.
Best Regards,
Saif