turtlebot/turtlebot4 #536: Create3 reports 66 nodes, suggests using Domain ID to limit

slowrunner, 2025-02-02:
### Robot Model

Turtlebot4 Lite

### ROS distro

Humble

### Networking Configuration

Discovery Server

### OS

Other Linux

### Built from source or installed?

Built from Source

### Package version

Create3 running I.0.0.FastDDS

### Type of issue

Other

### Expected behaviour

Do not expect suggestions to decrease load in Create3 log 

### Actual behaviour

Create3 Log has suggestion:
```
Feb  2 01:35:22 WaLI-CREATE3 user.notice create-platform: [INFO] [1738460122.265885175] [_do_not_use.system_health]: CPU usage: max 73 [%] mean 68 [%] RAM usage: 39/59 [MB]
Feb  2 01:35:22 WaLI-CREATE3 user.notice create-platform: [INFO] [1738460122.271452458] [_do_not_use.system_health]: Discovered 66 nodes! Considering using DOMAIN_ID to reduce the burden on the application

```

### Error messages

```bash
None
```

### To Reproduce

turtlebot-service-restart
? Perform undock, dock ?

### ROS Environments
- Create3 (added republish /stop_status to standard turtlebot4 republisher configuration)
- Raspberry Pi (5): turtlebot4.service plus three user nodes (wali_node, say_node, odometer)
- Turtlebot4_Desktop sometimes running rviz2 subscribed to /oakd/rgb/preview, /scan, /robot_description,/tf_static,/tf,  

### Other notes

Create3 Config:

[image]

Create3 Log:
[2025-02-01_Create3_Log.txt](https://github.com/user-attachments/files/18630862/2025-02-01_Create3_Log.txt)

Node List:
```
ubuntu@TB5WaLI:~/TB5-WaLI/wali_ws$ ros2 node list
/_do_not_use/_internal/composite_hazard
/_do_not_use/_internal/kinematics_engine
/_do_not_use/_internal/mobility
/_do_not_use/_internal/stasis
/_do_not_use/mobility_monitor
/_do_not_use/motion_control
/_do_not_use/robot_state
/_do_not_use/static_transform
/_do_not_use/system_health
/_do_not_use/ui_mgr
/analyzers
/create3_repub
/joint_state_publisher
/joy_linux_node
/launch_ros_29221
/oakd
/oakd_container
/odometer                <-- non-TB4-node:  Logs all robot travel
/robot_state_publisher
/rplidar_composition
/say                          <-- non-TB4-node: Text-to-speech server
/teleop_twist_joy_node
/turtlebot4_diagnostics
/turtlebot4_node
/wali_node                <--- non-TB4-node:  monitors battery to call /undock at 100%, /rotate at 20%, /dock at 18%
```

Raspberry Pi CPU Usage:
```
********** TB5-WaLI MONITOR ******************************
Saturday 02/01/25
 20:52:26 up 1 day, 20:12,  2 users,  load average: 0.67, 0.72, 0.73
temp=67.0'C
frequency(0)=2300023040
throttled=0x0
               total        used        free      shared  buff/cache   available
Mem:           7.8Gi       1.0Gi       5.9Gi        13Mi       1.1Gi       6.8Gi
Swap:             0B          0B          0B

%CPU %MEM CMD
16.5  0.9 /usr/bin/python3 /home/ubuntu/TB5-WaLI/wali_ws/install/turtlebot4_diagnostics/lib/turtlebot4_diagnostics/diagnostics_updater --ros-args -r __ns:=/ -r /diagnostics:=diagnostics -r /diagnostics_agg:=diagnostics_agg -r /diagnostics_toplevel_state:=diagnostics_toplevel_state

%CPU %MEM CMD
 8.7  0.5 /opt/ros/jazzy/lib/create3_republisher/create3_republisher --ros-args -r __ns:=/ --params-file /home/ubuntu/TB5-WaLI/wali_ws/install/turtlebot4_bringup/share/turtlebot4_bringup/config/republisher.yaml --params-file /tmp/launch_params_vy_exz1b -r cmd_vel:=cmd_vel_unstamped -r cmd_vel_stamped:=cmd_vel
 3.6  0.5 /home/ubuntu/TB5-WaLI/wali_ws/install/turtlebot4_node/lib/turtlebot4_node/turtlebot4_node --ros-args -r __ns:=/ --params-file /tmp/tmp8xqiit1_ --params-file /tmp/launch_params__6ky3_n5
 3.1  0.3 /home/ubuntu/TB5-WaLI/wali_ws/install/sllidar_ros2/lib/sllidar_ros2/sllidar_node --ros-args -r __node:=rplidar_composition -r __ns:=/ --params-file /tmp/launch_params_ump9bh3w
 3.0  2.4 /opt/ros/jazzy/lib/rclcpp_components/component_container --ros-args -r __node:=oakd_container -r __ns:=/
 2.5  0.9 /usr/bin/python3 /home/ubuntu/TB5-WaLI/wali_ws/install/wali/lib/wali/odometer
 1.1  0.8 /usr/bin/python3 /opt/ros/jazzy/lib/joint_state_publisher/joint_state_publisher --ros-args -r __node:=joint_state_publisher -r __ns:=/ --params-file /tmp/launch_params_c0nvjwd_ -r /tf:=tf -r /tf_static:=tf_static
 0.3  0.4 /opt/ros/jazzy/lib/robot_state_publisher/robot_state_publisher --ros-args -r __node:=robot_state_publisher -r __ns:=/ --params-file /tmp/launch_params_sybfpv08 --params-file /tmp/launch_params_7alpbuyk -r /tf:=tf -r /tf_static:=tf_static
 0.1  1.0 /usr/bin/python3 /home/ubuntu/TB5-WaLI/wali_ws/install/wali/lib/wali/wali_node
 0.0  0.9 /usr/bin/python3 /opt/ros/jazzy/bin/ros2 launch /tmp/turtlebot4.launch.py
 0.0  0.8 /usr/bin/python3 /home/ubuntu/TB5-WaLI/wali_ws/install/wali/lib/wali/say_node
 0.0  0.4 /usr/bin/python3 /opt/ros/jazzy/bin/ros2 run wali wali_node
 0.0  0.4 /usr/bin/python3 /opt/ros/jazzy/bin/ros2 run wali say_node
 0.0  0.4 /usr/bin/python3 /opt/ros/jazzy/bin/ros2 run wali odometer
 0.0  0.0 /bin/bash /usr/sbin/turtlebot4-start
```

slowrunner, 2025-02-04:
Two goals:  
1) understand possible reason Create3 claims seeing more nodes than actually exist
2) Learn how to optimize Create3 performance -( I returned a prior Create3 when RTABmap traffic crashed the Create3 before republisher release)

I have the feeling when I see this warning in the Create3 log that I need to cold boot my Turtlebot4 with center button 7 seconds or perhaps with /robot_power while off the dock?

Simba185, 2025-02-04:
Ok good ideas but I'm guessing its not tested as a working solution yet?

Clearpath really owes us to make a custom firmware for the Create3 that ACTUALLY works Reliably. After over 1 year of waiting for updates and fixes its still not reliable like a Yaskawa or other real robot.   I wonder if i will just have to do it my self ( or build it with my own hardware.)

smatarCPR, 2025-02-04:
Hello @slowrunner ,
Thank you for reaching out on the Turtlebot 4 Github page. I have a couple of questions regarding the current state of the Create3:

- I notice that you are currently running a discovery server setup. I wanted to double check that you have not connected the Create3 to a wi-fi network correct?
- In terms of the high number of nodes being launched, have you noticed any performance issues or at this point is it only the warning? Reviewing the details it appears that currently the Create3 is operating at about 60-70% of its capacity leaving some room for additional load.

Looking forward to hearing from you soon.
Best Regards,
Saif

slowrunner, 2025-02-04:
> * double check that you have not connected the Create3 to a wi-fi network correct?

Oh wow - totally ignored the first steps:
```
Create® 3

The Create® 3 needs to be updated to the latest firmware and have its Wi-Fi disabled.

Setup instructions:

Update to the [latest firmware](https://iroboteducation.github.io/create3_docs/releases/overview/) using the webserver.
Once updated, perform a [factory reset](https://iroboteducation.github.io/create3_docs/webserver/about/#:~:text=set%20to%20USB.-,Factory%20Reset,-A%20hyperlink%20to) to disconnect the Create® 3 from any Wi-Fi networks.
 Note
Ensure that this factory reset is done before the remaining setup as it will clear out the settings that are written to the Create® 3 by the Raspberry Pi.
```

I'll backtrack to these steps and re-perform the setup.

> * In terms of the high number of nodes being launched, have you noticed any performance issues or at this point is it only the warning?

For sure - it was complaining about 66 nodes and I ran turtlebot4_navigation async slam to map my house.  The mapping produced a terrible map, so I quit and restarted nav and rviz2 - the Create3 complained bout 99 nodes and hit 99% CPU and did not publish topics at required intervals until I stopped navigation process.  

I will repeat after following the Discovery Server setup (fully).

slowrunner, 2025-02-05:
Completed the factory reset, and re-configuring the Create3 - all back up and running well.

Ran tb4_navigation and the Create3 log is no longer reporting excessive nodes.

Thanks for the diagnosis.

closed 2025-02-05 by slowrunner