turtlebot/turtlebot4 #537: TurtleBot4 defective batteries and doesn't boot up

paoloelle, 2025-02-04:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Discovery Server

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

I cannot check the version of the package since one robot has only a battery problem and the other one doesn't boot up.

### Type of issue

Battery

### Expected behaviour

The robots don’t shut down once removed from the docking station and boot up properly.

### Actual behaviour

I am writing to report issues with two TurtleBot4 robots. Both robots are experiencing technical problems, as described below:
	•	TurtleBot 1: The battery does not hold a charge.
	•	TurtleBot 2: The battery does not hold a charge, and the Create3 does not boot up. Additionally, the Raspberry Pi screen does not turn on.

The problem occurs as soon as I remove the robot from the charging station, it shuts down immediately. I tested both robots with a fully functional battery, and this issue did not occur. As a result, I was able to update TurtleBot 1. However, TurtleBot 2 is experiencing another issue beyond the battery: the Create3 does not boot up (LED ring spinning white), and the Raspberry Pi screen remains off.

### Error messages

```bash

```

### To Reproduce

1. Place the robot on docking station
2. Remove the robot from the docking station

RustyCPR, 2025-02-04:
Hi there @paoloelle 
Thanks for creating this ticket. 

Could you remove the batteries for 30 minutes from the Create3 base to reset the battery logic circuitry. See instructions here: [LINK](https://github.com/turtlebot/turtlebot4/issues/428#issuecomment-2192172795)

Then reinstall the battery and please review these instructions and let me know if anything changes:[ LINK
](https://github.com/turtlebot/turtlebot4/issues/276#issuecomment-1732663588)
Thanks,
Rusty

paoloelle, 2025-02-05:
@RustyCPR I followed your instructions and the problems mentioned above still hold:

- TurtleBot 1: If I put the robot in the charging dock it turns on but the Raspberry Pi screen doesn't turn on. Once I remove the robot from the dock it suddenly shuts down. Wiring connections are ok.

- TurtleBot2: Create3 doesn't boot up and the Raspberry Pi screen doesn't turn on. Once I remove the robot from the dock it suddenly shuts down. Wiring connections are ok.

RustyCPR, 2025-02-05:
Hi @paoloelle 
Thanks for the reply. 
Can you please remove the Create3 adapter board and verify:
- The USB /BT toggle switch is set to USB
- Make sure the Battery power output cable and USB are securely connected. 

Please also verify: 
-Please check the other ends of the above connections are securely connected.  
-Please make sure the RPi ribbon cable is securely connected to the top PCBA. 
-When booting does the Create3 play a chime? 
-What is the color ring light on the Create3 showing: https://iroboteducation.github.io/create3_docs/hw/face/

[image]

[image]

[image]

The more details you provide the better I can help,
Thanks,
Rusty

paoloelle, 2025-02-10:
@RustyCPR  thanks for the suggestions. I've checked out the robots and here is a detailed description:

- TB1:

  - Ribbon cable is securely and correctly connected.
  - The usb/bt switch is set to USB.
  - Both the battery and the usb cable are securely connected both on the create3 adapter and the other end

  - If I put it on the station the following happens:
   - lidar spins for a few seconds
   - light ring shows "Spinning White" animation
   - plays chime
   - light ring shows "Partial White" animation. The animation shows 0% charge.
   - display is still off
   - after 30min on charging station it is still the same. (display off, animation the same and still shows 0%)

  - After I take it from the station the following happens:
   - light ring turns of immediately
   - does not respond to any button presses or anything

- TB2:
 - Ribbon cable is securely and correctly connected.
 - The usb/bt switch is set to USB.
 - Both the battery and the usb cable are securely connected both on the create3 adapter and the other end
 
 - If I put it on the station the following happens:

 - lidar spins for a few seconds
 - light ring shows "Spinning White" animation
 - display is off
 - after 30min on charging station it is still the same (display off, still "Spinning White animation")

- After I take it from the station the following happens:
 - light ring turns of immediately
   - does not respond to any button presses or anything

RustyCPR, 2025-02-11:
Thank you for the details @paoloelle 
Can you please confirm if the RPi is powered on while the Create3 is powered on the charger base? 
Can you check and verify if the USB-A to USB-B cable is connected from RPi to the PCB? 
Do you see any LEDs on the PCB board above the LCD? 

Thanks,
Rusty

paoloelle, 2025-02-12:
@RustyCPR For both of the robots the behavior is the same:

- The USB-A to USB-B cable is securely connected
- When the robot is on the charging station:
    * All of the LEDs above the LCD screen are turned off
    * The Lidar turns on and while it spins two LEDs on the RPi go on. A red one labeled PWR and a green one labeled ACT. Then Lidar and the LEDs of the RPi turn off at the same time and nothing happens anymore

RustyCPR, 2025-02-13:
Hi @paoloelle 
Thanks for your feedback! 

Are you able to keep the TB4 on the charger and while its charging get the Create3 logs and about page from the create3 webserver? https://turtlebot.github.io/turtlebot4-user-manual/setup/simple_discovery.html#create-3

Can you reimage the RPi but make sure that it is imaged with the standard version (or go into the turtlebot4 setup tool and make sure it is set to standard).  https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html#recovering-the-raspberry-pi

It appears you may have both a battery problem and a settings problem. 
If its a battery issue batteries can be sourced here:  https://www.irobot.com/en_US/roomba-accessories/4624864.html

Best regards,
Rusty

paoloelle, 2025-02-17:
HI @RustyCPR. These are my feedback:

- TB1
    - I'm not able to connect to `tb1_ip_address:8080` to open the webserver of the Create3
    - If I put the robot in the hotspot mode (pressing Button 1 and 2) I can connect to the Create3 webserver.
    - Attached you can find what the logs and the about page says.

[logs_TB1.txt](https://github.com/user-attachments/files/18827928/logs_TB1.txt)
[about_TB1.txt](https://github.com/user-attachments/files/18827935/about_TB1.txt)

- TB2
    - I was not able to connect to it. I can't even activate the robot's hotspot by pressing and holding Button 1 and 2.

I also tried with a functioning battery and these are the results:

- TB1: The robot works again perfectly. Like expected.
 
- TB2: The robot still doesn't work.

RustyCPR, 2025-02-20:
Hi @paoloelle 
Thank you for confirming and running the tests. 
For TB2, can you please: 
1. Place TB2 on charging dock and power on
2. SSH into [RPi](https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html#connect-the-raspberry-pi-to-your-network)
3. Run the [Setup tool](https://turtlebot.github.io/turtlebot4-user-manual/software/turtlebot4_setup.html#configuration-tools)
4. Change the ROBOT_NAMESPACE within Bash Setup, and apply settings
5. If error occurs please share logs `sudo journalctl -u turtlebot4 -r`

Thanks,
Rusty

paoloelle, 2025-02-26:
Hi @RustyCPR 
Unfortunately I can't ssh into TB2, because it is not in access point mode and I don't know the IP address (the display is off).

RustyCPR, 2025-03-03:
Hi @paoloelle 
Thanks for the details. 
What you are describing is when the Create3 has a invalid configuration. 
You keep describing trying to enable WIFI. Have you tried [SSH'ing into the RPi with a USB cable](https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html#recovering-the-raspberry-pi
)?
You can then get the IP by running: `ros2 topic echo /ip`
 Run the [Setup Tool](https://turtlebot.github.io/turtlebot4-user-manual/software/turtlebot4_setup.html#configuration-tools) 
Make sure your namespace is valid. 
Let me know,
Rusty

paoloelle, 2025-03-04:
Hi @RustyCPR 
Thanks for the feedback.
I finally was able to ssh into the TB4 by connecting to it via ethernet and plugging the USB-C cable into my pc. I ran the setup tool and corrected the namespace. After that, the raspberry pi kind of worked. All the topics appeared and also data was published on them. But the display and the four LEDs were still all off. 
Afterwards I connected the USB-C again to the TB4 power supply and placed it on the charging station. Then I had the same problem as before (Charge3 won't power on, display and LEDs stay off, I can't even ssh into the RPi anymore it just times out).
If I plug the USB-C cable into the computer again, it works again.

RustyCPR, 2025-03-10:
Hi @paoloelle 
Thanks for the details. It seems like progress is being made. 

_**I finally was able to ssh into the TB4 by connecting to it via ethernet and plugging the USB-C cable into my pc.**_

- A. Please make sure you have a ethernet connection from the remote PC to RPi. [SSH'ing into the RPi with a USB cable](https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html#recovering-the-raspberry-pi)?
- B. Verify if the RPi is connected to your WiFi Network 
- C. Perform a package update.  `sudo apt update && sudo apt upgrade`
- D. What is the updated namespace you used? 

**_Afterwards I connected the USB-C again to the TB4 power supply and placed it on the charging station. Then I had the same problem as before (Charge3 won't power on, display and LEDs stay off, I can't even ssh into the RPi anymore it just times out)._**

- Restore the USB-C connection to the Create3
- Keep the ethernet connection from the remote PC to RPi. 
- [Can you SSH into the RPi via Ethernet? ](https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html#recovering-the-raspberry-pi)

Thanks,
Rusty

paoloelle, 2025-03-12:
Hi @RustyCPR 
I followed your instructions:

- A. I established a working ethernet connection from my PC to the RPI.
- B. I checked again and the wifi is correctly configured.
- C. I ran it. Now there are no more updates available.
- D. I used the namespace: turtlebot4_13

If I put the bot on the charging station then restore the USB-C connection to the Create3, the Lidar stops spinning and does not turn on again. Also the ethernet gets disconnected. I think the RPI gets no power anymore and is turned off.

RustyCPR, 2025-03-13:
Hi @paoloelle 
Please check the Create3 power distribution board is fully inserted into the Create3 and the USB and power cable from this board are fully connected on both sides. See video. 

https://drive.google.com/file/d/1qsUDnqPVHvorud5E4nWRn53PREYXimiz/view?usp=drive_link

paoloelle, 2025-03-20:
Hi @RustyCPR ,
 I checked the power distribution board as well as the cables and everything is perfectly plugged in

RustyCPR, 2025-03-24:
HI @paoloelle  if the PCBA LEDs and LCD are not working - send me a message with the TB4 serial number to support[at]clearpathrobotic[dot]com and I'll look into sending a replacement PCBA. 
Thanks for your patience into debugging the issue.
Rusty

RustyCPR, 2025-04-17:
Hi @paoloelle I will close this ticket as we are corresponding offline. 
Thanks,
Rusty

closed 2025-04-17 by RustyCPR