turtlebot/turtlebot4 #542: Why does Turtlebot4 need external time server?  Why not the RPi?

slowrunner, 2025-02-08:
### Robot Model

Turtlebot4 Lite

### ROS distro

Humble

### Networking Configuration

Discovery Server

### OS

Ubuntu 22.04

### Built from source or installed?

Built from Source

### Package version

turtlebot4/turtlebot4  [dc744fe](https://github.com/turtlebot/turtlebot4/commit/dc744fee51ee681ae9092fda521e98a4a4c23330)
turtlebot4/turtlebot4_robot  [28e115f](https://github.com/turtlebot/turtlebot4_robot/commit/28e115f86930bb3273f1a1aa4738dbd8747513f0)

### Type of issue

Navigation (SLAM, Nav2 etc.)

### Expected behaviour

Expect turtlebot4 installation/setup would configure Create3 to sync time with Raspberry Pi.

# Why not run chrony on the RPi?

A long time ago the [Create3 folk suggested setting up chrony on the Raspberry Pi](https://iroboteducation.github.io/create3_docs/setup/compute-ntp/)  but the  [Issue 530]( https://github.com/turtlebot/turtlebot4/issues/530) suggests an external time server.  Why not setup chrony on the RPi?

### Actual behaviour

Create3 and Raspberry Pi connected LIDAR topics report unsynchronized frame clock times sometimes causing terrible SLAM quality.

[image]

### Error messages

```bash
Create3:

Feb  8 16:25:08 iRobot-883C456BE1E840DC89E64B28B4F75B53 user.notice ntpd: Script called in periodic mode
Feb  8 16:25:18 iRobot-883C456BE1E840DC89E64B28B4F75B53 user.notice create-platform: [INFO] [1739031918.563333611] [_do_not_use.system_health]: CPU usage: max 67 [%] mean 59 [%] RAM usage: 35/59 [MB]
Feb  8 16:26:18 iRobot-883C456BE1E840DC89E64B28B4F75B53 user.notice create-platform: [INFO] [1739031978.563532702] [_do_not_use.system_health]: CPU usage: max 63 [%] mean 56 [%] RAM usage: 35/59 [MB]
Feb  8 16:27:18 iRobot-883C456BE1E840DC89E64B28B4F75B53 user.notice create-platform: [INFO] [1739032038.569151499] [_do_not_use.system_health]: CPU usage: max 63 [%] mean 57 [%] RAM usage: 35/59 [MB]
Feb  8 16:27:25 iRobot-883C456BE1E840DC89E64B28B4F75B53 user.notice ntpd: ntpd: timed out waiting for 192.168.186.1, reach 0x00, next query in 522s
Feb  8 16:28:18 iRobot-883C456BE1E840DC89E64B28B4F75B53 user.notice create-platform: [INFO] [1739032098.569115222] [_do_not_use.system_health]: CPU usage: max 65 [%] mean 60 [%] RAM usage: 35/59 [MB]
Feb  8 16:28:50 iRobot-883C456BE1E840DC89E64B28B4F75B53 user.notice ntpd: ntpd: bad address '3.irobot.pool.ntp.org'
Feb  8 16:28:55 iRobot-883C456BE1E840DC89E64B28B4F75B53 user.notice ntpd: ntpd: bad address '2.irobot.pool.ntp.org'
Feb  8 16:29:00 iRobot-883C456BE1E840DC89E64B28B4F75B53 user.notice ntpd: ntpd: bad address '1.irobot.pool.ntp.org'
Feb  8 16:29:05 iRobot-883C456BE1E840DC89E64B28B4F75B53 user.notice ntpd: ntpd: bad address '0.irobot.pool.ntp.org'
```

### To Reproduce

1) Setup turtlebot4 (and Create3) with Discovery Server per Turtlebot4 Manual
2) launch async SLAM:
```
ros2 launch turtlebot4_navigation slam.launch.py sync:=false params:=/opt/ros/jazzy/share/turtlebot4_navigation/config/slam.yaml
```

### Other notes

Guidance is to set up an external time server for both Create3 and Raspberry Pi:  [Issue 530]( https://github.com/turtlebot/turtlebot4/issues/530)

Default Create3 ntp.conf:
```
# iRobot servers
server 0.irobot.pool.ntp.org iburst
server 1.irobot.pool.ntp.org iburst
server 2.irobot.pool.ntp.org iburst
server 3.irobot.pool.ntp.org iburst
# SBC servers
server 192.168.186.1 iburst
server 192.168.186.3 iburst
```

Default Raspberry Pi /etc/ntpsec/ntp.conf:
```
# Use servers from the NTP Pool Project. Approved by Ubuntu Technical Board
# on 2011-02-08 (LP: #104525). See https://www.pool.ntp.org/join.html for
# more information.
pool 0.ubuntu.pool.ntp.org iburst
pool 1.ubuntu.pool.ntp.org iburst
pool 2.ubuntu.pool.ntp.org iburst
pool 3.ubuntu.pool.ntp.org iburst

# Use Ubuntu's ntp server as a fallback.
server ntp.ubuntu.com

# Access control configuration; see /usr/share/doc/ntpsec-doc/html/accopt.html
# for details.
#
# Note that "restrict" applies to both servers and clients, so a configuration
# that might be intended to block requests from certain clients could also end
# up blocking replies from your own upstream servers.

# By default, exchange time with everybody, but don't allow configuration.
restrict default kod nomodify nopeer noquery limited

# Local users may interrogate the ntp server more closely.
restrict 127.0.0.1
restrict ::1

```

RPi Time Status:
```
ubuntu@TB5WaLI:~/TB5-WaLI/wali_ws$ ntpstat
synchronised to unspecified at stratum 2 
   time correct to within 71 ms
   polling server every 1024 s

ubuntu@TB5WaLI:~/TB5-WaLI/wali_ws$ timedatectl status
               Local time: Sat 2025-02-08 11:43:05 EST
           Universal time: Sat 2025-02-08 16:43:05 UTC
                 RTC time: Sat 2025-02-08 16:43:05
                Time zone: America/New_York (EST, -0500)
System clock synchronized: yes
              NTP service: n/a
          RTC in local TZ: no

ubuntu@TB5WaLI:~/TB5-WaLI/wali_ws$ ntpq -c rv
associd=0 status=0015 leap_none, sync_unspec, 1 event, clock_sync,
leap=00, stratum=2, precision=-23, rootdelay=66.984, rootdisp=36.931, refid=217.180.209.214, reftime=eb52030e.5a6ef22a 2025-02-08T16:21:34.353Z, tc=10, peer=17778,
offset=0.367545, frequency=0.691559, sys_jitter=2.527599, clk_jitter=0.949599, clock=eb520853.61521e02 2025-02-08T16:44:03.380Z, processor="aarch64",
system="Linux/6.8.0-1018-raspi", version="ntpd ntpsec-1.2.2", clk_wander=0.019468, tai=37, leapsec="2017-01-01T00:00Z", expire="2025-06-28T00:00Z", mintc=0

```

hilary-luo, 2025-02-20:
@slowrunner this response is going to focus on the conceptual topic since I haven't looked into the details of the setup script vs setup tool yet. 

I think you correctly figured out that the expected normal operation is that chrony be running on the pi (syncing with NTP servers on the internet) and that the create3 be served by the NTP server on the pi. Most people use the default images so that is probably why we haven't seen this come up before as it is set up correctly in those images. However there are limitations with this configuration. It ensures that the create3 timestamp always match with the pi timestamp and if the pi has an internet connection it also will match the internet time servers. This is sufficient when running all nodes onboard the Turtlebot4, or when working with computers where both the pi and the offboard computers have internet access and are syncing with the internet time servers.

If the system is not connected to the internet then you run into an issue where the pi4 does not have the ability to keep time while it is powered off, which can be problematic in itself if events are being scheduled or logged based on time of day. Additionally, the disconnected pi would not match the timestamp on other devices within the network. Whenever communication is intended between nodes on different computers it is important to have their time synchronized. This also becomes an issue if the offboard computers have just had time synchronization with the internet disabled. These are the situations where we recommend hosting a local server on the local network, to replace the usage of the internet time servers. Even in the case for devices like the pi5 where there is support for RTC, it is still important to have that time synchronized across devices.

slowrunner, 2025-02-21:
Super - The rationale for some folk to establish the time server on the visualization/remote computing platform was foreign to me due to my narrow "only onboard computing" focus.  Running the time server on the Pi is perfect for my use case, where my Pi is on 24/7/365 and connected to the Internet 99.x% of that.

Thanks.

closed 2025-02-21 by slowrunner