turtlebot/turtlebot4 #543: Inconsistent Navigation on Existing Maps

EdBrayUTS, 2025-02-10:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Discovery Server

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

`turtlebot4_navigation` version `1.0.5-1jammy.20250114.201201`

All packages on both remote PC and Raspberry Pi fully-updated at time of opening this issue.

### Type of issue

Navigation (SLAM, Nav2 etc.)

### Expected behaviour

The robot to be able to autonomously navigate to goals provied in RViz using a pre-existing map of the environment.

### Actual behaviour

This behaviour is very inconsistent. Very rarely, the everything works as expected. However typically something crashes when launching `nav2.launch.py`. An example error log is shown below. 

The error is not always the same, but it is typically `lifecycle_manager` being unable to reach either `controller_server` or `smoother_server` after 4 seconds by bond. In the below instance, it was `controller_server` that cannot be reached.

### Error messages

```bash
[INFO] [launch]: All log files can be found below /home/ed/.ros/log/2025-02-10-11-33-39-912698-Unhygienix-24041
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [controller_server-1]: process started with pid [24042]
[INFO] [smoother_server-2]: process started with pid [24044]
[INFO] [planner_server-3]: process started with pid [24046]
[INFO] [behavior_server-4]: process started with pid [24048]
[INFO] [bt_navigator-5]: process started with pid [24050]
[INFO] [waypoint_follower-6]: process started with pid [24052]
[INFO] [velocity_smoother-7]: process started with pid [24054]
[INFO] [lifecycle_manager-8]: process started with pid [24056]
[velocity_smoother-7] [INFO] [1739147620.022728425] [tb4.velocity_smoother]: 
[velocity_smoother-7] 	velocity_smoother lifecycle node launched. 
[velocity_smoother-7] 	Waiting on external lifecycle transitions to activate
[velocity_smoother-7] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[waypoint_follower-6] [INFO] [1739147620.026065141] [tb4.waypoint_follower]: 
[waypoint_follower-6] 	waypoint_follower lifecycle node launched. 
[waypoint_follower-6] 	Waiting on external lifecycle transitions to activate
[waypoint_follower-6] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[waypoint_follower-6] [INFO] [1739147620.026408458] [tb4.waypoint_follower]: Creating
[lifecycle_manager-8] [INFO] [1739147620.026702640] [tb4.lifecycle_manager_navigation]: Creating
[bt_navigator-5] [INFO] [1739147620.027269513] [tb4.bt_navigator]: 
[bt_navigator-5] 	bt_navigator lifecycle node launched. 
[bt_navigator-5] 	Waiting on external lifecycle transitions to activate
[bt_navigator-5] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[bt_navigator-5] [INFO] [1739147620.027338303] [tb4.bt_navigator]: Creating
[lifecycle_manager-8] [INFO] [1739147620.027507127] [tb4.lifecycle_manager_navigation]: Creating and initializing lifecycle service clients
[controller_server-1] [INFO] [1739147620.029482791] [tb4.controller_server]: 
[... 148 lines cut ...]
Running `sudo chronyc sources` on the Pi now shows:
```
MS Name/IP address         Stratum Poll Reach LastRx Last sample               
===============================================================================
^- alphyn.canonical.com          2   6   377    55  -2317us[-2317us] +/-  326ms
^- prod-ntp-4.ntp1.ps5.cano>     2   6   375   252  +3336us[+3336us] +/-  130ms
^- prod-ntp-5.ntp4.ps5.cano>     2   6   377   252  +2938us[+2938us] +/-  130ms
^- prod-ntp-3.ntp1.ps5.cano>     2   6   377   253  +1372us[+1372us] +/-  131ms
^- smtp.juneks.com.au            1   6   377   320   -189us[  +50us] +/- 3602us
^- auvps1.sameteem.com           5   6   376   253   -228us[ -228us] +/- 5530us
^- syd.clearnet.pw               2   6   377   252   -321us[ -321us] +/-   27ms
^- time.cloudflare.com           3   6   377   254   -589us[ -589us] +/- 4914us
^* <remote_pc_ip>                2   6   377   255   +453us[ +918us] +/- 6733us
```

And running `sudo chronyc clients` on the remote PC gives:
```
Hostname                      NTP   Drop Int IntL Last     Cmd   Drop Int  Last
===============================================================================
<raspberry_pi_ip>                 31      0   6   -    20       0      0   -     -
```

This indicates to me that the remote PC is the preferred source, but sync doesn't seem perfect. Could this be the root of the issue?

I've also observed the nodes started by `localization.launch.py` and `nav2.launch.py` take a long time to find services eg. `controller_server/get_state` (as in this example) or the `map_server` services. Sometimes they don't find them after 30 seconds, and I restart the command. Is this possibly related?

pvg6, 2025-02-11:
Same issue i am facing.. if find any solution please help me...
@EdBrayUTS

beardyFace, 2025-02-17:
I am having the exact same issue with the same setup (Discovery Server with Namespace) - for me it is consistently the smoother server that fails: 

failed to send response to /T14/smoother_server/change_state (timeout)

Sometimes it all works fine but more often than not it throws the timeout

EdBrayUTS, 2025-02-17:
@beardyFace thanks for the link. I've just tried installing packages within the `navigation2`, `turtlebot4`, and `bond_core` repositories from source (commits `1c68c212db01f9f75fcb8263a0fbb5dfa711bdea`, `e8b710f9c4976950af2b92693bc861c7837b36b5`, and `fb7ed7cbd5572c14f0176f987ac45ef7b45fc952` respectively).

The first time starting everything from a fresh boot everything worked as it should! But I've not been able to replicate it. The next time I launched everything I was met with:

```
[planner_server-3] [INFO] [1739833291.487637254] [tb4.planner_server]: Planner Server has GridBased  planners available.
[lifecycle_manager-8] [ERROR] [1739833294.433804758] [tb4.lifecycle_manager_navigation]: Failed to change state for node: planner_server. Exception: planner_server/get_state service client: async_send_request failed.
[lifecycle_manager-8] [ERROR] [1739833294.433905890] [tb4.lifecycle_manager_navigation]: Failed to bring up all requested nodes. Aborting bringup.
```

Then on several subsequent attempts (even after a restart of the robot) it gets stuck at:

```
[lifecycle_manager-8] [INFO] [1739833902.813031693] [tb4.lifecycle_manager_navigation]: Waiting for service controller_server/get_state...
```

I also tried directly over ethernet, which similarly worked once then not again. It seemed to get past waiting for servers, but struggled to call them, giving errors like:

```
[smoother_server-2] [WARN] [1739835197.561420748] [tb4.smoother_server.rclcpp]: failed to send response to /tb4/smoother_server/change_state (timeout): client will not receive response, at ./src/rmw_response.cpp:154, at ./src/rcl/service.c:314
```

So far none of the things I've tried (setting up timesync, building packages from source, going over ethernet, manually changing the `bond_timeout` parameter of the `lifecycle_manager` node) seems to have a statistically significant effect on whether things work or not, or even which of the many possible errors show up.

beardyFace, 2025-02-18:
I installed the latest versions of Nav2 and bond_core from Deb, as the releases seem to align with the current mains, but that didn't work. Same as you - it works maybe once if I am lucky, but it is inconsistent if it ever works at all.  

Thanks for taking the time to test from the source; that was my next step, saved me some time.  

I have tried running this with both SLAM, and localisation - same issue in each case, it does appear to be the Nav2 package failing to sync the service for some reason.

slowrunner, 2025-02-18:
I am seeing the same issue in Jazzy.  Just had to launch navigation 7 times to get all components to successfully come up.

beardyFace, 2025-02-20:
@slowrunner tried the suggestion, however no luck on my end with Humble. The use_sim_time params are all false but still times out. 

Interestingly, when I run it all without the modified params file the use_sim_time comes out as false with the examples already, even though they are true in the yaml files...

EdBrayUTS, 2025-02-20:
@slowrunner just tried that myself and no luck I'm afraid. Without changing anything, all the relevant nodes already had `use_sim_time` set to `False` when I checked after launching.

slowrunner, 2025-02-20:
Bummer @beardyFace and @EdBrayUTS 

Perhaps because I am using the RPi as the time server and running loc and nav on the RPi5, it improves for me, or perhaps Jazzy.  

Flakey behaviors are the toughest to diagnose.  Wishing you success soon.

slowrunner, 2025-02-24:
@EdBrayUTS IMO, this flakey nav2 bringup is not caused by our configurations.  With multiple folks seeing it with slightly different configurations, I am starting to believe that something changed in the nav2 code base that has found an edge case in  the bringup.

This morning, I did a ```turtlebot4-service-restart``` and rebooted my visualization platform and then had to launch navigation three times to get a working system.  Two days ago, I had to launch five times before I got a working system.  

I'm going to try removing the docking server plugin from the list of startup plugins, that was the major addition for the latest releases.  Somehow we have to characterize the failure, as it doesn't look like any of us have caught the attention of the god of nav2.  (@SteveMacenski)

hilary-luo, 2025-03-03:
Hi everyone, there is a lot going on here so please bear with me

@EdBrayUTS 
- After you set up the time sync with the remote PC did the `the timestamp on the message is earlier than all the data in the transform cache` messages go away? That should fully address the time sync issue but would be independent from the timeout issue.
- The issue regarding localization and nav2 nodes finding each other would be separate from the time sync. I have seen this type of issue before when running nav2 on a system with insufficient resources. Are you running this natively? or in a VM / container? Take a look at your resource utilization (CPU usage etc) during launch when the timeouts occur. If the resources are maxed out then it makes sense that the nodes are unable to fully launch in time. Let me know how this goes and see my note at the end about changing the timeouts
- Similarly can you monitor the resource usage on the pi during this time? You could try disabling the diagnostics on the pi (through the turtlebot4-setup tool) to see if that helps the situation. The diagnostics are known to create extra load on the pi. This should not impact the ability of nodes on the remote pc being able to discover eachother but overall would impact the reliability of the system as a whole.
- In regards to both the remote PC and the pi, try to make sure that all other processes are turned off when doing this test. 
- Finally, can you confirm that you have the whole system running on a 5GHz wifi network and that the create3 is disconnected from the internet (since you are using Humble + Discovery Server).

@pvg6 
- Do you have your own ticket open with the details about your system? I can't really say anything without knowing the details but hopefully the specifics I share for others can help you as well.

@beardyFace 
- Are you launching the navigation nodes on the pi or on an offboard computer?
- It makes sense that the use_sim_time change would not have impacted you on Humble as that issue is specific to Jazzy and will be fixed shortly.
- Consider the notes I shared for EdBrayUTS if they apply to you

@slowrunner 
- The issue with use_sim_time has been isolated to an error when the tb4 nav configs were ported to jazzy (I will be responding to you on that other discussion shortly about it). Are you still experiencing inconsistency with running nav despite your workaround and having switched to discovery server?

For everyone:
I would say that although those here are experiencing similar issues with inconsistent performance, we don't know enough about every individual system yet to know that the cause would be the same for each of you. I have most often seen these sorts of timeout issues when dealing with running navigation or simulations on low resource systems. This can be addressed by running nav on an offboard computer that has higher resources and to run all navigation nodes on that same  offboard computer so that discovery doesn't have to happen over the network for the navigation nodes. If you are using a virtual machine then make sure that you have allocated enough resources to it. Barring all of that, since it seems like everyone is on the edge of functional, you can play around with extending the various timeouts that you may be running into. See the timeouts set in the config files here: https://github.com/turtlebot/turtlebot4/tree/jazzy/turtlebot4_navigation/config. You can download the configs and then pass them in when you launch the respective nodes (params:=/path/to/config.yaml). If this works for you please share about which timeouts you needed to change to get the system working reliably. A solution could be that the default timeouts get extended.

beardyFace, 2025-03-03:
@hilary-luo 

Running the SLAM, Nav2, and Viz remotely - the Turtlebot4 is only operating its turtlebot4_robot package with a modified oakd_pro.yaml to turn the stereo pipeline on. Is it possible a poor wifi network (or an overloaded university network) could be causing the connections to fail/struggle to establish in the bond_timeout? The remote PC is relatively powerful and shouldn't be resource-limited.

hilary-luo, 2025-03-03:
@beardyFace depending on your networking configuration that is possible. If you are fully running on discovery server with the only server on the robot, then all of the discovery traffic has to make it to the robot server and back again in order for nodes on your offboard computer to discover each other. If you have a poor or congested wifi network, this could potentially be a problem, particularly if the robot is also sending stereo data across the network.

You could try and isolate out this issue by either
a) Testing over a wired connection with wifi disabled on all devices (make sure you update the server IP address on your offboard pc to point at the wired IP address of the pi)
b) Adding an extra discovery server locally on your offboard computer and launch all nodes on the offboard computer pointing at both servers (that way they can complete discovery locally but also can still access the robot nodes)
c) get a dedicated router to test with (needs to support 5 GHz with Wifi 5 speeds minimum)

Let me know how it goes

beardyFace, 2025-03-04:
Noted. I will test those conditions over the next week or two.

EdBrayUTS, 2025-03-07:
@hilary-luo thanks for the suggestions. I've run some initial tests and had some promising results I'll share with everyone now to see if they help anyone else. I've not checked each change independently to see their relative importance, but what I did was:

* Switched from the university WiFi network to a dedicated router that's not even connected to the internet: just the offboard PC and Turtlebot4 Raspberry Pi
* Cloned the `bond_core`, `navigation2`, and `turtlebot4` repos into a clean `ros2_ws` and built them
* Edited parameters in `turtlebot4/turtlrbot4_navigation/conig/nav2.yaml`:
  * Changed `bt_navigator/default_server_timeout` from `20` to `120`
  * Added `bt_navigator/wait_for_service_timeout` and set to `120000` (parameter is listed in their [docs](https://docs.nav2.org/configuration/packages/configuring-bt-navigator.html))
* Edited `navigation2/nav2_bringup/launch/navigation_launch.py` and `navigation2/nav2_bringup/launch/localization_launch.py` to increase the bond timeout. In each launch file:
  * Find the `lifecycle_manager` node
  * Pass another parameter to it called `bond_timeout` with a value of `120.0` (parameter is listed in their [docs](https://docs.nav2.org/configuration/packages/configuring-lifecycle.html))

I just got it working twice in a row, which is pretty unprecidented. The next bringup then failed, so not perfect though... I often get issues with AMCL actually, for example:
```
[amcl-2] [INFO] [1741314938.685094813] [tb4.amcl]: Subscribed to map topic.
[amcl-2] [WARN] [1741314938.789616876] [tb4.amcl.rclcpp]: failed to send response to /tb4/amcl/change_state (timeout): client will not receive response, at ./src/rmw_response.cpp:154, at ./src/rcl/service.c:314
```
But AMCL doesn't seem to have any timeout parameters we can change.

I'll do more tests when I get time to work out which the most important changes were. Is there a way to change the `bond_timeout` parameter on launch without having to edit the launch file in the Nav2 repo? They're not exposed this as a launch argument, so I don't think so but maybe someone knows a way.

In response to @hilary-luo 's specific questions:

> * After you set up the time sync with the remote PC did the `the timestamp on the message is earlier than all the data in the transform cache` messages go away? That should fully address the time sync issue but would be independent from the timeout issue.

No, this still happens sometimes. My guess is that network traffic is clogging up `tf` messages when everything starts up. On a successful startup, I get this warning intermittently but it eventually settles down. This doesn't seem like a major issue: enough 'correct' messages get through

> * Are you running this natively? or in a VM / container? Take a look at your resource utilization (CPU usage etc) during launch when the timeouts occur.

This is running natively on a pretty powerful offboard PC. A successful startup uses < 5% CPU on this machine.

> * Similarly can you monitor the resource usage on the pi during this time? 

A successful startup results in <25 % CPU usage on the Pi. Again, this doesn't seem unreasonable.
  
> * Finally, can you confirm that you have the whole system running on a 5GHz wifi network and that the create3 is disconnected from the internet

Both the univesity network I was on originally, and the router I'm on now, are 5 GHz networks. The Create3 is not connected to any network.

hilary-luo, 2025-03-07:
@EdBrayUTS Thanks for sharing this info about your system and your testing so far. You may already be aware of this, but with a system this complex I would suggest building these packages for release when you are testing (has performance implications). It won't affect the Turtlebot4 code, but likely would impact Nav2 performance (`colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release`). That being said, given the resources available it probably will not change the outcome on your machine.

In regards to the bond timeout, the easiest answer is that you can expose it as a launch argument yourself. You can look at the `autostart` parameter as an example of how to set this up, just ignore the part about the param_substitutions. So declare the Launch configuration (https://github.com/ros-navigation/navigation2/blob/e4ef813e6b5c6edc4307231e75a9cd01ebbd8c5b/nav2_bringup/launch/navigation_launch.py#L35) and create the Launch argument (https://github.com/ros-navigation/navigation2/blob/e4ef813e6b5c6edc4307231e75a9cd01ebbd8c5b/nav2_bringup/launch/navigation_launch.py#L96-L100) , pass it in as a launch parameter (https://github.com/ros-navigation/navigation2/blob/e4ef813e6b5c6edc4307231e75a9cd01ebbd8c5b/nav2_bringup/launch/navigation_launch.py#L233) and add the action with the rest of the launch option actions (https://github.com/ros-navigation/navigation2/blob/e4ef813e6b5c6edc4307231e75a9cd01ebbd8c5b/nav2_bringup/launch/navigation_launch.py#L332). A similar process can be done in the turtlebot4 launch file if you want to expose it all of the way out as an argument there.

Having done that work you could also contribute that change back to Nav2 since it could be useful for others as well (something to think about).

@ everyone here
A separate thought that I would like you guys to take a look at: is this issue repeatable that it always works the first time after boot and is flaky on subsequent boots? This is going in a bit of a different direction from the other troubleshooting but after you close the nav nodes etc. can you also stop the ROS2 Daemon (`ros2 daemon stop`) and then observe your processes that are running to make sure that there are no residual ROS processes that are still running on your offboard computer? One significant change for Nav2 between Humble and Jazzy is the increased use of Lifecycle nodes and thus more complex lifecycle management.

beardyFace, 2025-03-16:
@hilary-luo - It does appear that it all works fine the first time when everything has just been booted up on either end. Then there are issues on subsequent runs.

I did try restarting the daemon, but it didn't seem to have any impact although I can test further. Is there any more specific commands that can really knock out any processes still running?

slowrunner, 2025-03-17:
@beardyFace @hilary-luo It will be interesting to see if the next rclcpp update has any effect on the reliability.  I've been chasing the runaway oakd container after power_save does /stop_camera  /start_camera pair.  Luxonis made a change for their Jazzy branch that made the problem a "sometimes" and they [submitted an issue against ros2/rclcpp](https://github.com/ros2/rclcpp/issues/2763).  The ROS PMC developer responded they had found some multi-executor race conditions and could we try the latest code.  Luxonis reported the rclcpp fix indeed made the /stop_camera /start_camera sequence reliable again.   I am not able to test that fix at the moment, so will retest the power_save issue when the rclcpp mirrors get updated.

I'm crossing fingers that perhaps the lifecycle issues with navigation might also be race conditions that will be tamed by the rclcpp update.

EdBrayUTS, 2025-03-17:
Slight update from my end. I've isloated the changes I made and the things that made the difference seem to be the combination of:

- Dedicated router, not university network
- Changing `bond_timeout` to `120` in `navigation_launch.py` and `localization_launch.py` within `nav2_bringup`

The newer version of `bond_core` and changing the `bt_navigator` timeout parameters didn't seem to make too much difference. 

Similar to @beardyFace , I'm finding it typically works fine the first time when everything has just booted, but subsequent runs are less reliable.

I've also looked at the nodes that are running after I've cancelled everything (RViz, localisation, and navigation) as suggested @hilary-luo . I can see that often `/tb4/lifecycle_manager_navigation` and/or `/tb4/lifecycle_manager_localization` remains. Restarting the daemon seems to make these nodes disappear, and subsequent startups are more reliable.

With these changes (router, `bond_timeout`, and resarting daemon) I seem to get ~60% reliability when restarting. So making progress, but still not great.

Finally @hilary-luo I'm aware of how one can expose parameters through launch files, but since there are quite a few nested launch files in play here I was more asking if there was a cheaty short cut to avoid having to change all of them 😉  Seems not though

hilary-luo, 2025-03-18:
@beardyFace 
> I did try restarting the daemon, but it didn't seem to have any impact although I can test further. Is there any more specific commands that can really knock out any processes still running?

So after that first attempt has been ended (I assume you are using ctrl+c in the navigation terminals), then open up your preferred version of `top` to look at the processes. A lot of people use `htop`, I prefer `btop`. In there, filter the running processes and see if there are any `ros` or `lifecycle` processes remaining. If there are, then note down which ones are running, then end those processes. Most (if not all) top programs facilitate sending terminate / kill signals to the programs to end them. Then restart your navigation and see if it works the second time.

@slowrunner interesting, please do keep us up to date

@EdBrayUTS 
> I've also looked at the nodes that are running after I've cancelled everything (RViz, localisation, and navigation) as suggested @hilary-luo . I can see that often /tb4/lifecycle_manager_navigation [...]

Were those 2 the only ros processes that were left after everything else was shut down?

> With these changes (router, bond_timeout, and resarting daemon) I seem to get ~60% reliability when restarting. So making progress, but still not great.

Hmmm, better but not good. There has got to be something more happening behind the scenes. I'm going to have to think about this.

> Finally @hilary-luo I'm aware of how one can expose parameters through launch files, but since there are quite a few nested launch files in play here I was more asking if there was a cheaty short cut to avoid having to change all of them 😉 Seems not though

Haha, yah, no. If the parameter were being passed through directly to a node that was already receiving a parameter config file then there would be easier ways. I mean, you could modify the launch of that node to instead receive the same yaml as everything else and then put the parameters into the config (but of course specifying the node name etc in that config file) but I don't think that is really much less effort. It does mean that there is one less layer to modify though.