turtlebot/turtlebot4 #545: Depth Camera Configuration

FGJoseMiguel-AIST, 2025-02-12:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

I do not know

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

I'm not sure, the last one:

https://github.com/turtlebot/turtlebot4_simulator -> the last commit was 12f8f9ebea0b7178fd4c8043b5bd9b5d044b4870 4 months ago

### Type of issue

Camera

### Expected behaviour

I followed the instructions on how to simulate the robot:  

[https://turtlebot.github.io/turtlebot4-user-manual/software/turtlebot4_simulator.html](https://turtlebot.github.io/turtlebot4-user-manual/software/turtlebot4_simulator.html)  

I first want to run some tests using the RGB image and the depth image, but I noticed that, by default, the depth camera only activates when objects are very close. After installing all the robot resources with:  

```
sudo apt-get install ros-humble-turtlebot4
```

I don't know where the files that I need to modify to change this camera setting are located. Could you guide me on what to do?

### Actual behaviour

the depth camera only activates when objects are very close.

### Error messages

```bash

```

### To Reproduce

1. ros2 launch turtlebot4_ignition_bringup turtlebot4_ignition.launch.py slam:=true nav2:=true rviz:=true model:=standard

RustyCPR, 2025-02-28:
Hi there @FGJoseMiguel-AIST 
Thanks for writing in and sorry to see your having issues. 
Please see this reply - https://github.com/turtlebot/turtlebot4/issues/277#issuecomment-1815024838from from @hilary-luo and let me know if you have any additional questions. 
Best regards,
Rusty

RustyCPR, 2025-04-02:
Hi there @FGJoseMiguel-AIST 
Did my previous reply help? 
Let me know,
Thanks,
Rusty

FGJoseMiguel-AIST, 2025-04-07:
Hello, not really, the thing is that right now I'm using the robot only in simulation, not the real one, and because I didn't know how to configure the OAK-D, I changed it for a realsense

closed 2025-05-12 by RustyCPR