turtlebot/turtlebot4 #548: Turtlebot4 Raspberry Pi can't communicate with Create3 on discovery server

aalbaali, 2025-02-16:
### Robot Model

Turtlebot4 Lite

### ROS distro

Humble

### Networking Configuration

Discovery Server

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

```bash
Package: dpkg
Essential: yes
Status: install ok installed
Priority: required
Section: admin
Installed-Size: 6669
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Architecture: arm64
Multi-Arch: foreign
Version: 1.21.1ubuntu2.3
Depends: tar (>= 1.28-1)
Pre-Depends: libbz2-1.0, libc6 (>= 2.34), liblzma5 (>= 5.2.2), libselinux1 (>= 3.1~), libzstd1 (>= 1.4.0), zlib1g (>= 1:1.1.4)
Suggests: apt, debsig-verify
Breaks: libapt-pkg5.0 (<< 1.7~b), lsb-base (<< 10.2019031300)
Conffiles:
 /etc/alternatives/README 7be88b21f7e386c8d5a8790c2461c92b
 /etc/cron.daily/dpkg 94bb6c1363245e46256908a5d52ba4fb
 /etc/dpkg/dpkg.cfg f4413ffb515f8f753624ae3bb365b81b
 /etc/logrotate.d/alternatives 5fe0af6ce1505fefdc158d9e5dbf6286
 /etc/logrotate.d/dpkg 9e25c8505966b5829785f34a548ae11f
Description: Debian package management system
 This package provides the low-level infrastructure for handling the
 installation and removal of Debian software packages.
 .
 For Debian package development tools, install dpkg-dev.
Homepage: https://wiki.debian.org/Teams/Dpkg
Original-Maintainer: Dpkg Developers <debian-dpkg@lists.debian.org>

dpkg-query: package 'ros-humble-robot-base' is not installed and no information is available
Use dpkg --info (= dpkg-deb --info) to examine archive files.
```

### Type of issue

Networking

### Expected behaviour

I expect to have publisher count of topics published by the Create3 platform (e.g., `/imu` topic) to be more than 0, and I expect to see messages when I echo them.

### Actual behaviour

The raspberry pi isn't communicating with the Create3 platform. I came to this conclusion because:
1. the publishers of some Create3 topics (e.g., `/imu`) is 0;
2. can't echo any message.

### Error messages

```bash
No error message. Just nothing being echoed.
```

### To Reproduce

1. Follow the instructions on [this page](https://turtlebot.github.io/turtlebot4-user-manual/setup/discovery_server.html) to setup the discovery server on Turtlebot4.
2. Restart the robot and wait for the welcome message.
3. `ros2 topic info /imu`

### Other notes

- The robot operates as expected when on simple discovery.
- I disconnected the lidar and oak-d camera from the raspberry pi.

Side question: is it possible to ssh into the Create3 platform?

EdBrayUTS, 2025-02-16:
I also experienced the same issue when I was setting up my robot. Have you tried giving your robot a namespace? Take a look at #468 : there's a bug that requires a namespace to be present. It's been fixed in the Jazzy debian release, but not the humble one.

aalbaali, 2025-02-16:
> I also experienced the same issue when I was setting up my robot. Have you tried giving your robot a namespace? Take a look at [#468](https://github.com/turtlebot/turtlebot4/issues/468) : there's a [...]

Giving the robot a namespace fixed it. Thanks for the help!

closed 2025-02-16 by aalbaali