turtlebot/turtlebot4 #551: camera depth

Ubrillo, 2025-02-17:
### Robot Model

Turtlebot4 Standard

### ROS distro

Galactic

### Networking Configuration

Select One

### OS

Ubuntu 20.04

### Built from source or installed?

Installed

### Package version

tutlebot4 desktop

### Type of issue

Camera

### Expected behaviour

How can I  get depth data from turtlebot4 camera. when  I echo **/oakd/rgb/preview/image_raw/compressedDepth** topic, do data is shown

### Actual behaviour

[image]

### Error messages

```bash

```

### To Reproduce

.

### Other notes

.

slowrunner, 2025-02-18:
@Ubrillo  It will be interesting to see what official guidance you get, as there are quite a few ways to launch and use the camera.  

The [manual section](https://turtlebot.github.io/turtlebot4-user-manual/software/sensors.html#running-1) sort of hints at this but does not tell how to ensure "the automatically starting OAK-D node has been stopped or disabled."

The brute force way to accomplish this temporarily is:
```
ps -ef | grep oak
ubuntu@TB5WaLI:~/TB5-WaLI/wali_ws$ ps -ef | grep oak
ubuntu      1632    1467  2 Feb16 ?        01:10:25 /opt/ros/jazzy/lib/rclcpp_components/component_container --ros-args -r __node:=oakd_container -r __ns:=/
ubuntu     14577    2364  0 23:49 pts/0    00:00:00 grep --color=auto oak

$ kill 1632         <<--- PID of the oak container (first PID on the line, don't kill the parent PID!)
```

Now you can run the oak_example.launch.py:

```
ubuntu@TB5WaLI:~/TB5-WaLI/wali_ws$ ros2 launch turtlebot4_bringup oakd_example.launch.py 
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2025-02-17-23-55-18-936794-TB5WaLI-14631
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rgb_stereo_node-1]: process started with pid [14634]
[rgb_stereo_node-1] [184430101175A41200] [3.1] [1.012] [MonoCamera(1)] [error] OV7251 only supports THE_480_P/THE_400_P resolutions, defaulting to THE_480_P
[rgb_stereo_node-1] [184430101175A41200] [3.1] [1.013] [MonoCamera(0)] [error] OV7251 only supports THE_480_P/THE_400_P resolutions, defaulting to THE_480_P
```

and in rviz2
- Click Add -> By topic .. scroll down to /stereo ..  /depth, double click on Image
- You will see the stereo depth image

when done, ctrl-c to quit the oakd_example.launch and

restart the standard preview:
```
ros2 launch turtlebot4_bringup oakd.launch.py &

or 
```
turtlebot4-service-restart
```

smatarCPR, 2025-03-04:
Hello @Ubrillo ,
Thank you for reaching out on the Turtlebot 4 github page. I have the following notes on enabling the depth stereo topic for the camera:

Please note that Galactic has reached end of life and is no longer supported. It is highly recommended that you upgrade to Humble or Jazzy distros.

In order to enable to stereo topic follow this process:

1. Clone, source /opt/ros/humble/setup.bash and build turtlebot4_robot package
2. Modify the oakd_pro.yaml (or oakd_lite.yaml for Turtlebot 4 Lite) file to enable the stereo topic, see examples of different config files here: https://github.com/luxonis/depthai-ros/tree/humble/depthai_ros_driver/config
3. Link to your custom workspace using the turtlebot4-setup tool:

- run turtlebot4-setup
- navigate to ROS Setup -> Bash Setup
- select WORKSPACE_SETUP
- enter the path to your local setup.bash for your workspace. This will allow the turtlebot4 setup tool to locate and source these packages. e.g. /home/administrator/ros_ws/install/setup.bash

4. Reinstall the turtlebot4 service using the turtlebot4-setup tool under Robot Upstart -> Install.
- run turtlebot4-setup
- navigate to ROS Setup -> Robot Upstart
- select Install and agree as necessary

Now your robot should automatically run your local build whenever it is turned on.
The last few steps are using the turtlebot4-setup tool. The full usage of this tool is documented here: https://turtlebot.github.io/turtlebot4-user-manual/software/turtlebot4_setup.html#configuration-tools

Let me know if you have any questions or if I can help with anything else.
Best Regards,
Saif