turtlebot/turtlebot4 #553: Unable to move the robot with the controller or keyboard

YogiMind, 2025-02-20:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Discovery Server

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

Using Turtlebot 4 standard, flashed the Raspberry Pi with [turtlebot_standard_humble_1.0.4](http://download.ros.org/downloads/turtlebot4/) 
and followed the [Basic setup](https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html).

### Type of issue

Other

### Expected behaviour

Connect the controller and drive the robot.

### Actual behaviour

The robot does not move or respond to the controller. I checked the /cmd_vel topic and noticed it was registering the inputs from the controller. This points out the issue must lie within the turtlebot. When i did `ros2 launch turtlebot4_bringup robot.launch.py` and noticed there was an error **Invalid topic name** (see below). 
Could not find out why this happened.

I also noticed that the turtlebot4_base_node was not subscribing to the /cmd_vel topic which is probably related to the bringup problems.

### Error messages

```bash
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2025-02-19-17-42-58-770100-ubuntu-1762
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [turtlebot4_node-1]: process started with pid [1763]
[INFO] [turtlebot4_base_node-2]: process started with pid [1765]
[INFO] [create3_republisher-3]: process started with pid [1767]
[create3_republisher-3] [INFO] [1739986979.112038981] [create3_repub]: Creating republisher node with namespace '/' to interact with robot '//_do_not_use'
[create3_republisher-3] [INFO] [1739986979.114429418] [create3_repub]: Found 8 entities for robot_publishers
[create3_republisher-3] [INFO] [1739986979.118058155] [create3_repub]: Subscribing to topic '//_do_not_use/battery_state' and republishing it as '/battery_state' with type 'sensor_msgs/msg/BatteryState'
[create3_republisher-3] terminate called after throwing an instance of 'rclcpp::exceptions::InvalidTopicNameError'
[create3_republisher-3]   what():  Invalid topic name: topic name must not contain repeated '/':
[create3_republisher-3]   '//_do_not_use/battery_state'
[create3_republisher-3]     ^
...
```

### To Reproduce

1. Set up the network to [discovery server](https://turtlebot.github.io/turtlebot4-user-manual/setup/discovery_server.html)
2. Connect the controller by following the [basic setup](https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html)
3. Drive the robot with L1 and Joystick

### Other notes

Could someone please help me fix these problems. Would be very appreciated! 
Feel free to ask further questions to help pinpoint the cause

slowrunner, 2025-02-21:
> [create3_republisher-3]   what():  Invalid topic name: topic name must not contain repeated '/':
> [create3_republisher-3]   '//_do_not_use/battery_state'
> [create3_republisher-3]     ^

You have an extra slash in front of the do_not_use robot namespace.  Should be like this namespace:

[image]

Fix it, click Save, then click Restart Application link at the bottom 

(Browse to your <Raspberry pi IP>:8080 then click App Configuration)

Then on the RPi. Do turtlebot4-service-restart

YogiMind, 2025-02-25:
Thank you @smatarCPR for your help!
I added a namespace and after restarting the robot I can now move the robot with the controller!

Thanks again!

smatarCPR, 2025-02-28:
Hello @YogiMind ,
Glad I could help. I will go ahead and close this issue. Should you encounter any follow up issues, do not hesitate to reach out by simply replying to this thread or opening up a new issue.
Best Regards,
Saif

closed 2025-02-28 by smatarCPR