turtlebot/turtlebot4 #557: Navigation Issue

parvsytl, 2025-02-24:
> Hello Saif,
> 
> I hope you are doing well.
> 
> I recently followed the steps recommended by the TurtleBot team to configure NTP synchronization between my TurtleBot4 and Create3 base. After setting up the Raspberry Pi as the NTP server and configuring the Create3 to sync with it, everything worked perfectly at home. The robot was able to perform autonomous navigation without issues.
> 
> However, when I attempted to replicate the same setup at my office the next day, the autonomous navigation did not work. The only difference between the two setups is the network:
> 
>     Home: Home WiFi, Robot1
>     Office: Office WiFi, Robot2, Robot3
>     PC: The same in both locations
> 
> Since the setup worked at home, I assumed it was a network-related issue at the office. However, when I returned home and tried again, the same issue occurred—navigation was no longer functioning, even though I did not make any changes to the home robot or Create3 base.
> 
> At the moment, after performing all the recommended steps again at the office, the issue persists. Chrony and NTP appear to be running, but the robot does not navigate autonomously as expected.
> 
> I am attaching screenshots for your reference. Could you please help me identify the exact reason for this issue?
> 
> Looking forward to your assistance.
> 
> Regards,
> 
> Parvathy Santhosh
> 
> [image]
> 
> [image]
> 
> [image]
> 
> [image] 

 _Originally posted by @parvsytl in [#530](https://github.com/turtlebot/turtlebot4/issues/530#issuecomment-2662447317)_

slowrunner, 2025-02-24:
Out of curiosity, have you tried bringing up nav with the bot starting in a “white” area?  The area in the rviz2 screenshot looks very high cost.

smatarCPR, 2025-03-04:
Hello @parvsytl ,
Thank you for reaching out on the Turtlebot 4 Github page. I am very sorry to hear you are having issues with your Turtlebot. Based on the details provided I have the following advice:

When running RVIZ and setting the initial position on the remote computer, there is a chance that due to time synchronization that Nav2 is unable to process the initial position.  To combat this, setup NTP time synchronization on the remote computer you are using. Since in normal circumstances the Raspberry Pi is already setup for NTP through the internet you can setup your computer to do the same - https://documentation.ubuntu.com/server/how-to/networking/serve-ntp-with-chrony/index.html

Alternatively you can have your computer sync its time with the Raspberry Pi rather than a global ntp server.

Let me know if you have any questions or if I can help with anything else.
Best Regards,
Saif

closed 2025-04-22 by smatarCPR