turtlebot/turtlebot4 #559: Create3 Republisher Stops Republishing

slowrunner, 2025-02-25:
### Robot Model

Turtlebot4 Lite

### ROS distro

Jazzy

### Networking Configuration

Discovery Server

### OS

Ubuntu 24.04

### Built from source or installed?

Built from Source

### Package version

turtlebot4/turtlebot4  [dc744fe](https://github.com/turtlebot/turtlebot4/commit/dc744fee51ee681ae9092fda521e98a4a4c23330)
turtlebot4/turtlebot4_robot  [28e115f](https://github.com/turtlebot/turtlebot4_robot/commit/28e115f86930bb3273f1a1aa4738dbd8747513f0)

turtlebot4-setup:About
```
TurtleBot 4 Open Source Robotics Platform.  

Model: lite  
Version: 2.0.2  
ROS: Jazzy  
Hostname: TB5WaLI  
IP: 10.0.0.178  
The TurtleBot 4 was created in a partnership between Open Robotics and Clearpath Robotics.  

```

Ubuntu 24.04
```
PRETTY_NAME="Ubuntu 24.04.2 LTS"
NAME="Ubuntu"
VERSION_ID="24.04"
VERSION="24.04.2 LTS (Noble Numbat)"
VERSION_CODENAME=noble
ID=ubuntu
ID_LIKE=debian

```
ROS 2 Jazzy
```
ROS_VERSION=2
ROS_PYTHON_VERSION=3
ROS_SUPER_CLIENT=True
ROS_DOMAIN_ID=5
ROS_AUTOMATIC_DISCOVERY_RANGE=SUBNET
ROS_DISTRO=jazzy
ROS_DISCOVERY_SERVER=127.0.0.1:11811;

```
Raspberry Pi 5 8GB
```
Linux TB5WaLI 6.8.0-1018-raspi #20-Ubuntu SMP PREEMPT_DYNAMIC Fri Jan 17 12:35:36 UTC 2025 aarch64 aarch64 aarch64 GNU/Linux
```

NTP Setup:  
- RPi is Discovery Server  
- Create3 is client  

```
sudo chronyc clients
Hostname                      NTP   Drop Int IntL Last     Cmd   Drop Int  Last
===============================================================================
192.168.186.2                   4      0  11   -  1116       0      0   -     -
```

### Type of issue

Other

### Expected behaviour

# Expect Create3 republished topics to always be available.

### Actual behaviour

### Create3 topics /battery_state and /dock_status stopped being published roughly 10 hours after full system boot. 

This is the second morning I have found the bot in this "not publishing Create3 topics" between 5 and 6AM after running this script overnight.  ([Prior time documented here Issue #554 ](https://github.com/turtlebot/turtlebot4/issues/554) )

### Error messages

```bash
No error messages found - Create3 topics stop being "published"
```

### To Reproduce

What I did with system:
1) remove bot from dock
2) sudo shutdown -h now on RPi
3) Hold Create3 Power button till shutdown tone (7 seconds)
4) Place bot on dock to startup bot
5) launched localization 
6) launched navigation (no global cost map), ctrl-c 
7) launched navigation (no global cost map), ctrl-c 
8) launched navigation - success - see global cost map
9) closed rviz2, ctrl-c navigation, ctrl-c localization
10) launched my monitor.sh - It died around 5:33 AM after "up 10:30"
```
Collecting ROS Info

********** TB5-WaLI MONITOR ******************************
Tuesday 02/25/25
 05:33:14 up 10:13,  3 users,  load average: 1.87, 1.44, 1.18
temp=64.8'C
frequency(0)=2100018048
throttled=0x0
               total        used        free      shared  buff/cache   available
Mem:           7.8Gi       990Mi       6.0Gi        14Mi       1.0Gi       6.8Gi
Swap:             0B          0B          0B

Voltage: 16.60  Current: 0.610  Watts: 10.12
Battery: 0.97  Docked: True
---
 
Collecting ROS Info                   <<<<---- STUCK HERE - /battery_state and /dock_status no longer published

```

### Other notes

## monitor.sh

```
#!/bin/bash

while true; \
do echo -e "Collecting ROS Info"; \

# Check Battery Percentage
batt=$(ros2 topic echo --once --field percentage --qos-reliability best_effort --qos-durability volatile /battery_state)
batt="${batt:0:4}"

# Checking Docking State
docked=$(ros2 topic echo --once --field is_docked --qos-reliability best_effort --qos-durability volatile /dock_status)

curr=$(ros2 topic echo --once --field current --qos-reliability best_effort --qos-durability volatile /battery_state); \
curr="${curr:0:5}"; \
# echo "Current: $curr"; \

volt=$(ros2 topic echo --once --field voltage --qos-reliability best_effort --qos-durability volatile /battery_state); \
volt="${volt:0:5}"; \
watts=`(echo "scale=1; ($volt * $curr)" | bc -l)`; \
watts="${watts:0:5}"; \

echo -e "\n********** TB5-WaLI MONITOR ******************************"; \

echo -n `date +"%A %D"`; \
echo ""; \
uptime; \
vcgencmd measure_temp && vcgencmd measure_clock arm && vcgencmd get_throttled; \
free -h; \
echo ""; \
echo "Voltage: $volt  Current: $curr  Watts: $watts"; \
echo "Battery: $batt  Docked: $docked"; \

# ps -e -o %cpu,%mem,cmd | grep  'CPU\|MEM\|CMD\|rplidar\|robot\|joint\|oak\|nav\|slam\|turtlebot4\|wali' | grep -v "grep" | sort -rk 
1 ; \
# sleep 5; \
echo " "; \
done

```

## node list and topic list in "stopped publishing" state
```
ubuntu@TB5WaLI:~/TB5-WaLI/wali_ws$ ros2 node list
/create3_repub
/joint_state_publisher
/joy_linux_node
/launch_ros_1377
/oakd
/oakd_container
/odometer
/robot_state_publisher
/rplidar_composition
/say
/teleop_twist_joy_node
/turtlebot4_node
/wali_node

ubuntu@TB5WaLI:~/TB5-WaLI/wali_ws$ ros2 topic list
/battery_state
/cmd_audio
/cmd_lightring
/cmd_vel
/cmd_vel_unstamped
/diagnostics
/dock_status
/function_calls
/imu
/interface_buttons
/ip
/joint_states
/joy
/joy/set_feedback
/oakd/rgb/preview/camera_info
/oakd/rgb/preview/image_raw
/oakd/rgb/preview/image_raw/compressed
/oakd/rgb/preview/image_raw/compressedDepth
/oakd/rgb/preview/image_raw/theora
/oakd/rgb/preview/image_raw/zstd
/odom
/parameter_events
/robot_description
/rosout
/scan
/stop_status
/tf
/tf_static
/wheel_status

```

## Current time stamp
```
ubuntu@TB5WaLI:~/TB5-WaLI/wali_ws$ uptime; ros2 topic echo --once /joy
 09:42:05 up 14:22,  3 users,  load average: 0.41, 0.26, 0.22
header:
  stamp:
    sec: 1740494528
    nanosec: 270926560
  frame_id: joy

```

## BUT /battery_state (and /dock_status) are no longer being updated:
```
ubuntu@TB5WaLI:~/TB5-WaLI/wali_ws$ uptime; ros2 topic echo --once /battery_state
 09:42:43 up 14:23,  3 users,  load average: 0.32, 0.26, 0.22
header:
  stamp:
    sec: 1740479599            <<<<--- NOTE timestamp is 4.14 hours old  (around 5:33 AM)
    nanosec: 253609684
  frame_id: ''
voltage: 16.61400032043457
temperature: 38.64999771118164
current: 0.6060000061988831
charge: 1.9079999923706055
capacity: 1.9709999561309814
design_capacity: 1.9709999561309814
percentage: 0.9700000286102295
power_supply_status: 0
power_supply_health: 0
power_supply_technology: 0
present: true
cell_voltage: []
cell_temperature: []
location: ''
serial_number: ''
---

slowrunner, 2025-02-26:
Again republisher is in weird state - after 8h 57m the node list no longer shows the /_do_not_use/ nodes.  This time I did not run localization or navigation or even rviz2 visualization.  

Only my wali_node subscribed to /battery_state  and /dock_status, and my monitor script running echo --once repeatedly on those two topics roughly every 20 seconds.

Why should that kill the Create3 republisher containerized nodes?

hilary-luo, 2025-03-03:
On your robot, can you please edit /etc/systemd/logind.conf to uncomment the `RemoveIPC` line and set it as `RemoveIPC=no`, then reboot the robot. See if the issue is still reproducible after that change.

A functionally equivalent change is going to be rolled out for all users shortly. Your symptoms sound similar enough to another issue that they may be related although your situation is presenting a bit differently. After the official update gets rolled out you wouldn't need that modification in the /etc/systemd/logind.conf file anymore.

hilary-luo, 2025-03-03:
If the issue is still reproducible after that change then please provide both the create3 log and the log from the pi (journalctl) from the boot where it happened.

slowrunner, 2025-03-03:
@hilary-luo 
# New Twist: Republisher not working at all with suggested mod: ```RemoveIPC=no``` in logind.conf

But my Raspberry Pi claims "becoming mindful." so maybe it will heal itself!
```
Mar 03 14:31:15 TB5WaLI kernel: Yama: becoming mindful.
```
... Getting serious, sorry ...

### Indications of new problem:
- Node list does not show containerized /_do_not_use/ nodes:
```
ubuntu@TB5WaLI:~/TB5-WaLI/wali_ws$ ros2 node list
/create3_repub                 <--- running but no /_do_not_use nodes listed
/joint_state_publisher
/joy_linux_node
/launch_ros_2309
/oakd
/oakd_container
/odometer
/robot_state_publisher
/rplidar_composition
/say
/teleop_twist_joy_node
/turtlebot4_node
/wali_node
```
- Republisher has registered publishers:
```
ubuntu@TB5WaLI:~/TB5-WaLI/issues/create3stops_pub$ ros2 topic list
/battery_state
/cmd_audio
/cmd_lightring
/cmd_vel
/cmd_vel_unstamped
/diagnostics
/dock_status
/function_calls
/imu
/interface_buttons
/ip
/joint_states
/joy
/joy/set_feedback
/oakd/rgb/preview/camera_info
/oakd/rgb/preview/image_raw
/oakd/rgb/preview/image_raw/compressed
/oakd/rgb/preview/image_raw/compressedDepth
/oakd/rgb/preview/image_raw/theora
/oakd/rgb/preview/image_raw/zstd
/odom
/parameter_events
/robot_description
/rosout
/scan
/stop_status
/tf
/tf_static
/wheel_status
```

- turtlebot4.service status
```
ubuntu@TB5WaLI:~/TB5-WaLI/wali_ws$ systemctl status turtlebot4.service 
● turtlebot4.service - "bringup turtlebot4"
     Loaded: loaded (/usr/lib/systemd/system/turtlebot4.service; enabled; preset: enabled)
     Active: active (running) since Mon 2025-03-03 14:34:41 EST; 58min ago
   Main PID: 2211 (turtlebot4-star)
      Tasks: 147 (limit: 9375)
     Memory: 282.8M (peak: 303.7M)
        CPU: 5min 19.904s
     CGroup: /system.slice/turtlebot4.service
             ├─2211 /bin/bash /usr/sbin/turtlebot4-start
             ├─2309 /usr/bin/python3 /opt/ros/jazzy/bin/ros2 launch /tmp/turtlebot4.launch.py
             ├─2313 /home/ubuntu/TB5-WaLI/wali_ws/install/turtlebot4_node/lib/turtlebot4_node/turtlebot4_node --ros-args -r __ns:=/ --params-file /tmp/tmp0iqkq5_3 --params-file /tmp/launch_params_5cxpj0ss
             ├─2314 /opt/ros/jazzy/lib/create3_republisher/create3_republisher --ros-args -r __ns:=/ --params-file /home/ubuntu/TB5-WaLI/wali_ws/install/turtlebot4_bringup/share/turtlebot4_bringup/config/republisher.yaml --params-file /tmp/launc>
             ├─2315 /opt/ros/jazzy/lib/joy_linux/joy_linux_node --ros-args -r __node:=joy_linux_node -r __ns:=/ --params-file /tmp/launch_params_089df3yt -r /diagnostics:=diagnostics
             ├─2316 /opt/ros/jazzy/lib/teleop_twist_joy/teleop_node --ros-args -r __node:=teleop_twist_joy_node -r __ns:=/ --params-file /tmp/tmpjqf5pvll --params-file /tmp/launch_params_f1ihpbxz
             ├─2317 /home/ubuntu/TB5-WaLI/wali_ws/install/sllidar_ros2/lib/sllidar_ros2/sllidar_node --ros-args -r __node:=rplidar_composition -r __ns:=/ --params-file /tmp/launch_params_4j3uylh6
             ├─2318 /opt/ros/jazzy/lib/robot_state_publisher/robot_state_publisher --ros-args -r __node:=robot_state_publisher -r __ns:=/ --params-file /tmp/launch_params_2aju_80y --params-file /tmp/launch_params_i2t5bgtj -r /tf:=tf -r /tf_stati>
             ├─2319 /usr/bin/python3 /opt/ros/jazzy/lib/joint_state_publisher/joint_state_publisher --ros-args -r __node:=joint_state_publisher -r __ns:=/ --params-file /tmp/launch_params_3wpzq3j4 -r /tf:=tf -r /tf_static:=tf_static
             └─2452 /opt/ros/jazzy/lib/rclcpp_components/component_container --ros-args -r __node:=oakd_container -r __ns:=/

Mar 03 14:35:15 TB5WaLI turtlebot4-start[2309]: [INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/oakd' in container '/oakd_container'
Mar 03 14:35:16 TB5WaLI turtlebot4-start[2309]: [component_container-8] [INFO] [1741030516.803731801] [oakd]: Starting camera.
Mar 03 14:35:16 TB5WaLI turtlebot4-start[2309]: [component_container-8] [INFO] [1741030516.814669055] [oakd]: No ip/mxid specified, connecting to the next available device.
Mar 03 14:35:19 TB5WaLI turtlebot4-start[2309]: [component_container-8] [INFO] [1741030519.313306352] [oakd]: Camera with MXID: 184430101175A41200 and Name: 3.1 connected!
Mar 03 14:35:19 TB5WaLI turtlebot4-start[2309]: [component_container-8] [INFO] [1741030519.314161816] [oakd]: USB SPEED: HIGH
Mar 03 14:35:19 TB5WaLI turtlebot4-start[2309]: [component_container-8] [INFO] [1741030519.335468915] [oakd]: Device type: OAK-D-LITE
Mar 03 14:35:19 TB5WaLI turtlebot4-start[2309]: [component_container-8] [INFO] [1741030519.337768455] [oakd]: Pipeline type: RGB
Mar 03 14:35:20 TB5WaLI turtlebot4-start[2309]: [component_container-8] [WARN] [1741030520.448933124] [oakd]: IMU enabled but not available!
Mar 03 14:35:20 TB5WaLI turtlebot4-start[2309]: [component_container-8] [INFO] [1741030520.449115106] [oakd]: Finished setting up pipeline.
Mar 03 14:35:20 TB5WaLI turtlebot4-start[2309]: [component_container-8] [INFO] [1741030520.667592976] [oakd]: Camera ready!
lines 1-29/29 (END)
```
[2025-03-03_create3.log](https://github.com/user-attachments/files/19060281/2025-03-03_create3.log)

[2025-03-03_RPi.log](https://github.com/user-attachments/files/19060289/2025-03-03_RPi.log)

hilary-luo, 2025-03-03:
There is no path where making that change would makes sense to cause the issue you describe since that change only impacts disabling a particular action from happening when all user sessions are logged out. So it is much more likely that the issue you are seeing is independent and happened when you tested this unrelated fix.

In your create3 log there are a bunch of errors about Network unreachable. How is your create3 connected to the pi and is the create3 also connected to the wifi? I think the issue is with the connectivity to the create3 in this case.

slowrunner, 2025-03-04:
I'll gave it another try with only rebooting the RPi, and repub is working correctly - I started the "overnight monitor.sh" test:
```

ubuntu@TB5WaLI:~/TB5-WaLI/wali_ws$ ros2 node list
/_do_not_use/_internal/composite_hazard
/_do_not_use/_internal/kinematics_engine
/_do_not_use/_internal/mobility
/_do_not_use/_internal/stasis
/_do_not_use/mobility_monitor
/_do_not_use/motion_control
/_do_not_use/robot_state
/_do_not_use/static_transform
/_do_not_use/system_health
/_do_not_use/ui_mgr
/create3_repub
/joint_state_publisher
/joy_linux_node
/launch_ros_1367
/oakd
/oakd_container
/odometer
/robot_state_publisher
/say
/teleop_twist_joy_node
/turtlebot4_node
/wali_node

ubuntu@TB5WaLI:~/TB5-WaLI/wali_ws$ ./monitor.sh 
Collecting ROS Info

********** TB5-WaLI MONITOR ******************************
Monday 03/03/25
 21:32:36 up 4 min,  2 users,  load average: 0.26, 0.42, 0.22
temp=59.8'C
frequency(0)=1900019584
throttled=0x0
               total        used        free      shared  buff/cache   available
Mem:           7.8Gi       875Mi       6.6Gi        12Mi       529Mi       6.9Gi
Swap:             0B          0B          0B

Voltage: 14.63  Current: -0.78  Watts: -11.4
Battery: 0.57  Docked: False
---
 
Collecting ROS Info

```

### In the earlier test I brought the system up from a cold and dark start:
- Ensure bot not docked
- RPi shutdown -h now
- Create3 ctr btn till descending notes
- Place bot on dock, 
  - Create3 starts, 
  - RPi boots, starts chrony, turtlebot4.service, then my WaLI nodes 
  - Create3 announces alive eventually.

> How is your create3 connected to the pi and is the create3 also connected to the wifi?

The Create3 is only connected to the Pi via Ethernet over the USB - RPi: 192.168.182.3, Create3: 192.168.182.2.
The Create3 is not connected to WiFi.

```
>>>>>>> system IP address
unknown

>>>>>>> system version
I.0.0.FastDDS

>>>>>>> system sku
RCi3099

>>>>>>> system battery
Percentage:66
Capacity(mAh):1944

>>>>>>> system properties
   PRODUCT         : create3
   SERIAL          : e17628
   CLEAN MODE      : normal
   CONNS SVC MODE  : normal
   AP RADIO        : wlan1/g/6/[SHORT-GI-20][GF]/Create-883C/US
   AP IP ADDR      : 192.168.10.1
   CL IP           : dhcp/#/#/#
   CL DNS          : #/#/#
   CL DHCPCD T/O   : 3
   CL REG DOMAIN   : #
   ETH0 IP         : static/192.168.186.2/192.168.186.3
   USB0 IP         : 192.168.186.2/192.168.187.2/192.168.186.3
   SYSTEM_ACCESS   : locked
   DEVELOPMENT MODE: disabled
   RADIO_LIST      : wlan0 wlan1
   HOSTNAME        : iRobot-883C456BE1E840DC89E64B28B4F75B53
   ROBOT ID        : 883C456BE1E840DC89E64B28B4F75B53
   BOOT COUNT      : 52
   LANGUAGE PACK   : unknown
   EXTRA LANGUAGE  : not_installed

>>>>>>> system date
Tue Mar  4 02:22:02 UTC 2025

>>>>>>> system uptime
 02:22:03 up  5:41,  load average: 1.32, 1.13, 1.06

>>>>>>> system memory
              total        used        free      shared  buff/cache   available
Mem:          59088       33844        1416         252       23828       23296
Swap:             0           0           0

>>>>>>> system ifconfig
lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:1115 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1115 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:60402 (58.9 KiB)  TX bytes:60402 (58.9 KiB)

usb0      Link encap:Ethernet  HWaddr FE:90:2E:78:0D:AD  
          inet addr:192.168.186.2  Bcast:192.168.186.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:12282 errors:0 dropped:0 overruns:0 frame:0
          TX packets:2212959 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:5729507 (5.4 MiB)  TX bytes:1029187373 (981.5 MiB)

>>>>>>> system hciconfig

[Factory Reset](javascript:confirm_factory_reset())
```

closed 2026-04-07 by RustyCPR