turtlebot/turtlebot4 #562: ros topics aren't published on startup

andpuc23, 2025-02-28:
### Robot Model

Turtlebot4 Standard

### ROS distro

Jazzy

### Networking Configuration

Simple Discovery

### OS

Ubuntu 24.04

### Built from source or installed?

Installed

### Package version

ros-jazzy-turtlebot4-base 2.0.1-2noble.20241228.002030

### Type of issue

Networking

### Expected behaviour

- I start the robot
- it loads
- makes the "happy" sound
- all 5 LEDs are on
- I see the data being published on topics like /battery_state

### Actual behaviour

- Robot starts
- makes the sound
- COMM and BATT LEDs are off
- screen shows 0 battery state
- topics published are /parameter_events and /rosout

### Error messages

```bash

```

### To Reproduce

Not sure how to reproduce this as this appeared at some moment without my intervention
Start the robot?

### Other notes

I've seen issues #269, #507, didn't help. Tried forgetting the network, switching ROS_DOMAIN_ID to 1 on robot and pc - no luck.

Tried running diagnostics, see the error log, not quite sure what this means:

```
ubuntu@turtlebot4:~$ ros2 launch turtlebot4_diagnostics diagnostics.launch.py
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2025-02-28-07-41-04-198398-turtlebot4-1851
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): launch configuration 'namespace' does not exist
```

slowrunner, 2025-02-28:
You should add screenshots of the Configuration page and Home page, or very, very accurately list the namespace, RMW, and Enable FastDDS check box, and the Firmware version shown in your Home or About page.

Also you probably should list your turtlebot4-setup -> ROS Setup -> Bash Setup values

slowrunner, 2025-03-01:
diagnostics are started automatically if you selected enable diagnostics in turtlebot4-setup.

If you did not enable them, they can be started with:
```
ros2 launch turtlebot4_diagnostics diagnostics.launch.py namespace:=/
```

then on your visualization platform:
```
ros2 launch turtlebot4_viz view_diagnostics.launch.py
```
(You can ignore the Create3 Mouse Error - republisher does not repub mouse topic by default)

[image]

andpuc23, 2025-03-01:
@slowrunner thank you for your response!

Here is my application config: 
[image]
[image]

Not quite sure what should be shown at home page, but here it is:
[image]

I can confirm that the diagnostics are enabled

[image]
But if i run `ros2 topic list` at robot it doesn't show them:
```
ubuntu@turtlebot4:~$ ros2 topic list
/parameter_events
/rosout
```

Tried running your command for starting the diagnostics tool - they start indeed, thanks for that, here's what I see at my pc:

[image]

I find it rather suspicious that there is no /battery_state published, even though robot is docked, and (if undocked) the ring lights show the percentage somewhat accurately (it loses some part when I leave it undocked for an hour or 2)

Thanks,
andpuc23

slowrunner, 2025-03-01:
@andpuc23 Your turtlebot4-setup ROS_DOMAIN_ID does not match what you are showing in the Create3 Configuration page.

I'm not totally good with remembering the configuration steps so suggest a re-read of the manual!

My guess:
- Set turtlebot4-setup -> ROS Setup -> Bash Setup: change ROS_DOMAIN_ID to 1
  - Save
- Robot Upstart -> Install
? Turtlebot4 Setup: Apply Settings ?
- Exit
- sudo shutdown -h now
- Remove bot from dock
- Press Create3 center button till descending notes
- Place back on dock
- Wait forever for second rising notes
- remote ssh back into the RPi 
  - ```printenv | grep ROS```          (see ROS_DOMAIN_ID is what you wanted)
  - ```ros2 topic echo /battery_state```

closed 2025-03-03 by andpuc23