turtlebot/turtlebot4 #563: Failed to activate Controller : diffdrive_controller

mohitydv09, 2025-03-03:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

I do not know

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

1.0.5-1jammy.20250121.134002

### Type of issue

Navigation (SLAM, Nav2 etc.)

### Expected behaviour

Robot Undock

### Actual behaviour

When running :
`ros2 launch turtlebot4_ignition_bringup turtlebot4_ignition.launch.py`

The simulated robot won't move from its place; terminal shows `diffdrive_controller` error.

### Error messages

```bash
[ruby $(which ign) gazebo-1] [ERROR] [1740970431.267199346] [controller_manager]: Switch controller timed out after 5.000000 seconds!
[spawner-32] [ERROR] [1740970431.270110117] [spawner_joint_state_broadcaster]: Failed to activate controller : joint_state_broadcaster
[ERROR] [spawner-32]: process has died [pid 33669, exit code 1, cmd '/opt/ros/humble/lib/controller_manager/spawner joint_state_broadcaster -c controller_manager --ros-args -r __ns:=/'].
[INFO] [spawner-46]: process started with pid [34385]
[spawner-46] [INFO] [1740970432.219270380] [spawner_diffdrive_controller]: waiting for service /controller_manager/list_controllers to become available...
[ruby $(which ign) gazebo-1] [INFO] [1740970434.992716695] [controller_manager]: Loading controller 'diffdrive_controller'
[spawner-46] [INFO] [1740970435.125048749] [spawner_diffdrive_controller]: Loaded diffdrive_controller
[ruby $(which ign) gazebo-1] [INFO] [1740970435.139157273] [controller_manager]: Configuring controller 'diffdrive_controller'
[ruby $(which ign) gazebo-1] [ERROR] [1740970440.170862487] [controller_manager]: Switch controller timed out after 5.000000 seconds!
[spawner-46] [ERROR] [1740970440.174629033] [spawner_diffdrive_controller]: Failed to activate controller : diffdrive_controller
[ERROR] [spawner-46]: process has died [pid 34385, exit code 1, cmd '/opt/ros/humble/lib/controller_manager/spawner diffdrive_controller -c controller_manager --ros-args -r __ns:=/ --params-file /opt/ros/humble/share/irobot_create_control/config/control.yaml'].
[parameter_bridge-9] [INFO] [1740970440.726227088] [cmd_vel_bridge]: Passing message from ROS geometry_msgs/msg/Twist to Gazebo gz.msgs.Twist (showing msg only once per type)
[ruby $(which ign) gazebo-1] [WARN] [1740970440.726342223] [diffdrive_controller]: Can't accept new commands. subscriber is inactive
[turtlebot4_node-30] [INFO] [1740970440.825815008] [turtlebot4_node]: OAKD stopped
[turtlebot4_node-30] [INFO] [1740970440.825902920] [turtlebot4_node]: RPLIDAR stopped
[ruby $(which ign) gazebo-1] [WARN] [1740970440.826291201] [diffdrive_controller]: Can't accept new commands. subscriber is inactive
[ruby $(which ign) gazebo-1] [WARN] [1740970440.968934786] [diffdrive_controller]: Can't accept new commands. subscriber is inactive
[ruby $(which ign) gazebo-1] [WARN] [1740970441.116520041] [diffdrive_controller]: Can't accept new commands. subscriber is inactive
[ruby $(which ign) gazebo-1] [WARN] [1740970441.233784494] [diffdrive_controller]: Can't accept new commands. subscriber is inactive
[ruby $(which ign) gazebo-1] [WARN] [1740970441.379537266] [diffdrive_controller]: Can't accept new commands. subscriber is inactive
[ruby $(which ign) gazebo-1] [WARN] [1740970441.553315848] [diffdrive_controller]: Can't accept new commands. subscriber is inactive
[ruby $(which ign) gazebo-1] [WARN] [1740970441.701348672] [diffdrive_controller]: Can't accept new commands. subscriber is inactive
[ruby $(which ign) gazebo-1] [WARN] [1740970441.732051920] [diffdrive_controller]: Can't accept new commands. subscriber is inactive
[ruby $(which ign) gazebo-1] [WARN] [1740970441.897931871] [diffdrive_controller]: Can't accept new commands. subscriber is inactive
```

### To Reproduce

run ros2 launch turtlebot4_ignition_bringup turtlebot4_ignition.launch.py

Congying1112, 2025-03-12:
exactly the same!!

Congying1112, 2025-03-12:
I resolved it by building from source

smatarCPR, 2025-03-17:
Hello @mohitydv09 ,
Thank you for reaching out on the Turtlebot 4 Github page. I am very sorry to hear that you are having issues with the simulator. Would it be possible to provide/check the following:

- In the bottom left hand corner of the Gazebo window there should be a play button. Have you pressed this button. Would it be possible to provide screenshot of the Gazebo simulator following its launch?
- What sort of setup are you using to run the simulation, is it a virtual machine or are you running Ubuntu natively?

Let me know the results of these checks. Looking forward to hearing from you soon.
Best Regards,
Saif

closed 2025-04-14 by smatarCPR