turtlebot/turtlebot4 #565: no connection between create3 and raspberry pi

IacopoFornai, 2025-03-04:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

I do not know

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

i cannot provide the package version since i cannot connect to raspberry pi

### Type of issue

Other

### Expected behaviour

establish the connection between create3 and raspberry pi

### Actual behaviour

i was updating the turtlebot4 standard to ros2 jazzy version. 
 Cannot open the webserver Create3 , i was able to update Create3 only (using Create3 as Access Point). No connection between Create3 and Raspberry Pi.

### Error messages

```bash

```

### To Reproduce

turn on the turtlebot4 by putting it on the charging station

IacopoFornai, 2025-03-26:
Hi @smatarCPR 

connecting with ethernet cable, i'm able to ssh into the raspberry, i already did a factory reset of the raspberry pi and flashed the latest Raspberry pi image, i also already installed ros2 jazzy. 

at startup the turtlebot just shows this ip: 10.42.0.1
moreover only 3 leds are on: mtr, wifi, power. comm and batt leds are off
and the lidar is always rotating without stopping

Looking forward to hearing from you soon.
best regards,

smatarCPR, 2025-04-23:
Hello @IacopoFornai 
Apologies for the delay in response. I have the following questions regarding the current state of the Turtlebot:

- After installing Jazzy, were you able to follow the basic setup process to get the Turtlebot 4 connected to wi-fi and setup the networking configuration - https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html#robot
- Did you update the firmware for the Create3? The recommended version of the firmware is the latest Iron version - https://iroboteducation.github.io/create3_docs/releases/overview/#iron
- The lidar spinning constantly is normal, so long the /scan topic exists and being published, the Lidar will spin to provide scan data.

Looking forward to hearing from you soon.
Best Regards,
Saif

IacopoFornai, 2025-04-30:
hello @smatarCPR i followed the instructions from scratch and now the turtlebot is working again, probably there was an error in the configuration, thanks for your help!

closed 2025-04-30 by IacopoFornai