turtlebot/turtlebot4 #567: Problem Report: TurtleBot 4 unresponsive screen and lidar after unboxing

TechmakeS, 2025-03-05:
### Robot Model

Turtlebot4 Standard

### ROS distro

Select One

### Networking Configuration

Select One

### OS

Select One

### Built from source or installed?

Select One

### Package version

None because I can't it works

### Expected behaviour

None because I can't operate TurtleBot 4 not responding screen and lidar after unpacking

### Actual behaviour

There is a reported problem with the TurtleBot 4 we purchased. The unit was
unpacked (Figures 1 and 2) (received with scratches (Figure 3)) and attached to its base; however, it is not
working as expected.

### Error messages

```bash

```

### To Reproduce

None because I can't operate TurtleBot 4 not responding screen and lidar after unpacking.

### Other notes

[Informe de ProblemaTurtleBot 4 no responde pantalla y lidar despues del desempaquetado.pdf](https://github.com/user-attachments/files/19098365/Informe.de.ProblemaTurtleBot.4.no.responde.pantalla.y.lidar.despues.del.desempaquetado.pdf)

hilary-luo, 2025-03-06:
@TechmakeS You have submitted a software bug report. In the future please read through the options and select a `Troubleshooting` issue if you do not have a specific error to report in this particular software package. In this case I will reassign it for you.

RustyCPR, 2025-03-10:
Hi @TechmakeS 
Thanks for writing in and sorry to see your having issues with the TB4. 

Some items to verify:
1. Make sure the Create3 power distribution board is fully inserted into the Create3 base, this will power the RPi, PCBa, lidar and camera. 
https://github.com/turtlebot/turtlebot4/issues/394#issuecomment-2045931389

2. When booting does the Create3 play a chime?
-What is the color ring light on the Create3 showing: https://iroboteducation.github.io/create3_docs/hw/face/

3. Please verify the other connections between the Create3 Power Distribution board and the TB4 PCBA and RPi: 
https://github.com/turtlebot/turtlebot4/issues/537#issuecomment-2638012695

Thank you,
Rusty

TechmakeS, 2025-03-14:
[image]

[Informe de ProblemaTurtleBot 4 no responde pantalla.docx](https://github.com/user-attachments/files/19251428/Informe.de.ProblemaTurtleBot.4.no.responde.pantalla.docx)

RustyCPR, 2025-03-17:
**_### Translation from Informe de ProblemaTurtleBot 4 no responde pantalla.docx_**

Subject: Issue Report: TurtleBot 4 still not responding to screen
Reporting an issue with the TurtleBot 4 we purchased. The recommended tests were performed https://github.com/turtlebot/turtlebot4/issues/537#issuecomment-2638012695. 

All connections were checked:
All connections were verified and connected correctly:
• The USB/BT toggle switch is set to USB
• The Battery power output cable and USB are securely connected.
- Check the other ends of the above connections are securely connected.
- The RPi ribbon cable is securely connected to the top PCBA.
After this, it was reconnected to its base:
And the following is shown: At startup, the creator starts in red, as shown in the figure: Robot error. Cycle power. https://iroboteducation.github.io/create3_docs/hw/face/ in the documentation it indicates the cycle power error but I didn't find what procedure should be followed in this case.?

[image]
After a while the color changed and a sound was heard.

[image]

The LIDAR signal is now being received, and a connection was established via SSH following the instructions in the same link https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html#recovering-the-raspberry-pi. However, the display still doesn't receive a signal, even though it's off.

RustyCPR, 2025-03-17:
Hi @TechmakeS 

Some questions to help me understand the issue:
1. Are some of the PCBA LEDs on? 
[image]

>If yes - PCBA LEDs are on - check the LCD ribbon cable is fully inserted:

[image]

>If no - LEDs are not on check the power cable from the Create 3 power distribution board. 

[image]

TechmakeS, 2025-03-31:
Hi,

None of the LEDs on the PCBA are lit.

[image]

After this, the power supply cable for the Create 3 and PCB A was checked. As can be seen in the images, they are connected correctly.

[image]

[image]

[image]

It was tested again and still no LEDs turn on and the screen remains off.

TechmakeS, 2025-04-02:
HI, I still haven't been able to resolve the issue with Turtlebot. Is there a way to schedule a meeting with technical support, us and the user?

Best Regards,
Lea Salazar
Departamento de Compras

[signature_1392675457]

Techmake Solutions, S.A. de C.V.
-   81 2556 4023 ó 24
-   ***@***.******@***.***>
-   www.techmake.com<http://www.techmake.com/>
Av. Luis Elizondo 400, Col. Altavista Monterrey, NL,
México, 64840
[signature_582344522]<https://youtube.com/techmakeE>   [signature_2058480464] <https://facebook.com/techmakeE>    [signature_2048821146] <https://www.instagram.com/techmakesolutions/>

Fecha: lunes, 17 de marzo de 2025, 2:14 p.m.
Para: turtlebot/turtlebot4 ***@***.***>
CC: Compras Techmake ***@***.***>, Mention ***@***.***>

image.png (view on web)<https://github.com/user-attachments/assets/6a6e186d-d155-4366-8aad-bf5b898cfed9>

image.png (view on web)<https://github.com/user-attachments/assets/a76b5b82-4cef-4db5-a6c7-a74931957390>

image.png (view on web)<https://github.com/user-attachments/assets/7c68eb76-5a86-4a20-80a5-b1ffbb88fc83>

image.png (view on web)<https://github.com/user-attachments/assets/6a6e186d-d155-4366-8aad-bf5b898cfed9>

image.png (view on web)<https://github.com/user-attachments/assets/a76b5b82-4cef-4db5-a6c7-a74931957390>

image.png (view on web)<https://github.com/user-attachments/assets/7c68eb76-5a86-4a20-80a5-b1ffbb88fc83>

RustyCPR, 2025-04-03:
Hi Lea - thanks for the details. It seems like the PCBA requires replacement. 
Can you please email **_support(at)clearpathrobotics(dot)com_** with Subject title Github 567 and include the TB4 serial number photo (located at the bottom of your TB4)? 
I'll review if the TB4 is still within the 1 year warranty period.
Thank you,
Rusty

TechmakeS, 2025-04-03:
Yes, of course, I just sent the requested email.

Best Regards,
Lea Salazar
Departamento de Compras

[signature_1392675457]

Techmake Solutions, S.A. de C.V.
-   81 2556 4023 ó 24
-   ***@***.******@***.***>
-   www.techmake.com<http://www.techmake.com/>
Av. Luis Elizondo 400, Col. Altavista Monterrey, NL,
México, 64840
[signature_582344522]<https://youtube.com/techmakeE>   [signature_2058480464] <https://facebook.com/techmakeE>    [signature_2048821146] <https://www.instagram.com/techmakesolutions/>

Fecha: miércoles, 2 de abril de 2025, 6:25 p.m.

RustyCPR, 2025-04-17:
Hi @TechmakeS I will close this issue as its resolved. 
Thanks,
Rusty

closed 2025-04-17 by RustyCPR