turtlebot/turtlebot4 #569: Turtltebot4 Power Issues

dhruvapaul, 2025-03-11:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Simple Discovery

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

I do not know

### Type of issue

Battery

### Expected behaviour

Upon placing the robot on to the dock, the lidar should continuously spin and the raspberry pi should light up and display something.

### Actual behaviour

Upon placing the robot on to the dock, the lidar stops spinning after a few seconds, the raspberry pi never shows anything, and the center light button keeps spinning white. I cannot connect to the robot with a wired connection. I suspect this is an issue of the power but I do not know how to move forward.

### Error messages

```bash

```

### To Reproduce

1. Place robot on to the dock
2. Observe the actual behavior, even after waiting indefinitely

smatarCPR, 2025-03-17:
Hello @dhruvapaul ,
Thank you for reaching out on the Turtlebot 4 Github page. I am very sorry to hear that you are having issues with your Turtlebot 4 unit.

Some items to verify:

Make sure the Create3 power distribution board is fully inserted into the Create3 base, this will power the RPi, PCBa, lidar and camera.
https://github.com/turtlebot/turtlebot4/issues/394#issuecomment-2045931389

When booting does the Create3 play a chime?
-What is the color ring light on the Create3 showing: https://iroboteducation.github.io/create3_docs/hw/face/

Please verify the other connections between the Create3 Power Distribution board and the TB4 PCBA and RPi:
https://github.com/turtlebot/turtlebot4/issues/537#issuecomment-2638012695

Looking forward to hearing from you soon.
Best Regards,
Saif

dhruvapaul, 2025-04-08:
The Create3 power distribution board is full inserted. When booting, the light spins white continuously, all while there is no response from the Raspberry Pi.