turtlebot/turtlebot4 #570: Nav2 stopped working, *failed to bring up all requested nodes*

YogiMind, 2025-03-12:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Discovery Server

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

Package: ros-humble-turtlebot4-desktop
Version: 1.0.0-1jammy.20250121.155754

Package: ros-humble-turtlebot4-navigation
Version: 1.0.5-1jammy.20250210.230608

### Type of issue

Navigation (SLAM, Nav2 etc.)

### Expected behaviour

I have a namespace for my robot: /raphael

I launch slam:
ros2 launch turtlebot4_navigation slam.launch.py namespace:=/raphael
then nav2:
ros2 launch turtlebot4_navigation nav2.launch.py namespace:=/raphael
and finally rviz:
ros2 launch turtlebot4_viz view_robot.launch.py namespace:=/raphael

In RViz I can set the initial pose estimate
then I set a Goal estimate and the turtlebot starts moving to the goal

### Actual behaviour

When setting the goal pose, the robot does not move
instead i get some errors in my terminals

### Error messages

```bash
ros2 launch turtlebot4_navigation nav2.launch.py namespace:=/raphael

...
...

[raphael.lifecycle_manager_navigation]: Activating controller_server
[controller_server-1] [INFO] [1741702134.261919167] [raphael.controller_server]: Activating
[controller_server-1] [INFO] [1741702134.261998437] [raphael.local_costmap.local_costmap]: Activating
[controller_server-1] [INFO] [1741702134.262016525] [raphael.local_costmap.local_costmap]: Checking transform
[controller_server-1] [INFO] [1741702134.262093700] [raphael.local_costmap.local_costmap]: start
[controller_server-1] [INFO] [1741702134.269342664] [raphael.local_costmap.local_costmap]: StaticLayer: Resizing static layer to 249 X 106 at 0.050000 m/pix
[controller_server-1] [INFO] [1741702134.513188514] [raphael.controller_server]: Creating bond (controller_server) to lifecycle manager.
[planner_server-3] [INFO] [1741702134.576051439] [raphael.global_costmap.global_costmap]: StaticLayer: Resizing costmap to 249 X 106 at 0.050000 m/pix
[controller_server-1] [INFO] [1741702135.462246091] [raphael.local_costmap.local_costmap]: StaticLayer: Resizing static layer to 251 X 106 at 0.050000 m/pix
[controller_server-1] [INFO] [1741702136.462251676] [raphael.local_costmap.local_costmap]: StaticLayer: Resizing static layer to 251 X 106 at 0.050000 m/pix
[lifecycle_manager-8] [ERROR] [1741702136.521638533] [raphael.lifecycle_manager_navigation]: Server controller_server was unable to be reached after 4.00s by bond. This server may be misconfigured.
[lifecycle_manager-8] [ERROR] [1741702136.521685745] [raphael.lifecycle_manager_navigation]: Failed to bring up all requested nodes. Aborting bringup.
[controller_server-1] [INFO] [1741702142.062231689] [raphael.local_costmap.local_costmap]: StaticLayer: Resizing static layer to 251 X 106 at 0.050000 m/pix
[controller_server-1] [INFO] [1741702142.326389401] [raphael.local_costmap.local_costmap]: Message Filter dropping message: frame 'rplidar_link' at time 1741702141.454 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[controller_server-1] [INFO] [1741702142.415998244] [raphael.local_costmap.local_costmap]: Message Filter dropping message: frame 'rplidar_link' at time 1741702141.458 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
------------------------------------------------------------------------------------

[rviz2-1] [INFO] [1741702774.005971664] [rviz2]: Setting estimate pose: Frame:map, Position(-0.569635, 0.0111301, 0), Orientation(0, 0, -0.997857, 0.0654397) = Angle: -3.01062
[rviz2-1] Start navigation
[rviz2-1] [INFO] [1741702780.216311806] [raphael.rviz_navigation_dialog_action_client]: NavigateToPose will be called using the BT Navigator's default behavior tree.
[rviz2-1] [ERROR] [1741702780.216844835] [raphael.rviz_navigation_dialog_action_client]: Goal was rejected by server
```

### To Reproduce

replace namespace with appropriate for you

launch slam:
ros2 launch turtlebot4_navigation slam.launch.py namespace:=/namespace
then nav2:
ros2 launch turtlebot4_navigation nav2.launch.py namespace:=/namespace
and finally rviz:
ros2 launch turtlebot4_viz view_robot.launch.py namespace:=/namespace

in RViz:
set initial pose
set Goal pose

### Other notes

The weird thing is that this worked perfectly fine one week ago, following the exact same procedure.

smatarCPR, 2025-03-24:
Hello @YogiMind ,
Thank you for reaching out on the Turtlebot 4 Github page. I am very sorry to hear that you are having issues with your Turtlebot 4 Unit. Based on the details provided:

- We typically see the message `controller_server-1] [INFO] [1741702142.415998244] [raphael.local_costmap.local_costmap]: Message Filter dropping message: frame 'rplidar_link' at time 1741702141.458 for reason 'the timestamp on the message is earlier than all the data in the transform cache'` when there is a time desync between the three devices. Our recommendation is to: 

1. Setup NTP config on the Create3 to so that it time syncs with the RPi - see instructions in [ticket #216](https://github.com/turtlebot/turtlebot4/issues/216)
2. Setup NTP server on your remote PC on the network -  https://ubuntu.com/server/docs/how-to-serve-the-network-time-protocol-with-chrony. Please follow the instructions up until and including the section that is "Enable serving the Network Time Protocol". Specifically you'll need to install chrony and then "allow" the RPi access to the server. You also need to make sure that you restart chrony after making any changes to the file (sudo systemctl restart chrony.service). As an additional note make sure that this device has a static IP so that the RPi can always find it at a given IP address.
3. Redirect the Pi to sync with the new NTP server - Open the /etc/chrony.conf file and add a new server line similar to the one you've used for the Create3 but changing the IP address to point at the NTP server set up in the previous step. This is what will tell the RPi to sync to your computer and then you can verify that after they have restarted chrony that you are able to sync properly.

Let me know how this recommendation works for you. Looking forward to hearing from you soon.
Best Regards,
Saif M.

smatarCPR, 2025-04-22:
Hello @YogiMind ,
This issue is being closed due to inactivity. If you are still experiencing the issue, feel free to reopen this ticket when you are ready to continue the troubleshooting process.
Best Regards,
Saif

closed 2025-04-22 by smatarCPR