turtlebot/turtlebot4 #571: simulator not getting topic info

dan9thsense, 2025-03-15:
### Robot Model

Turtlebot4 Standard

### ROS distro

Jazzy

### Networking Configuration

Simple Discovery

### OS

Ubuntu 24.04

### Built from source or installed?

Installed

### Package version

dpkg -s ros-jazzy-turtlebot4-simulator
Package: ros-jazzy-turtlebot4-simulator
Status: install ok installed
Priority: optional
Section: misc
Installed-Size: 41
Maintainer: rkreinin <rkreinin@clearpathrobotics.com>
Architecture: amd64
Version: 2.0.2-1noble.20250305.002613
Depends: ros-jazzy-turtlebot4-gz-bringup, ros-jazzy-turtlebot4-gz-gui-plugins, ros-jazzy-turtlebot4-gz-toolbox, ros-jazzy-ros-workspace
Description: Metapackage for Turtlebot4 simulations

### Type of issue

Other

### Expected behaviour

Just trying to get the simulator to launch. Expected gazebo to come up, the robot model to be shown, and the interfaces to work.

### Actual behaviour

Gazebo harmonic runs and the robot panel appears but is unresponsive. No robot model appears.

continuous repeated messages:
[create-7] [INFO] [1742057216.629996336] [ros_gz_sim]: Waiting messages on topic [robot_description].
[create-8] [INFO] [1742057217.530157809] [ros_gz_sim]: Waiting messages on topic [standard_dock_description].

### Error messages

```bash
These messages are repeated over and over:
[create-7] [INFO] [1742056312.223504244] [ros_gz_sim]: Waiting messages on topic [robot_description].
[create-8] [INFO] [1742056312.323225705] [ros_gz_sim]: Waiting messages on topic [standard_dock_description].

Here is the full console log:
ros2 launch turtlebot4_gz_bringup turtlebot4_gz.launch.py
[INFO] [launch]: All log files can be found below /home/dbarry/.ros/log/2025-03-15-12-46-44-366056-legion-63223
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [gazebo-1]: process started with pid [63228]
[INFO] [parameter_bridge-2]: process started with pid [63229]
[INFO] [robot_state_publisher-3]: process started with pid [63230]
[INFO] [joint_state_publisher-4]: process started with pid [63232]
[INFO] [robot_state_publisher-5]: process started with pid [63233]
[INFO] [static_transform_publisher-6]: process started with pid [63234]
[INFO] [create-7]: process started with pid [63235]
[INFO] [create-8]: process started with pid [63237]
[INFO] [parameter_bridge-9]: process started with pid [63238]
[... 390 lines cut ...]
[gazebo-1] [Dbg] [SimulationRunner.cc:560] [Dbg] [SimulationRunner.cc:560] Exiting postupdate worker thread (1)
[gazebo-1] [Dbg] [SimulationRunner.cc:560] Exiting postupdate worker thread (0)
[ERROR] [gazebo-1]: process has died [pid 63228, exit code -2, cmd 'ruby $(which gz) sim warehouse.sdf -r -v 4 --gui-config /home/dbarry/turtlebot4/install/turtlebot4_gz_bringup/share/turtlebot4_gz_bringup/gui/standard/gui.config --force-version 8'].
```

### To Reproduce

start a discovery server on the computer:
`fastdds discovery --server-id 0`
launch the default sim bring up file:
`ros2 launch turtlebot4_gz_bringup turtlebot4_gz.launch.py`

### Other notes

Following the [simulation instructions page.](https://turtlebot.github.io/turtlebot4-user-manual/software/turtlebot4_simulator.html#gazebo-bringup)
There is a line on that page saying, "It is recommended to configure the computer for simple discovery prior to running any simulations." but it is unclear how to do that since the network instructions for simple server only apply to the turtlebot, not the computer. There is this line as well, "The simulation can be run in discovery server mode but the discovery server referenced must exist and be accessible for discovery as well as not having any conflicts in the topics launched." I created a discovery server on the laptop as follows:
fastdds discovery --server-id 0
### Server is running ###
  Participant Type:   SERVER
  Security:           NO
  Server ID:          0
  Server GUID prefix: 44.53.00.5f.45.50.52.4f.53.49.4d.41
  Server Addresses:   UDPv4:[0.0.0.0]:11811

But it does not seem to matter whether this exists or not.

dan9thsense, 2025-03-15:
I also tried running from source built packages but same behavior. Note I listed "Simple discovery" but it is actually a discovery server (see "other notes")

dan9thsense, 2025-03-15:
Partly solved. I changed locations and forgot to reset my ROS_IP. The messages are gone and the robot now appears but it is unresponsive to move commands

dan9thsense, 2025-03-15:
nodes show up but not topics:
ros2 topic list
/parameter_events
/rosout

ros2 node list
/bumper_contact_bridge
/buttons_msg_bridge
/camera_bridge
/camera_stf
/cliff_front_left_bridge
/cliff_front_right_bridge
/cliff_side_left_bridge
/cliff_side_right_bridge
/clock_bridge
/cmd_vel_bridge
/dock_state_publisher
/hazards_vector_publisher
/hmi_buttons_msg_bridge
/hmi_display_msg_bridge
/hmi_led_msg_bridge
/interface_buttons_node
/ir_intensity_front_center_left_bridge
/ir_intensity_front_center_right_bridge
/ir_intensity_front_left_bridge
/ir_intensity_front_right_bridge
/ir_intensity_left_bridge
/ir_intensity_right_bridge
/ir_intensity_side_left_bridge
/ir_intensity_vector_publisher
/joint_state_publisher
/kidnap_estimator_publisher
/lidar_bridge
/mock_publisher
/motion_control
/odom_base_tf_bridge
/pose_bridge
/pose_republisher_node
/robot_state
/robot_state_publisher
/rplidar_stf
/sensors_node
/tf_odom_std_dock_link_publisher
/turtlebot4_gz_hmi_node
/turtlebot4_node
/ui_mgr
/wheel_status_publisher

dan9thsense, 2025-03-15:
Looking at the gazebo topics with
`gz topic -l`
shows a full list of topics, including /cmd_vel so I tried setting up a ros bridge for cmd_vel with
`ros2 run ros_gz_bridge parameter_bridge /cmd_vel@geometry_msgs/msg/Twist@gz.msgs.Twist`
but it did not help.

dan9thsense, 2025-03-15:
It is now mostly working. The ros2 topic list still does not show any of the robotic topics but gazebo loads, displays the environment and the robot model, and responds to cmd_vel messages.

Here are the things needed that are not in the simulation page instructions:

Discovery server
`fastdds discovery --server-id 0`

ros gz bridge, for example:
ros2 run ros_gz_bridge parameter_bridge /cmd_vel@geometry_msgs/msg/Twist@gz.msgs.Twist

monitor the gz topic:
gz topic -e --topic /cmd_vel

send a command:
ros2 topic pub /cmd_vel geometry_msgs/msg/Twist '{linear: {x: 0.5, y: 0.0, z: 0.0}, angular: {x: 0.0, y: 0.0, z: 0.5}}' -r 1

smatarCPR, 2025-04-02:
Hello @dan9thsense ,
Thank you for reaching out on the Turtlebot 4 Github page. Apologies for the delay in response on this. I have the following details/recommendations to share you with you regarding this matter:

- Based on your current setup, there is no need for any additional packages to get the Gazebo simulation working.
- In addition, you should not need a discovery server for running the Gazebo simulation
- If you wish to continue using a discover server ensure the following items:

1. Ensure that for every new terminal window or tab opened you source the setup.bash file so that the networking related environment variables have been passed through and are applied to the terminal session
2. There are no additional network bridges being used which may be causing communication conflicts between your computer, the discovery server, and the Gazebo simulation

- My recommendation in terms of using the discover would it be possible to revert the changes you made to the packages being used, and return to the default settings, then provide the terminal outputs after you have sourced the setup.bash file for discovery server settings.

Looking forward to hearing from you soon.
Best Regards,
Saif

dan9thsense, 2025-04-03:
The setup.bash file is sourced in .bashrc, so it is always sourced with new terminals. Going back to the default package settings (by starting entirely over from scratch) had no impact. The entire project works except that "ros2 topic list" and "ros2 topic echo some_topic" do not work if there is a discovery server running (they do work if there is no discovery server, but then, of course, I am unable to access ROS messages from a remote terminal). With the discovery server running, nodes that subscribe to topics do receive them normally, both locally and remotely. For example, rviz2 runs just fine on a remote machine.

Unless you have some other ideas, I'm giving up at this point and just living with the inability to use ros2 topic list or ros2 topic echo ...

smatarCPR, 2025-04-23:
Hello @dan9thsense ,
Apologies for the delay in response. I have the following questions regarding the setup of the simulator:

- Are the terminals being setup with super client permissions? The default generator for the setup.bash file relies on the use of super client permissions to generate and make modifications to setup.bash
- Would it be possible to provide a copy of the setup.bash being generated?

Looking forward to hearing from you soon.
Best Regards,
Saif

smatarCPR, 2025-06-02:
Hello @dan9thsense 
This issue is being closed due to inactivity. If you are still experiencing the issue, feel free to reopen this ticket when you are ready to continue the troubleshooting process. 
Best Regards,
Saif

closed 2025-06-02 by smatarCPR