turtlebot/turtlebot4 #572: Unable to sync time to get NAV working

Simba185, 2025-03-18:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Simple Discovery

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

standard

### Type of issue

Select One

### Expected behaviour

When i am trying to do Navigation, there is no way to solve the tyime stamp issues ( causing dropped frames)

Pi Create and user laptop are all set to NTP and clocks are the same but the delivered frames are .5 to 1 second late.

How to fix this?

### Actual behaviour

cannot map or navigate

### Error messages

```bash
dropped frames due to time sync
```

### To Reproduce

lanunch smal and rviz

### Other notes

Lidar has to be manually started rotating.

RustyCPR, 2025-03-20:
Hi Tristan, 
Sorry to see your having issues. 
Can you please provide the below details of the issue: 

- Provide the Turtlebot log `journalctl -b > turtlebot4_log.txt`
- Terminal outputs off all active terminal screens. 
- Details to how you set up NTP

Lidar has to be manually started rotating.
> See your past ticket to stop lidar from spinning on power up  [LINK](https://github.com/turtlebot/turtlebot4/issues/299)
Let us know if this resolves that.

Simba185, 2025-03-21:
Hi there Rusty, 

Can you please provide instructions for the turtlebot4_log.txt

And no this is not the power state issue but in fact the lidar is powered and trying to spin but it will not start with additional assistance. ( poor design with rubber band).

Please see the screenshots below.
 
As you can see the robot not working well and is never consistent.

[image]
[image]
[image]
[image]

RustyCPR, 2025-03-26:
Hi there Tristan - got your message. 
Please make sure you applied a namespace and rebooted the robot. 
If issues are still present - logs please. 
Thanks,
Rusty

Simba185, 2025-03-26:
Hi there Rusty,

Turtlebot4- setup configured the Create 3 namespace to "/Do_not_use" is this correct? If so what namespace am I setting. ( as this is not listed in the "Discovery server setup - Robot setup"

Also the robot is unable to reboot, it needs the battery removed.

hilary-luo, 2025-03-26:
@Simba185 There is a known bug in Humble where a namespace must be applied for the create3 republisher to work properly. I should have clarified what I meant when I said that the namespace would show as `_do_not_use` followed by your chosen namespace and to make sure to apply your own namespace. When we say apply a namespace we are referring to in the `turtlebot4-setup` tool -> ROS Setup -> Bash Setup -> ROBOT_NAMESPACE. You can set it to a valid ROS namespace of your choice (eg. robot1). After applying this you should be able to reboot the robot. After rebooting, all topics should show up as /robot1/topic_name and the create3 will show it's namespace as /_do_not_use/robot1

Simba185, 2025-03-26:
Hi there Hilary,

After applying the  ROBOT_Namespace in setup.bash the robot took over 20 min to fully boot and turn on the COMM and BAT Led's

I will apply the User PC configuration to see if we can get the nav working.

Tristan

hilary-luo, 2025-03-27:
@Simba185 There are some steps we can take to troubleshoot the boot time but I'd like to see if the boot time is persistent before we get into that and also get the Nav working.

Simba185, 2025-03-27:
So I applied the settings as per the "User PC" for discovery server but there are no topics other than 
/parameter_events 
/rosout

I will send the bash files shortly.

What could be the problem?

hilary-luo, 2025-03-27:
The most common causes are that

1. the bash settings that you created have not been applied. (usually the bashrc was not sourced after the changes were applied)
2. The daemon was not restarted after the changes were applied or the topic echo has not been run twice after restart of the daemon
3. The settings on the computer are incorrect (for example the wireless IP address is used but the computer is only connected over wired, or the Server ID is incorrect)
4. The computers are not on the same network (can't ping)

Simba185, 2025-03-27:
Hi there Hilary

All 4 steps have been checked.

I think these pictures show the proof.

The robot is showing all topics on ssh

hilary-luo, 2025-03-27:
In your screenshots that you have shared, the second one shows a setup file for Discovery server ID 4 and the third screenshot is showing a setup file for simple discovery. Without context of what files you are showing and what is being done with them I can't draw conclusions. 

You can tell the discovery server ID by counting semicolons. The first IP before the semicolon would be for Server ID 0, next would be for Server ID 1 etc.

A better way to compare and check that both terminals on the robot and the offboard computer have the same settings look at the environment variables `printenv | grep ROS`. This filter does not capture all relevant settings but it captures the main ones.

Simba185, 2025-03-27:
These are the files created by the discovery server setup.

What details do you need to advise the solution, as it is not working.

Please call me as we pay for support and don't receive it.

hilary-luo, 2025-03-27:
@Simba185 I can support you directly here, support calls have to be scheduled through support, and not through me.

> These are the files created by the discovery server setup.
I am looking for specifically which device this file is on and the exact path to the file.

I would suggest first running the aforementioned `printenv | grep ROS` on both the offboard computer and on the robot and you can share the output of that here (specifying which is which).

hilary-luo, 2025-03-27:
ROS needs to be in all caps

Simba185, 2025-03-27:
user pc:

[image]

Simba185, 2025-03-27:
ohhhhh no wonder this dosn't make any sence

Robot:

[image]

Simba185, 2025-03-27:
user pc:

[image]

they seem the same other than the discovery server.

hilary-luo, 2025-03-27:
Yup, so that confirms that both devices are configured with the same settings which is good. Next can you confirm that

1. You can ping the robot at the 192.168.50.124 address (or that you are ssh'd in from this computer using that IP address)?

Then I want you to do `ros2 daemon stop` on both the robot and the offboard computer and then retry `ros2 topic list` twice on both of them. Tell me which one(s) are successful.

Simba185, 2025-03-27:
both robot and pc can ping ip 192.168.50.124

only robot has complete topic list after stopping daemon.

[image]

hilary-luo, 2025-03-27:
Next I want to check on the settings of the server that is running on the robot. Can you share the output of `systemctl status discovery.service`.

While you are doing that, I also want to double check that you updated these packages to the latest with sudo apt update / upgrade?

Simba185, 2025-03-27:
ubuntu@ubuntu:~$ systemctl status discovery.server
Unit discovery.server.service could not be found.

yikes.....

and fully upgraded.

[image]

hilary-luo, 2025-03-27:
`systemctl status discovery.service` not server

Simba185, 2025-03-27:
lets hope thats better

[image]

hilary-luo, 2025-03-27:
Can you try restarting the discovery server service on the robot? `sudo systemctl restart discovery.service`

It does look healthy but I am trying to narrow down possibilities.

Simba185, 2025-03-27:
so stop the ros2 daemon? I was thinking it was already stopped.

Simba185, 2025-03-27:
still only topics on robot.

[image]

Can we schedule a call for Friday? unless you can remote in a take a look.

hilary-luo, 2025-03-27:
Every time we run ros2 topic list it will start the daemon again. Every time I request that we stop the daemon I am asking for it to essentially be reset. Did you restart the discovery service first before what you shared there? I don't see it in the screenshot but I am assuming that you did it out of frame.

At this point the wifi network should be eliminated as a factor. Can you connect your offboard computer to the robot over a wired connection, reconfigure the offboard computer to use the raspberry pi's wired address and then do the daemon stop and then topic list twice? Unfortunately this isn't something I can really facilitate remotely so let's get through this step first.

Simba185, 2025-03-27:
yes discovery restart was done prior to daemon stop

I tried to stop/start and topic list a few times but no luck. i figured the ros topic list  restarted the daemon, you can see that on the robot

can you advise what settings need to change to the new ip on the user pc?

hilary-luo, 2025-03-27:
It's the ip address in the ROS_DISCOVERY_SERVER environment variable that needs to be updated on the offboard computer and then resourced in your terminal. Where it previously had the Wifi IP address, it needs to be updated with the wired address (192.168.185.3). Of course also make sure that you can ping / ssh using that address.

Simba185, 2025-03-27:
ok can you please clarify how to do this without breaking anything.

ROS_DISCOVERY_SERVER=192.168.185.3 or do we need the semicolons.

hilary-luo, 2025-03-27:
You need the semicolons, I would just suggest using the script again to configure it. If you want to do it manually then you are best to edit the bash file to replace only the IP address in that line (leaving the semicolons and quotes etc) then re-source the bashrc.

Simba185, 2025-03-27:
ok did that still the same issues

[image]

hilary-luo, 2025-03-27:
Can you print the environment variables again to ensure that it was applied properly?

Simba185, 2025-03-27:
yep its there

[image]

Simba185, 2025-03-28:
Hi there Hilary, 

Yes Native installation of Ubunbu.

Can you share how to get the .bashrc file? I have tried to make a copy and send to usb stick but its not working.

Simba185, 2025-03-28:
files for robot here

[turtlebot_log1.txt](https://github.com/user-attachments/files/19506633/turtlebot_log1.txt)
[setup.bash.txt](https://github.com/user-attachments/files/19506636/setup.bash.txt)
[ip a output.txt](https://github.com/user-attachments/files/19506635/ip.a.output.txt)
[bashrc.txt](https://github.com/user-attachments/files/19506638/bashrc.txt)
[50-wifis.yaml.txt](https://github.com/user-attachments/files/19506634/50-wifis.yaml.txt)
[40-ethernets.yaml.txt](https://github.com/user-attachments/files/19506637/40-ethernets.yaml.txt)

Simba185, 2025-03-28:
files from user pc here

[setup.bash.txt](https://github.com/user-attachments/files/19506653/setup.bash.txt)
[ipa.txt](https://github.com/user-attachments/files/19506652/ipa.txt)
[bashrc.txt](https://github.com/user-attachments/files/19506654/bashrc.txt)

hilary-luo, 2025-03-28:
From your user PC can you please also provide `/etc/turtlebot4_discovery/setup.bash`

Simba185, 2025-03-28:
no problem.
[setup.bash discovery.txt](https://github.com/user-attachments/files/19507216/setup.bash.discovery.txt)

RustyCPR, 2025-05-12:
Closing this ticket due to no-updates or activity.

closed 2025-05-12 by RustyCPR