turtlebot/turtlebot4 #573: TB4's nav2 behaves unstably in humble.

DongDong-Juice, 2025-03-25:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Discovery Server

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

dpkg -s ros-$ROS_DISTRO-turtlebot4-PACKAGE_WITH_ISSUE
dpkg-query: package 'ros-humble-turtlebot4-package_with_issue' is not installed and no information is available
Use dpkg --info (= dpkg-deb --info) to examine archive files.

### Type of issue

Navigation (SLAM, Nav2 etc.)

### Expected behaviour

Follow the specified waypoint

### Actual behaviour

Disobedient like an excited puppy

### Error messages

```bash
[component_container_isolated-1] [INFO] [1742874401.928505780] [local_costmap.local_costmap]: Message Filter dropping message: frame 'rplidar_link' at time 1742874400.889 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[component_container_isolated-1] [INFO] [1742874401.928505748] [global_costmap.global_costmap]: Message Filter dropping message: frame 'rplidar_link' at time 1742874400.889 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[component_container_isolated-1] [INFO] [1742874401.959422610] [local_costmap.local_costmap]: Message Filter dropping message: frame 'rplidar_link' at time 1742874401.025 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[component_container_isolated-1] [INFO] [1742874401.959493437] [global_costmap.global_costmap]: Message Filter dropping message: frame 'rplidar_link' at time 1742874401.025 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[component_container_isolated-1] [INFO] [1742874402.005114902] [local_costmap.local_costmap]: Message Filter dropping message: frame 'rplidar_link' at time 1742874401.148 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[component_container_isolated-1] [INFO] [1742874402.005322716] [global_costmap.global_costmap]: Message Filter dropping message: frame 'rplidar_link' at time 1742874401.148 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[component_container_isolated-1] [INFO] [1742874402.255754784] [local_costmap.local_costmap]: Message Filter dropping message: frame 'rplidar_link' at time 1742874401.283 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[component_container_isolated-1] [INFO] [1742874402.255781249] [global_costmap.global_costmap]: Message Filter dropping message: frame 'rplidar_link' at time 1742874401.283 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[component_container_isolated-1] [INFO] [1742874402.451378743] [local_costmap.local_costmap]: Message Filter dropping message: frame 'rplidar_link' at time 1742874401.665 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[component_container_isolated-1] [INFO] [1742874402.451381447] [global_costmap.global_costmap]: Message Filter dropping message: frame 'rplidar_link' at time 1742874401.665 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[component_container_isolated-1] [INFO] [1742874402.466781880] [global_costmap.global_costmap]: Message Filter dropping message: frame 'rplidar_link' at time 1742874401.666 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[component_container_isolated-1] [INFO] [1742874402.466781746] [local_costmap.local_costmap]: Message Filter dropping message: frame 'rplidar_link' at time 1742874401.666 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[component_container_isolated-1] [INFO] [1742874402.481566479] [global_costmap.global_costmap]: Message Filter dropping message: frame 'rplidar_link' at time 1742874401.800 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[component_container_isolated-1] [INFO] [1742874402.481783505] [local_costmap.local_costmap]: Message Filter dropping message: frame 'rplidar_link' at time 1742874401.800 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[component_container_isolated-1] [INFO] [1742874402.628620266] [amcl]: Message Filter dropping message: frame 'rplidar_link' at time 1742874400.889 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[component_container_isolated-1] [INFO] [1742874402.645847154] [local_costmap.local_costmap]: Message Filter dropping message: frame 'rplidar_link' at time 1742874401.921 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[component_container_isolated-1] [INFO] [1742874402.646707849] [global_costmap.global_costmap]: Message Filter dropping message: frame 'rplidar_link' at time 1742874401.921 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[component_container_isolated-1] [INFO] [1742874402.659428676] [amcl]: Message Filter dropping message: frame 'rplidar_link' at time 1742874401.025 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[component_container_isolated-1] [INFO] [1742874402.683916447] [local_costmap.local_costmap]: Message Filter dropping message: frame 'rplidar_link' at time 1742874402.166 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[component_container_isolated-1] [INFO] [1742874402.683987821] [global_costmap.global_costmap]: Message Filter dropping message: frame 'rplidar_link' at time 1742874402.166 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[component_container_isolated-1] [INFO] [1742874402.705081779] [amcl]: Message Filter dropping message: frame 'rplidar_link' at time 1742874401.148 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[component_container_isolated-1] [INFO] [1742874402.955773059] [amcl]: Message Filter dropping message: frame 'rplidar_link' at time 1742874401.283 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[component_container_isolated-1] [INFO] [1742874403.030912406] [global_costmap.global_costmap]: Message Filter dropping message: frame 'rplidar_link' at time 1742874402.193 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[component_container_isolated-1] [INFO] [1742874403.031011410] [local_costmap.local_costmap]: Message Filter dropping message: frame 'rplidar_link' at time 1742874402.193 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[component_container_isolated-1] [INFO] [1742874403.151401336] [amcl]: Message Filter dropping message: frame 'rplidar_link' at time 1742874401.665 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[component_container_isolated-1] [INFO] [1742874403.166831180] [amcl]: Message Filter dropping message: frame 'rplidar_link' at time 1742874401.666 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[component_container_isolated-1] [INFO] [1742874403.181758545] [amcl]: Message Filter dropping message: frame 'rplidar_link' at time 1742874401.800 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[component_container_isolated-1] [INFO] [1742874403.275002603] [global_costmap.global_costmap]: Message Filter dropping message: frame 'rplidar_link' at time 1742874402.314 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[component_container_isolated-1] [INFO] [1742874403.275012705] [local_costmap.local_costmap]: Message Filter dropping message: frame 'rplidar_link' at time 1742874402.314 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[... 30 lines cut ...]
[component_container_isolated-1] [INFO] [1742877197.603412071] [global_costmap.global_costmap]: Message Filter dropping message: frame 'rplidar_link' at time 1742877196.472 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[... cut ...]
### To Reproduce

1. Build the nav2 package from source 
2. Run ros2 launch nav2_simple_commander waypoint_follower_example_launch.py ​​
3. View

### Other notes

I know there is a nice TB4 nav2 package. But I want to do 
waypoint_follow. So I want to implement it using only nav2 (so that it can be used for other robots too).
Is there anything more specific about nav2 waypoint_follow, like how to do it, or some detailed documentation or relationships? (I've already looked at the docs, but they didn't help.)

RustyCPR, 2025-04-07:
Hi good afternoon @DongDong-Juice 
Thanks for writing in - Waypoint Follower is outside of our support scope for TB4 support. 
I would advise you to confirm you can get [Nav2](https://turtlebot.github.io/turtlebot4-user-manual/tutorials/navigation.html) successfully running (multiple times) on your TB4 and from there if its successfully runs to tackle Waypoint Follower example. 
You'll probably be in a better position to get it running or debug if standard NAV2 runs. 
If you have Waypoint Follower issues occur after confirming Nav2 runs; your best resource is here: https://github.com/ros-navigation/navigation2/issues
 
Let me know how it goes,
Rusty

DongDong-Juice, 2025-04-08:
hi @RustyCPR 

Many parts of nav2 are the same as TB4, because TB4 is also based on nav2. So if TB4 provides the necessary parts for path planning, nav2 will do the path planning and TB4 will only receive the final move command and move!

However, there is a minor but important issue.

TB4 and remote PC are connected to the search server. So... ROS timestamps are very different.

Especially /scan, /odom, /amcl etc which are important for realtime performance were problematic.

I tried ntp configuration to solve this issue, but the best solution was to modify the timeout in nav2_parms.

Now the desired path movement function works fine with nav2.

A minor issue is that TB4 sometimes lacks performance. When running nav2, CPU shows an average of 98-100.

If you are unlucky, TB4 will pass out because of too many commands :/

The only solution in this situation is to put TB4 into sleep mode and then wake it up again... Any other good ideas for this?

Also, I am curious, I would like to create a service (that will work automatically like Ubuntu services!) for nav2 (to be used with TB4). Any good advice? If you have experience, please give me some good advice.

And now that this issue has been resolved, we'll close it.

closed 2025-04-08 by DongDong-Juice