turtlebot/turtlebot4 #574: Localization and Navigation don't work every time

dkavroulakis, 2025-03-26:
### Robot Model

Turtlebot4 Lite

### ROS distro

Humble

### Networking Configuration

Discovery Server

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

Package: ros-humble-turtlebot4-navigation
Status: install ok installed
Priority: optional
Section: misc
Installed-Size: 2470
Maintainer: rkreinin <rkreinin@clearpathrobotics.com>
Architecture: amd64
Version: 1.0.5-1jammy.20241128.064448
Depends: ros-humble-nav2-bringup, ros-humble-nav2-simple-commander, ros-humble-slam-toolbox, ros-humble-ros-workspace
Description: Turtlebot4 Navigation

All packages on both remote PC and Raspberry Pi fully-updated at time of opening this issue.

### Type of issue

Navigation (SLAM, Nav2 etc.)

### Expected behaviour

I expect the localization and nav2 navigation stack to start up without errors so that it could be used for navigation.
The robot to be able to autonomously navigate to goals provided in RViz using a pre-existing map of the environment.

### Actual behaviour

Oftentimes, when I start the localization node, I get an error indicating that AMCL or map_server has failed and is unresponsive. Additionally, when I start the navigation node after running localization, I receive an error stating that the smoother server, controller_server, or bt_navigator has failed and will not respond.

### Error messages

```bash
[amcl-2] [WARN] [1741609758.180257840] [turtlebot4_lite_maritsa.amcl.rclcpp]: failed to send response to /turtlebot4_lite_maritsa/amcl/change_state (timeout): client will not receive response, at ./src/rmw_response.cpp:154, at ./src/rcl/service.c:314

[controller_server-1] [INFO] [1742233902.822780759] [turtlebot4_lite_maritsa.controller_server]: Controller Server has FollowPath controllers available. [controller_server-1] [WARN] [1742233902.925065890] [turtlebot4_lite_maritsa.controller_server.rclcpp]: failed to send response to /turtlebot4_lite_maritsa/controller_server/change_state (timeout): client will not receive response, at ./src/rmw_response.cpp:154, at ./src/rcl/service.c:314

[bt_navigator-5] [WARN] [1742372887.665845554] [turtlebot4_lite_maritsa.bt_navigator.rclcpp]: failed to send response to /turtlebot4_lite_maritsa/bt_navigator/change_state (timeout): client will not receive response, at ./src/rmw_response.cpp:154, at ./src/rcl/service.c:314

[smoother_server.rclcpp]: failed to send response to /turtlebot4_lite_maritsa/smoother_server/change_state (timeout): client will not receive response, at ./src/rmw_response.cpp:154, at ./src/rcl/service.c:314
```

### To Reproduce

1. Run localization to load map: ros2 launch turtlebot4_navigation localization.launch.py map:=/path/to/map.yaml namespace:=/turtlebot4_lite_maritsa
2. Run navigation node: ros2 launch turtlebot4_navigation nav2.launch.py namespace:=/turtlebot4_lite_maritsa

### Other notes

Most of the time, I have to restart the turtlebot4 lite robot multiple times to get everything running successfully.

Current Firmware Version: H.2.6 (Humble)

I have tried both ways of building: from source and using installed packages, but nothing works properly.

smatarCPR, 2025-04-07:
Hello @dkavroulakis 
Thank you for reaching out on the Turtlebot 4 github page. I am very sorry to hear you are encountering issues with your Turtlebot. I have the following questions regarding your current setup:

- How are you currently launching the navigation and SLAM packages? Are you launching the navigation package on the offboard computer? Is there a change in behavior if you attempt to launch everything from Pi? 
- Would it be possible to provide the full terminal outputs when launching the navigation stack? Include where it was launched from, the exact command used, and the full output until termination.

Looking forward to hearing from you soon.
Best Regards,
Saif

smatarCPR, 2025-05-13:
Hello @dkavroulakis 
Thank you for providing the requested details. Based on the terminal outputs and details shared:

- It is indeed possible the issue may be caused by a weak or unstable Wi-Fi network. What sort of network are you working on? Does it have a great deal of network traffic? Is it possible to improve the network stability and strength using a dedicated router?
- If the navigation works without issues when launched on the Pi, but instability occurs when launched remotely, this could be due to a breakdown in communication between the Pi and remote computer. Double check the following items:

1. Check that the package versions are matched between the Raspberry Pi and the Remote Computer. ALso ensure that both are using the latest packages.
2. Is the Remote Computer running Ubuntu natively or is it a virtual machine?

Let me know the results of these checks. Looking forward to hearing from you soon.
Best Regards,
Saif

lasfarW3r, 2025-06-04:
Hi @Jannkar, Im facing the same Problem, the thing is it sometimes even when I restart the tb4 it still doesn't work, but the SLAM works perfectly fine, and I noticed that @smatarCPR talked about internet strength what if I was using AP mode in that case the Turtlebot4 doesn't have acces to the internet still the connection with computer is doable  does the navigation really require   internet ? 
is possible to share the solution if you find it. 

Best Regards, 
Yasir

smatarCPR, 2025-06-10:
Hello @dkavroulakis ,
Thank you for providing these details. In terms of the latency you are seeing we have the following recommendations to all users:

- When running discovery server, ensure that you have unset ros super client in terminal running the navigation stack. This will prevent excessive network traffic, ensuring the navigation stack is sending and receiving only the relevant nodes+topic data.
- As @Jannkar mentioned, connection to the internet is not necessary for navigation. When discussing network strength we are typically referring to the integrity of the connection between devices. This is primarily affected by the network bandwidth available and the other devices or data used on the network.

Let me know if you have any follow up questions or if I can help with anything else.
Best Regards,
Saif

Jannkar, 2025-06-11:
Hi @smatarCPR!

Thanks for mentioning the ROS super client. We are using it in the navigation Docker container.

However, it sounds to me that there is something more behind this issue that would require further investigation. We have been using RViz on our laptops to visualize camera and laser data simultaneously from the Turtlebot, and there was no visible delay in either one. The camera image was pristine and real-time, which to me tells that the WiFi itself can handle the network traffic. And since the navigation stack mainly subscribes to the odometry and laser scan data, which won't cause huge network traffic, it sounds to me that there is something more to this than just the network traffic. 

Especially, since on the first launch we are able to run the navigation stack without any issue, and also at the same time visualize the camera data without any delay. The issues occurred only after restarting the navigation stack.

dkavroulakis, 2025-07-02:
Hello @hilary-luo, your solution works perfectly. Thank you very much for your assistance.

closed 2025-07-02 by dkavroulakis

kimandreas, 2025-08-06:
@hilary-luo what would be the discovery configuration/setup if you operate in multi-robot environment, for example two pc's and two turtlebot4's with each pc's connected to respective turtlebot which is set up as the discover server.  localization and nav2 for each turtlebot runs on the its respective pc.    All devices can be connected to a single wifi router.

just setting up multi-robot in the discovery setup.bash (i.e. in both pc's) as

export ROS_DISCOVERY_SERVER=";;;;;192.168.0.24:11811;192.168.0.5:11811;"

seems to have issues in smooth execution of localization and nav2

kimandreas, 2025-08-07:
@hilary-luo Thanks for the quick response.  I do need to connect multiple pc and turtlebot4 to communicate with each other as putting multiple pcs and robots via common single wifi router is very limiting.  I will open another ticket as you suggested.