turtlebot/turtlebot4 #575: TB4: No communication between RPi and Create3. Boot never finishes

guenterbauch, 2025-03-30:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Simple Discovery

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

used the standard humble.img and flashed the SD-Card via raspberry-imager.

### Type of issue

Networking

### Expected behaviour

I expect a connection between the RPi and the create3 computer. 

### Actual behaviour

The Create3 circle is showing the booting animation. 

ssh into RPi is working, also connection via miniHDMI and keyboard works.

### Error messages

```bash
After changing ROBOT_NAMESPACE from [] to "/robot1", the following message is printed in terminal:

Error: Unable to set Create3 options.Please ensure that the Create3 is fully booted and apply again.                                                                                                                                                                                                                                                                                                 
                                                                                                                                                                                                                                                                                                                                                                                                     
 Details:                                                                                                                                                                                                                                                                                                                                                                                            
Error writing ROS settings to Create3                                                                                                                                                                                                                                                                                                                                                                
                                                                                                                                                                                                                                                                                                                                                                                                     
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current                                                                                                                                                                                                                                                                                                                      
                                 Dload  Upload   Total   Spent    Left  Speed                                                                                                                                                                                                                                                                                                                        
  0     0    0     0    0     0      0      0 --:--:--  0:00:04 --:--:--                                                                                                                                                                                                                                                                                                                                                                                                      
curl: (28) Failed to connect to 192.168.186.2 port 80 after 129456 ms: Connection timed out                                                                                                                                                                                                                                                                                                          
                                                                                                                                                                                                                                                                                                                                                                                                     
Press Q, Esc, or CTRL+C to go back.                                                                                                                                                                                                                                                                                                                                                                  
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
> Okay
```

### To Reproduce

1. Place the tb4 on the dock
2. Wait for boot of Create3
3. ??
4. Boot never finishes

### Other notes

ubuntu@ubuntu:~$ ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN group default qlen 1000
    link/ether d8:3a:dd:1a:36:4e brd ff:ff:ff:ff:ff:ff
3: usb0: <BROADCAST,MULTICAST,UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
    link/ether f2:c9:3e:10:ac:ed brd ff:ff:ff:ff:ff:ff
    inet 192.168.186.3/24 brd 192.168.186.255 scope global noprefixroute usb0
       valid_lft forever preferred_lft forever
4: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
    link/ether d8:3a:dd:1a:36:4f brd ff:ff:ff:ff:ff:ff
    inet 192.168.2.17/24 brd 192.168.2.255 scope global dynamic noprefixroute wlan0
       valid_lft 85358sec preferred_lft 85358sec
    inet6 fe80::da3a:ddff:fe1a:364f/64 scope link 
       valid_lft forever preferred_lft forever

guenterbauch, 2025-03-30:
[image]

My actions so far:

- Removing battery according to the create3 doc
- Veryfing that the toggle is set to usb
- Veryfying optimal cable position 
- Multiple flashing of SD-Card
- Reading older issues on github like https://github.com/iRobotEducation/create3_docs/discussions/200 

Kind regards
Len

guenterbauch, 2025-04-02:
Hi @DongDong-Juice,

thanks for the reply. I tried again.

- changed the RPi from AP to Client Mode, entered the 2.4 Ghz WiFi credentials (local network)
- got the IP addr 192.168.2.17 assigned per DHCP, ssh-ed into the RPi succesfully.
- can reach the standard gateway 192.168.2.1

I cannot access the webserver under 192.168.2.17:8080

Last week I could reach the Create3. I remember that I did some changes to namespace and wifi, but then pressed factory reset. 

Create3 Light still shows the booting animation. Do you have another suggestion?

Kind regards
Len

DongDong-Juice, 2025-04-02:
Hi @guenterbauch 

If you can connect to TB4, can you check it?
source /etc/turtlebot4/setup.bash
ros2 daemon stop ; ros2 daemon start
ros2 topic list

Also

ifconfig 

Does the USB0 interface have an IP address? If so, can you try the ping command?

guenterbauch, 2025-04-02:
Hi @DongDong-Juice,

ubuntu@ubuntu:~$ source /etc/turtlebot4/setup.bash
ubuntu@ubuntu:~$ ros2 daemon stop; ros2 daemon start
The daemon has been stopped
The daemon has been started
ubuntu@ubuntu:~$ ros2 topic list
/battery_state
/cmd_vel
/diagnostics
/diagnostics_agg
/diagnostics_toplevel_state
/dock_status
/function_calls
/hazard_detection
/hmi/buttons
/hmi/display
/hmi/display/message
/hmi/led
/imu
/interface_buttons
/ip
/joint_states
/joy
/joy/set_feedback
/mouse
/oakd/imu/data
/oakd/rgb/preview/camera_info
/oakd/rgb/preview/image_raw
/oakd/rgb/preview/image_raw/compressed
/oakd/rgb/preview/image_raw/compressedDepth
/oakd/rgb/preview/image_raw/theora
/parameter_events
/robot_description
/rosout
/scan
/tf
/tf_static
/wheel_status

ubuntu@ubuntu:~$ ifconfig
Command 'ifconfig' not found, but can be installed with:
sudo apt install net-tools
ubuntu@ubuntu:~$ ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN group default qlen 1000
    link/ether d8:3a:dd:1a:36:4e brd ff:ff:ff:ff:ff:ff
3: usb0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc fq_codel state DOWN group default qlen 1000
    link/ether f2:c9:3e:10:ac:ed brd ff:ff:ff:ff:ff:ff
    inet 192.168.186.3/24 brd 192.168.186.255 scope global noprefixroute usb0
       valid_lft forever preferred_lft forever
    inet6 fe80::f0c9:3eff:fe10:aced/64 scope link
       valid_lft forever preferred_lft forever
4: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
    link/ether d8:3a:dd:1a:36:4f brd ff:ff:ff:ff:ff:ff
    inet 192.168.2.17/24 brd 192.168.2.255 scope global dynamic noprefixroute wlan0
       valid_lft 85906sec preferred_lft 85906sec
    inet6 fe80::da3a:ddff:fe1a:364f/64 scope link
       valid_lft forever preferred_lft forever

ping command

ubuntu@ubuntu:~$ ping 192.168.186.3
PING 192.168.186.3 (192.168.186.3) 56(84) bytes of data.
64 bytes from 192.168.186.3: icmp_seq=1 ttl=64 time=0.106 ms
64 bytes from 192.168.186.3: icmp_seq=2 ttl=64 time=0.117 ms
64 bytes from 192.168.186.3: icmp_seq=3 ttl=64 time=0.157 ms
64 bytes from 192.168.186.3: icmp_seq=4 ttl=64 time=0.130 ms
64 bytes from 192.168.186.3: icmp_seq=5 ttl=64 time=0.120 ms
64 bytes from 192.168.186.3: icmp_seq=6 ttl=64 time=0.123 ms
^C
--- 192.168.186.3 ping statistics ---
6 packets transmitted, 6 received, 0% packet loss, time 5107ms
rtt min/avg/max/mdev = 0.106/0.125/0.157/0.015 ms
ubuntu@ubuntu:~$ ping 192.168.186.2
PING 192.168.186.2 (192.168.186.2) 56(84) bytes of data.
From 192.168.186.3 icmp_seq=3 Destination Host Unreachable
From 192.168.186.3 icmp_seq=4 Destination Host Unreachable
From 192.168.186.3 icmp_seq=5 Destination Host Unreachable
^C
--- 192.168.186.2 ping statistics ---
8 packets transmitted, 0 received, +3 errors, 100% packet loss, time 7151ms
pipe 4

I am looking forward to further suggestions.

Kind regards
Len

DongDong-Juice, 2025-04-02:
@guenterbauch 
ros2 topic echo /battery_state 
Can I get information through the command?

DongDong-Juice, 2025-04-02:
There seems to be a problem with the discovery server setup between create3 and tb4. You need to be able to access the create3 website mentioned above to fix it.
Maybe There seems to be a problem with the namespace setup.

I recommend ssh into TB4 and reconfigure the namespace and wifi with the turtlebot4-setup command, or revert to the initial state.

guenterbauch, 2025-04-02:
Hi @DongDong-Juice ,

> [@guenterbauch](https://github.com/guenterbauch) ros2 topic echo /battery_state Can I get information through the command?

no answer is given.

> There seems to be a problem with the discovery server setup between create3 and tb4. You need to be able to access the create3 website mentioned above to fix it. Maybe There seems to be a problem [...]

Yes, that is my problem. However, I cannot do that. I also tried to connect a USB-C to Ethernet adapter to the Create3 USB C Port, changed the IP address to the suggested input (https://iroboteducation.github.io/create3_docs/setup/network-config/#ethernet-over-usb). No answer to pinging 192.168.186.2.

Kind regards
Len

DongDong-Juice, 2025-04-03:
Hi @guenterbauch 

Have you tried this?
https://turtlebot.github.io/turtlebot4-user-manual/setup/simple_discovery.html#create-3

The most important thing to solve this problem is to connect to wifi on create3. I think it should be connected to discovery. If you have no way to access the webpage, I recommend factory resetting the entire create3, including tb4. That seems to be the best way to solve this problem.

Based on what you said at the beginning... I think the problem occurred while putting the namespace in create3. So I think it would be better to create a new environment and try again slowly. Sorry for not giving a clear solution.

P.S. I had the same problem on my TB4. It was solved after I reverted everything to a brand new one.

guenterbauch, 2025-04-03:
Hi @DongDong-Juice ,

resetting TB4 is possible, but I am unable to reset the Create3. Unfortunately that is my main constraint. 
At this point, I doubt the Create3 has power? Tried to get a Bluetooth connection (by changing the toggle), but cannot find the device. Thanks for your efforts! 

I will wait for an official reply.

Kind regards
Len

DongDong-Juice, 2025-04-04:
@guenterbauch 

Sorry for the late reply.

I need to check back slowly to fix this issue.

Have you tried this command on TB4 that you are connecting to via ssh? turtlebot4-setup?
And have you reconfigured your wifi?

Related information is in the
https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html

Connecting your Raspberry Pi to your network

section.

If turtlebot4-setup doesn't work, see

https://turtlebot.github.io/turtlebot4-user-manual/software/turtlebot4_setup.html#configuration-tools

Configuration Tools
section.

The above process requires you to reconfigure your wifi.
This is different from switching to an AP to get an IP address.

Finally, if create3 is not powered, TB4 will not turn on. This is because TB4 is powered by create3.

guenterbauch, 2025-04-04:
> 
> Have you tried this command on TB4 that you are connecting to via ssh? turtlebot4-setup? And have you reconfigured your wifi?
> 
> Related information is in the https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html
> 
> Connecting your Raspberry Pi to your network
> 
> section.

Yes, I did that. I have an ip-addr.

>Finally, if create3 is not powered, TB4 will not turn on. This is because TB4 is powered by create3.

Okay!

I also checked the forwarding service, which was suggested in https://github.com/turtlebot/turtlebot4/issues/234#issuecomment-1660890402

ubuntu@ubuntu:~$ sudo systemctl status webserver.service
● webserver.service - Create 3 Webserver forwarding service
     Loaded: loaded (/etc/systemd/system/webserver.service; enabled; vendor preset: enabled)
     Active: active (running) since Mon 2024-08-12 15:54:21 UTC; 1h 5min ago
   Main PID: 802 (socat)
      Tasks: 1 (limit: 4422)
     Memory: 1.4M
        CPU: 13ms
     CGroup: /system.slice/webserver.service
             └─802 /usr/bin/socat TCP-LISTEN:8080,fork,reuseaddr tcp:192.168.186.2:80

DongDong-Juice, 2025-04-04:
@guenterbauch 
cat ~.bashrc
Can you tell me the contents?

create3 and TB4 are connected to the discovery server of ros. So I want to check if the settings are correct.

Here is my bashrc contents.

export ROBOT_SETUP=/etc/turtlebot4/setup.bash
source $ROBOT_SETUP
source /etc/turtlebot4/aliases.bash
source /etc/turtlebot4/setup.bash
source /opt/ros/humble/setup.bash
export ROS_DISCOVERY_SERVER={turtlebot4_IP}:11811
export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp <- don't worry about fast

guenterbauch, 2025-04-04:
> [@guenterbauch](https://github.com/guenterbauch) cat ~.bashrc Can you tell me the contents?

ubuntu@ubuntu:~$ cat ~.bashrc
cat: '~.bashrc': No such file or directory

Kind regards
Len

DongDong-Juice, 2025-04-07:
hi @guenterbauch 

I checked a few things.
First ping

  ping 192.168.186.2
  PING 192.168.186.2 (192.168.186.2) 56(84) bytes of data. 64 bytes from 192.168.186.2: icmp_seq=1 ttl=64 time=0.542 ms
  64 bytes from 192.168.186.2: icmp_seq=2 ttl=64 time=0.500 ms
  ^C
  --- 192.168.186.2 ping statistics ---
  2 packets transmitted, 2 received, 0% packet loss, time 1030ms
  rtt min/avg/max/mdev = 0.500/0.521/0.542/0.021 ms

and ip a

  ip a
  1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
   link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
   inet 127.0.0.1/8 scope host lo
   valid_lft forever preferred_lft forever
   inet6 ::1/128 scope host
   valid_lft forever preferred_lft forever
  2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN group default qlen 1000
   link/ether d8:3a:dd:08:60:1c brd ff:ff:ff:ff:ff:ff
  3: usb0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
   link/ether f2:c9:3e:10:ac:ed brd ff:ff:ff:ff:ff:ff
   inet 192.168.186.3/24 brd 192.168.186.255 scope global noprefixroute usb0
   valid_lft forever preferred_lft forever
   inet6 fe80::f0c9:3eff:fe10:aced/64 scope link
   valid_lft forever preferred_lft forever
  4: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
   link/ether d8:3a:dd:08:60:1d brd ff:ff:ff:ff:ff:ff
   inet 172.19.201.100/24 ​​brd 172.19.201.255 scope global dynamic noprefixroute wlan0
  valid_lft 84588sec preferred_lft 84588sec
  inet6 fe80::da3a:ddff:fe08:601d/64 scope link
  valid_lft forever preferred_lft forever

I think we can cautiously confirm that the internet connection is the problem by trying this. So let's try to connect to the internet!

Have you tried it?

Approach from Getting Started with the Create® 3? in https://iroboteducation.github.io/create3_docs/faq/faq/

If you can access the create3 website, your wifi must be 2.4g, not 5g!

Let me know how it goes :) I hope it works.

DongDong-Juice, 2025-04-07:
[image]

It is better to leave namespace empty when there is only one robot.

guenterbauch, 2025-04-08:
Hi @DongDong-Juice ,

thanks for your effort. I understand your guidance, however I cannot activate the AccessPoint of the Create3.
It still shows the spinning animation mentioned here https://iroboteducation.github.io/create3_docs/hw/face/ .

I still checked whether the WiFi of the create3 was displayed in the WLAN search of my host PC (it could be that the light animation hadn't updated to cyan blue?). Unfortunately, this was not the case.

The current status is as follows: The RPi is reachable via SSH and Ethernet. After inserting the battery and connecting it to the dock, the TB4 starts. The light ring remains in the "spinning white" animation, and no "happy sound" is played. 

I look forward to more input.

Kind regards
Len

RustyCPR, 2025-04-10:
Hi @guenterbauch 
Thanks for writing in and sorry your TB4 has encoutered issues. 
If your running Simple Discovery - please remove the namespace, or switch to [Discovery Server](https://turtlebot.github.io/turtlebot4-user-manual/setup/discovery_server.html#discovery-server)

https://turtlebot.github.io/turtlebot4-user-manual/setup/simple_discovery.html#application-configuration
[image]

Let me know how this works out for you. 
Best regards,
Rusty

guenterbauch, 2025-04-11:
Hi @RustyCPR ,

I reflashed the SD card to Humble and connected the RPi to a local network.

My ROS settings -- no connection to Create3

[image]

Then I changed the settings to discovery server:

[image]

That gives me this output:

[image]

The settings in the RPi are then resetted.

Kind regards
Len

RustyCPR, 2025-05-05:
Hi there @guenterbauch 

1. Can you please thoroughly verify the wire connections are connected between RPi and Create3? - see details here: https://github.com/turtlebot/turtlebot4/issues/586#issuecomment-2828856261

2. Power on TB4 and wait 5 minutes to ensure that any boot up operations on the create3 that are happening in the background are completed.  

3. Then ssh into the rpi and install nmap on the RPi: `sudo apt install nmap`

4. Run: `ping 192.168.186.2` to ensure that nothing has changed, let this run for a while and copy/paste the output here. 

5. Scan for the available connections by running: `nmap -sn 192.168.186.3/24` and share the output here. 

6. Scan for details about the create3 IP address by running:  `nmap 192.168.186.2`   and share the output 

7. Please share the file contents of `/etc/netplan/` directory

8. Please share the log from the raspberry pi from the boot where all of these things were attempted:`journalctl -b >> journallog.txt`

Thank you,
Rusty

guenterbauch, 2025-05-06:
> Hi there [@guenterbauch](https://github.com/guenterbauch)
> 
> 1. Can you please thoroughly verify the wire connections are connected between RPi and Create3? - see details here: [Rpi no longer boots up #586 (comment)](https://github.com/turtlebot/turtlebot4/issues/586#issuecomment-2828856261)

All connected.

> 2. Power on TB4 and wait 5 minutes to ensure that any boot up operations on the create3 that are happening in the background are completed.

> 3. Then ssh into the rpi and install nmap on the RPi: `sudo apt install nmap`

Done.
 
> 4. Run: `ping 192.168.186.2` to ensure that nothing has changed, let this run for a while and copy/paste the output here.

```
From 192.168.186.3 icmp_seq=312 Destination Host Unreachable
^C
--- 192.168.186.2 ping statistics ---
317 packets transmitted, 0 received, +74 errors, 100% packet loss, time 323571ms
pipe 4
```
 
> 5. Scan for the available connections by running: `nmap -sn 192.168.186.3/24` and share the output here.

```
ubuntu@ubuntu:~$ nmap -sn 192.168.186.3/24
Starting Nmap 7.80 ( https://nmap.org ) at 2024-08-12 16:26 UTC
Nmap scan report for 192.168.186.3
Host is up (0.00059s latency).
Nmap done: 256 IP addresses (1 host up) scanned in 15.72 seconds
```

> 6. Scan for details about the create3 IP address by running:  `nmap 192.168.186.2`   and share the output

```
ubuntu@ubuntu:~$ nmap 192.168.186.2
Starting Nmap 7.80 ( https://nmap.org ) at 2024-08-12 16:27 UTC
Note: Host seems down. If it is really up, but blocking our ping probes, try -Pn
Nmap done: 1 IP address (0 hosts up) scanned in 3.10 seconds

```
```
ubuntu@ubuntu:~$ nmap -Pn 192.168.186.2
Starting Nmap 7.80 ( https://nmap.org ) at 2024-08-12 16:28 UTC
Nmap scan report for 192.168.186.2
Host is up (0.056s latency).
All 1000 scanned ports on 192.168.186.2 are filtered
```

Nmap done: 1 IP address (1 host up) scanned in 18.97 seconds

> 7. Please share the file contents of `/etc/netplan/` directory

```
ubuntu@ubuntu:~$ cat /etc/netplan/*.yaml
network:
    version: 2
    ethernets:
        renderer: NetworkManager
        eth0:
            addresses:
            - 192.168.185.3/24
            dhcp4: true
            optional: true
        usb0:
            addresses:
            - 192.168.186.3/24
            dhcp4: false
# This file was automatically created by the turtlebot4-setup tool and should not be manually modified

network:
    version: 2
    wifis:
        renderer: NetworkManager
        wlan0:
            access-points:
                LNI4.02:
                    band: 2.4GHz
                    password: strenggeheim
            dhcp4: true
```

> 8. Please share the log from the raspberry pi from the boot where all of these things were attempted:`journalctl -b >> journallog.txt`> 

[journallog.txt](https://github.com/user-attachments/files/20054260/journallog.txt)

Thanks for your guidance.

Kind regards
Len

DongDong-Juice, 2025-05-07:
@guenterbauch 
Long time no see.

The result of the information confirmed through the method suggested by the expert is one thing.

The internet connection between create3 and rpi is not working.

Have you tried changing the IP address of create3? It seems that the current search connection and ping are not working because the connection to 192.168.186.2 is not working.

Also, the search connection is not working, so the battery information and robot status information are not working.

Have you solved the problem related to this issue? Or check it.

RustyCPR, 2025-05-21:
Hi there @guenterbauch 
Thank you for the details and logs. We are looking into this. 

Can you confirm your team did not change wired subnet settings on the Create?https://iroboteducation.github.io/create3_docs/webserver/set-wired-subnet/ 

Regards,
Rusty

guenterbauch, 2025-05-29:
Hi @RustyCPR,

thanks for following up., I have been OOO due to holidays.

I remember that the page has been openend. 
I cannot confirm, that the wired subnet settings has been not changed.
In fact, I recall changing something and hitting save on the webserver. 
Could have been this menu?

Kind regards
Len

guenterbauch, 2025-06-24:
Hi @RustyCPR,

I don't remember if we made a change there. Is it possible to get the tb4 back to work, if a change has been made there and we don't know the subnet?

Kind regards 
Len