turtlebot/turtlebot4 #576: TB4 change SLAM alogrithm

Ragnar7982, 2025-03-31:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Simple Discovery

### OS

Ubuntu 22.04

### Built from source or installed?

Built from Source

### Package version

ros2 Humble Ubuntu22.04

### Type of issue

Navigation (SLAM, Nav2 etc.)

### Expected behaviour

Hi, is there any way that can change TB4 2D slam algorithm? cuz tb4_ws can not change(read-only), if I wanna use others method, how do I do ?

### Actual behaviour

Hi, is there any way that can change TB4 2D slam algorithm? cuz tb4_ws can not change(read-only), if I wanna use others method, how do I do ?

### Error messages

```bash

```

### To Reproduce

.

smatarCPR, 2025-04-07:
Hello @Ragnar7982 
Thank you for reaching out on the Turtlebot 4 Github page. I am very sorry to hear that you are having issues with your Turtlebot 4 unit. Regarding your question, I have the following inquiries/recommendations:

- Where is the tb4_ws coming from? This is not a part of the default configuration. Is this the custom workspace you have made to house your source installation?
- In terms of making modifications to how the 2D SLAM algorithm works, the Turtlebot uses the SLAM toolbox, and custom configurations can be passed in when launching - https://github.com/SteveMacenski/slam_toolbox

Let me know if you have any follow up questions or if I can help with anything else.
Best Regards,
Saif

smatarCPR, 2025-05-12:
Hello @Ragnar7982 
This issue is being closed due to inactivity. If you are still experiencing the issue, feel free to reopen this ticket when you are ready to continue the troubleshooting process. 
Best Regards,
Saif

closed 2025-05-12 by smatarCPR