turtlebot/turtlebot4 #578: Localization with Generated Map too Slow

Saitamacode, 2025-04-03:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Simple Discovery

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

I do not know

### Type of issue

Navigation (SLAM, Nav2 etc.)

### Expected behaviour

After generating a custom map with SLAM, if I were to test Localization navigation of the turltbot4 with that custom map, the robot should be able to move with little delay.

### Actual behaviour

There is immense delays such that the robot only moves after 5-10 minutes of waiting, if prompted to move around the custom map through Rviz. Directly moving the robot with teleop works normally, it is just movement through Rviz that has such high delays.

### Error messages

```bash

```

### To Reproduce

1. Generate map with SLAM
2. Use map to navigate and move with Localization

RustyCPR, 2025-04-07:
Hi @Saitamacode 

Thanks for writing in. I could use more details to assist with the issue. 

- Could you please detail where you are launching SLAM? 
- Are you on a 5GHz WiFi network? 
- Could you please provide full terminal outputs of everything you are running? 

Thanks,
Rusty

RustyCPR, 2025-04-28:
Hi there @Saitamacode 
Any progress or updates on this request? 
If its resolved  - please let me know,
Thanks,
Rusty

RustyCPR, 2025-06-16:
Hi there @Saitamacode 
No update from your side. I will close this thread. Please reopen if you need help. 
Thanks,
Rusty

closed 2025-06-16 by RustyCPR