turtlebot/turtlebot4 #581: Issue with trying to upgrade battery

ark686, 2025-04-10:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Discovery Server

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

Not applicable

### Type of issue

Battery

### Expected behaviour

Tried to upgrade the runtime of the Turtlebot4 by getting a larger battery (roughly 3 times the Amp-Hours).  

### Actual behaviour

Installed and charged up fine, and by monitoring the time it took to discharge, was getting roughly three times the run time.

One oddity was that /battery_state did not report any larger "capacity" compared to smaller battery (assume the Create 3 base assumes battery capacity size).

But when left on dock, after getting fully charged, it would then start to discharge (while still on the dock) and eventually die.  I am thinking that the Create 3 base is thinking there is something wrong with the battery and shutting it down for "protection".  Is there a setting that can be changed in the Create 3 to get it to not discharge the fully charged larger battery when docked?

Note this occurs consistently, with two different larger batteries in multiple different Turtlebot 4s.

### Error messages

```bash

```

### To Reproduce

1.  Install new larger battery
2.  Place on dock

smatarCPR, 2025-04-16:
Hello @ark686 
Thank you for reaching out on the Turtlebot 4 github page. I have the following information regarding upgrading the battery on the Turtlebot 4: unfortunately upgrading the battery is not a supported function of the Turtlebot 4. We have not tested and will be unable to provide details regarding making this upgrade work.

If you wish to provide additional power to external payloads you plan on integrating on the Turtlebot, you may be able to mount an external power system, completely separate from the Turtlebot's main power system and use that.

Let me know if you have any questions or if I can help with anything else.
Best Regards,
Saif

closed 2025-05-12 by smatarCPR

callum-sh, 2025-09-16:
hey @smatarCPR, appreciate the response and am super curious about your recommendations on mounting an external power system and if you have any examples you can send my way? Hoping to do this to mine ASAP

RustyCPR, 2025-09-16:
Hi @callum-sh 
You could use a 12V external powered USB Hub similar to this: https://www.startech.com/en-ca/usb-hubs/5g4aibs-usb-hub-na
If you search Amazon for 12V battery banks - you'll find many options for rechargeable USB/12V batteries with varying capacity. 
Best regards,
Rusty