turtlebot/turtlebot4 #582: RPLiDAR not working

scd5027psu, 2025-04-11:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Discovery Server

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

No specific package version, it doesn't work even when echoing the topic

### Type of issue

LIDAR

### Expected behaviour

I expect the LIDAR to output usable information instead of just inf for range and 0 for intensity. I also expect the LIDAR to emit laser my other (functioning one) does.

### Actual behaviour

The LIDAR does not return usable information. I am able to see the /scan topic but the values are all inf (for range) and 0 (for intensity). It powers up and spins but no light is coming from the emitter. I know this because we have another Turtlebot 4 with a functioning LIDAR and on that unit the light can be seen when looking at it through a camera.

### Error messages

```bash

```

### To Reproduce

1) run: ros2 topic echo /scan

### Other notes

[scan_output(1).txt](https://github.com/user-attachments/files/19713665/scan_output.1.txt)

I attached outputs of the /scan topic run from the User PC with the robot in both docked and undocked states as well as directly ssh'ed into the raspberry pi.

scd5027psu, 2025-04-11:
I also double checked the USB and pin connectors, all connections are well seated.

smatarCPR, 2025-04-23:
Hello @scd5027psu 
Thank you for reaching out on the Turtlebot 4 Github page. I am very sorry to hear that you are having issues with your Turtlebot 4. Would it be possible to provide the following:

- In RVIZ if you add the scan topic are you able to see the scan rings? Would it be possible to provide a screenshot of the RVIZ window.
- Would it also be possible to provide a quick bag file. Please include all topics and use the following command: `sudo apt-get install ros-$ROS_DISTRO-rosbag2-storage-mcap; ros2 bag record -s mcap --include-hidden-topics --all`
- Lastly, would it be possible to provide the Turtlebot 4 logs using `journalctl -b > turtlebot4_log.txt`

Looking forward to hearing from you soon.
Best Regards,
Saif

BilelAllani, 2025-04-30:
Hello @scd5027psu, @smatarCPR, 

I’ve run into the same issue on my TurtleBot 4 — the LIDAR spins, but all /scan values are either 0.0 or inf. There’s no visible IR emission (confirmed by phone camera), unlike a known working LIDAR unit.

@scd5027psu Did you ever manage to find a solution or determine if it was a hardware issue?

@smatarCPR  Thanks for the detailed support steps. I’ve gathered the following as requested:

Turtlebot4_log.txt from journalctl -b: 
[turtlebot4_log.txt](https://github.com/user-attachments/files/19983524/turtlebot4_log.txt)

System properties from Create3 webserver: 
[sys_properties.txt](https://github.com/user-attachments/files/19983913/sys_properties.txt)

Recorded Bag file: 
https://filebin.net/p427e5kgbxdwh46o

RViz screenshot: no rings visible
[image]

Environment notes:
No crash or accident occurred.
Indoor, semi-controlled environment — low dust, low vibration, clean floor

Troubleshooting steps already tried:
Changed USB ports
Connected RPLIDAR directly to PC
Cleaned the laser lens
Connected via SSH to the Pi, stopped turtlebot service, and manually launched rplidar.launch.py, trying different parameters each time.

I also found someone reporting the same symptoms here: https://github.com/turtlebot/turtlebot4/issues/287 — unfortunately the issue was closed due to inactivity.

Let me know if any additional logs or diagnostics are needed.
Thanks. 
Bilel

scd5027psu, 2025-05-02:
> Hello [@scd5027psu](https://github.com/scd5027psu) Thank you for reaching out on the Turtlebot 4 Github page. I am very sorry to hear that you are having issues with your Turtlebot 4. Would it be [...]

Sure, the requested information is attached. However, I would like to restate that it is pretty clearly a hardware issue for the following reasons:
I have two turtlebot 4 mobile robot. One is working and one has the malfunctioning RPLidar. I put the RPLidar from the working Turtlebot onto the Turtlebot that wasn't showing any Lidar info and it started working as normal. I also placed the non-working Lidar onto the original working Turtlebot and the Lidar scan information was zeros and infs, indicating it was not working.
On top of this, I used my phone camera and I cannot see light coming from the emitter from the non-functioning RPLidar with it installed on either Turtlebot. The other, functioning Lidar showed light coming from the emitter when it was installed on both Turtlebot units.
The non-functioning Lidar is spinning and I double checked the pin connects, however it is not emitting. It seems pretty clear it is an issue with the RPLidar...

As promised here is the requested information:

[turtlebot4_log.txt](https://github.com/user-attachments/files/20016652/turtlebot4_log.txt)

[rosbag2_2025_04_30-10_51_06.zip](https://github.com/user-attachments/files/20016655/rosbag2_2025_04_30-10_51_06.zip)

[RvizScreenshot.zip](https://github.com/user-attachments/files/20016660/RvizScreenshot.zip)
(I don't know why but github wouldn't let me upload the screenshot as a .png so I zipped it)

smatarCPR, 2025-05-06:
Hello @scd5027psu ,
Thank you for providing the requested details from the details provided we believe that this indeed the result of a hardware failure. Would it be possible to reach out to our support team at the following email address: support@clearpathrobotics.com. Please provide the following details when reaching out:

- Please provide the serial number of the Turtlebot unit affected. This can be found on the bottom of the Create3 unit
- Please fill the attached Customer setup form

Our support team will determine if the warranty period for this Turtlebot 4 unit is still valid and take it from there.

Let me know if you have any question or if I can help with anything else.

@BilelAllani would it be possible to open a new ticket for this? This way we'll be able to better track which Turtlebots are affected by this issue.

Best Regards,
Saif

[Customer_Setup_Form_-_Support (1).pdf](https://github.com/user-attachments/files/20069182/Customer_Setup_Form_-_Support.1.pdf)

smatarCPR, 2025-05-26:
Hello @scd5027psu 
/I will go ahead and close this issue. Should you encounter any follow up issues, do not hesitate to reach out by simply replying to this thread or opening up a new issue.
Best Regards,
Saif

closed 2025-05-26 by smatarCPR