turtlebot/turtlebot4 #583: Robot doesn't boot after renaming  default username

Technion-COGROB, 2025-04-14:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Simple Discovery

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

I need a separate non-privileged user for the student in my Lab, so I've renamed the default username (ubuntu) to my lab's admin username and added a new user for the students (without adding it to the sudo group). I used the preconfigured image from the Turtlebot4 manual site.
I've  changed the network configuration from AP mode to Client mode (and successfully connecting it to my Lab's WiFi AP). After saving the changes  and rebooting, the robot won't start. the power button isn't responding - I am forced to leave the robot off the charging station for it to "die" of battery depletion.

Where in the code do I update the username so it will boot with the new user?

Thank you!

### Type of issue

Other

### Expected behaviour

Robot booting app, buttons on base (Reate 3) and user buttons by the LCD working.

### Actual behaviour

All robot buttons non-functioning, robot not booting.

### Error messages

```bash

```

### To Reproduce

Rename default username from `ubuntu` to something else

smatarCPR, 2025-04-16:
Hello @Technion-COGROB 
Thank you for reaching out on the official Turtlebot github page. I am very sorry to hear that you are having issues with your Turtlebot. I have the following questions/recommendations:

- When you say you've renamed the default username, which method did you use? Did you make the changes in the Turtlebot 4 setup utility?
- In terms of the robot not rebooting, are you able to force a restart by: placing the robot into storage mode by pressing and holding the center button on the Create3 for 7 seconds or more, then placing it on the charging dock?
- Have you made any modifications to the ROS namespace being used? I see you are using Simple Discovery which should not need a namespace, but I wanted to double check you were not using it.

Let me know the results of these checks. Looking forward to hearing from you soon.
Best Regards,
Saif

Technion-COGROB, 2025-04-17:
Hello Saif,

Thanks for getting back to me so quickly.

   - I renamed the user by using the usermod command. I wasn't aware
   there's an option for it in the setup utility (turlebot4-setup?). Can you
   tell me how to do it?
   - When I hold the middle (power) button for 7 second or more, it just
   blinks for a second and continues to stay on.
   - Haven't made any changes to the code. I followed the Turtlebot 4 Manual
   <https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html#robot>
   to the letter.
   - I planned to configure the robot to use Discovery Server, but didn't
   get that far yet.

   1. Is there a support chat I can consult to get help with my
   specific setup needs?
   2. Is there a technical guide I can send my University's IT Security
   dept. to tell them how to setup the WiFi Router for the robot (ports,
   protocols,etc.?)

Thank you!

Amichai Rotman
Cognitive Robotics Lab Unix Support
To schedule a Linux Support click here
<https://calendar.app.google/EjtNQ3Y67kZcnpVF6>
Tel:    +972 48294241 <048294241>
Email:  ***@***.***  | Web:  https://dds.technion.ac.il
<https://dds.technion.ac.il>

smatarCPR, 2025-05-06:
Hello @Technion-COGROB 
Apologies for the delayed response on this matter. I escalated the matter internally to the software development team regarding this. There advice was as follows:

In order to change the user account that runs the Turtlebot 4 services you will need to perform the following steps:

1. Perform a source installation of the Turtlebot 4 environment -  https://turtlebot.github.io/turtlebot4-user-manual/software/turtlebot4_robot.html#source-installation
2. In the code, make modifications everywhere that the username is used. This can be a very involved process, an alternative suggestion is to change the password to the main administrator account to block access for the students) and add additional user accounts.

Let me know if you have any questions or if I can help with anything else.
Best Regards,
Saif

Technion-COGROB, 2025-05-07:
Thank you for your reply.

To use the Install from Source method, I need to install an Ubuntu 22.04 Server image on the SD card, rename the user and the clone the repository and follow the instructions you've sent, correct?

smatarCPR, 2025-05-13:
Hello @Technion-COGROB ,
Apologies for the delayed response on this. You can use the Raspberry Pi images here as a basis - http://download.ros.org/downloads/turtlebot4/

There is no need to start from a blank installation. After flashing this image you can perform the steps you have listed. Just be sure to change the bash setup in the Turtlebot 4 setup utility to use the source installed files - https://turtlebot.github.io/turtlebot4-user-manual/software/turtlebot4_setup.html

Let me know if you have any questions or if I can help with anything else.
Best Regards,
Saif

smatarCPR, 2025-06-16:
Hello @Technion-COGROB 
This issue is being closed due to inactivity. If you are still experiencing the issue, feel free to reopen this ticket when you are ready to continue the troubleshooting process. 
Best Regards,
Saif

closed 2025-06-16 by smatarCPR