turtlebot/turtlebot4 #584: SLAM mapping/map update way to slow in rviz2

vhczz, 2025-04-22:
### Robot Model

Turtlebot4 Standard

### ROS distro

Jazzy

### Networking Configuration

Discovery Server

### OS

Ubuntu 24.04

### Built from source or installed?

Installed

### Package version

I dont know

### Type of issue

Navigation (SLAM, Nav2 etc.)

### Expected behaviour

When trying to visualize the process of mapping in Rviz2 the map should update smoothly as we move the robot.

### Actual behaviour

The map update is every 1-5 minutes

### Error messages

```bash

```

### To Reproduce

1. Run synchronous SLAM on a terminal
2. Run view_navigation
3. Change the topics on Rviz2  so that the match the namespace of the robots I'm working with respectively
4. Move the robot and map

### Other notes

I am using Discovery Servery as I have 2 turtlebots4, each one of them is namespaced and everytime I run rviz I have to change the topic to match the namespace and map.

smatarCPR, 2025-04-24:
Hello @vhczz ,
Thank you for reaching out on the Turtlebot 4 Github page. I am very sorry to hear that you are encountering troubles with your Turtlebot 4 unit. I have the following questions regarding your current setup:

- Is this the same Turtlebot from #580 ?
- Where are you launching the SLAM launch file? Is it when SSH'ed into the Raspberry Pi on the Turtlebot or from your remote computer?
- How is communication between the Create3, Raspberry Pi and your remote computer? Are you able to echo topics from the Create3, such as /odom on both the Raspberry Pi and your remote computer? Does sending Create3 commands work as expected - https://turtlebot.github.io/turtlebot4-user-manual/tutorials/driving.html#create-3-actions
- What type of network are you working on? Is it a 5.0 GHz or 2.4 GHz network?
- Have you made modifications to the default configuration for SLAM to generate a higher resolution map?
- Are there other robots on the same network or is this the only robot on the network?

Looking forward to hearing from you soon.
Best Regards,
Saif

vhczz, 2025-04-25:
Hello @smatarCPR 

1. Yes the Turtlebot is the same from issue #580 , we were able to create a map but we've had issues (like the one on this issue)

2. I launch asynchronous SLAM in my computer bash terminal (not ssh'ed into the RPi)

3. The communication between Create3, RPi and my computer seem to be working just fine as well. 

4. We are using a 5Ghz network

5. We haven't done any modifications to the SLAM parameters

6. We do have more robots on the same network such as another turtlebot, (they are namespaced). Also there are some other robots connected to the same network such as a UR5, UR3 and three U-Lite6 which are not always on but most of the time they are.

I'll be waiting for your response, thank you.

vhczz, 2025-04-25:
Here is a video of the turlebot driving around but not generating the map:

https://github.com/user-attachments/assets/e7696d8e-d988-4136-9386-16ae562f0773

smatarCPR, 2025-04-29:
Hello @vhczz 
Thank you for providing these details. Based on these I have the following questions/observations:

- Would it be possible to provide the full terminal outputs from launching the navigation. There may be messages there explaining why the mapping is stopping so quickly after starting
-  You mentioned that you had multiple robots on the same network. How are these robots setup? Are they setup in a discovery server configuration? Is it a fully connected setup where every robot can see the topics from the other robots (https://docs.clearpathrobotics.com/docs/ros/networking/ros2-networking#fully-connected), more akin to a command center (https://docs.clearpathrobotics.com/docs/ros/networking/ros2-networking#command-center) or are the robots completely isolated from each other? 
- Is the navigation stack being launched as a super client?

Looking forward to hearing from you soon.
Best Regards,
Saif

vhczz, 2025-05-14:
Hello, this is the terminal output from slam:

victor-chavez@victor-chavezr:~$ ros2 launch turtlebot4_navigation slam.launch.py namespace:=/Tortuga1 super_client:=True
[INFO] [launch]: All log files can be found below /home/victor-chavez/.ros/log/2025-05-13-19-27-11-156500-victor-chavezr-11649
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [sync_slam_toolbox_node-1]: process started with pid [11666]
[sync_slam_toolbox_node-1] [INFO] [1747186033.314554844] [Tortuga1.slam_toolbox]: Configuring
[sync_slam_toolbox_node-1] [INFO] [1747186033.337881454] [Tortuga1.slam_toolbox]: Using solver plugin solver_plugins::CeresSolver
[INFO] [launch.user]: [LifecycleLaunch] Slamtoolbox node is activating.
[sync_slam_toolbox_node-1] [INFO] [1747186033.385920408] [Tortuga1.slam_toolbox]: Activating
[sync_slam_toolbox_node-1] [WARN] [1747186033.678260301] [Tortuga1.slam_toolbox]: Failed to compute odom pose
[sync_slam_toolbox_node-1] [WARN] [1747186033.792354410] [Tortuga1.slam_toolbox]: Failed to compute odom pose
[sync_slam_toolbox_node-1] [WARN] [1747186033.958861023] [Tortuga1.slam_toolbox]: Failed to compute odom pose
[sync_slam_toolbox_node-1] [WARN] [1747186034.092354695] [Tortuga1.slam_toolbox]: Failed to compute odom pose
[sync_slam_toolbox_node-1] [WARN] [1747186034.209868712] [Tortuga1.slam_toolbox]: Failed to compute odom pose
[sync_slam_toolbox_node-1] [INFO] [1747186034.540533823] [Tortuga1.slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1747186034.287 for reason 'discarding message because the queue is full'
[sync_slam_toolbox_node-1] [WARN] [1747186034.588522893] [Tortuga1.slam_toolbox]: Failed to compute odom pose
[sync_slam_toolbox_node-1] [WARN] [1747186034.652129009] [Tortuga1.slam_toolbox]: Failed to compute odom pose
[sync_slam_toolbox_node-1] [WARN] [1747186034.754358404] [Tortuga1.slam_toolbox]: Failed to compute odom pose
[sync_slam_toolbox_node-1] [WARN] [1747186034.847250785] [Tortuga1.slam_toolbox]: Failed to compute odom pose
[sync_slam_toolbox_node-1] [INFO] [1747186035.043058217] [Tortuga1.slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1747186034.937 for reason 'discarding message because the queue is full'

These are the logs from rviz:

victor-chavez@victor-chavezr:~$ ros2 launch turtlebot4_viz view_navigation.launch.py namespace:=/Tortuga1 superclient:=True
[INFO] [launch]: All log files can be found below /home/victor-chavez/.ros/log/2025-05-13-19-27-32-959414-victor-chavezr-11709
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rviz2-1]: process started with pid [11712]
[rviz2-1] [INFO] [1747186053.222901982] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-1] [INFO] [1747186053.222968541] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-1] [INFO] [1747186053.235136301] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-1] [INFO] [1747186053.294440314] [Tortuga1.rviz2]: Subscribing to: /Tortuga1/mobile_base/sensors/bumper_pointcloud
[rviz2-1] 
[rviz2-1] [INFO] [1747186053.294685502] [Tortuga1.rviz2]: Subscribing to: /Tortuga1/mobile_base/sensors/bumper_pointcloud
[rviz2-1] 
[rviz2-1] [INFO] [1747186053.297566156] [Tortuga1.rviz2]: Subscribing to: /Tortuga1/local_costmap/voxel_marked_cloud
[rviz2-1] 
[rviz2-1] [INFO] [1747186053.306632957] [Tortuga1.rviz2]: Subscribing to: /Tortuga1/global_costmap/voxel_marked_cloud
[rviz2-1] 
[rviz2-1] [INFO] [1747186053.318842990] [Tortuga1.rviz2]: Subscribing to: /Tortuga1/mobile_base/sensors/bumper_pointcloud
[rviz2-1] 
[rviz2-1] [INFO] [1747186053.319453854] [Tortuga1.rviz2]: Subscribing to: /Tortuga1/local_costmap/voxel_marked_cloud
[rviz2-1] 
[rviz2-1] [INFO] [1747186053.321149666] [Tortuga1.rviz2]: Subscribing to: /Tortuga1/global_costmap/voxel_marked_cloud
[rviz2-1] 
[rviz2-1] [WARN] [1747186053.341548799] [rcl.logging_rosout]: Publisher already registered for node name: 'rviz2'. If this is due to multiple nodes with the same name then all logs for the logger named 'Tortuga1.rviz2' will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[rviz2-1] [WARN] [1747186053.346880765] [rcl.logging_rosout]: Publisher already registered for node name: 'rviz2'. If this is due to multiple nodes with the same name then all logs for the logger named 'Tortuga1.rviz2' will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[rviz2-1] [INFO] [1747186054.352775167] [Tortuga1.rviz2]: Waiting for the slam_toolbox node configuration.
[rviz2-1] [INFO] [1747186054.358144997] [Tortuga1.rviz2]: Start the slam_toolbox node state check.
[rviz2-1] [ERROR] [1747186054.826760418] [rviz2]: Lookup would require extrapolation into the future.  Requested time 1747186054.943156 but the latest data is at time 1747186054.822856, when looking up transform from frame [rplidar_link] to frame [map]
[rviz2-1] [INFO] [1747186065.372826575] [Tortuga1.rviz2]: Message Filter dropping message: frame 'rplidar_link' at time 1747186065.249 for reason 'unknown'
[rviz2-1] [INFO] [1747186083.422810591] [rviz2]: Trying to create a map of size 210 x 249 using 1 swatches
[rviz2-1] [ERROR] [1747186083.454888640] [rviz2]: rviz/glsl120/indexed_8bit_image.vert
[rviz2-1] rviz/glsl120/indexed_8bit_image.frag
[rviz2-1]  GLSL link result : 
[rviz2-1] active samplers with a different type refer to the same texture image unit

I am using a discovery server with 2 robots with diferent namespaces and I tried to run then as superclient

vhczz, 2025-06-02:
Hello I managed to create a map correctly but only after I took out the namespace off the robot. I am guessing the map wasn't being published under the correct topic somehow?... I have another problem that might be related to handling namespaces #598 which was causing this problem as well I feel like.