turtlebot/turtlebot4 #585: Basic setup - Cannot connect to network in client mode

miguelriemoliveira, 2025-04-22:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

I do not know

### OS

Ubuntu 24.04

### Built from source or installed?

Installed

### Package version

not sure how to respond to this one.

### Type of issue

Networking

### Expected behaviour

The turtlebot4 whould connect to the wifi and I should see an ip address on the lcd display.

### Actual behaviour

The turtlebot4 does not connect to the hotspot when configured in client mode. I connected through hdmi and was able to see that its not connected. Also, I do not see a connection on my phone, and the lcd display says ip unknown.

### Error messages

```bash

```

### To Reproduce

Follow the basic setup manual up until point 

**Connect the Raspberry Pi to your network**

Then, after reboot, my TB4 does not connect to the network.

RustyCPR, 2025-04-22:
Hi there @miguelriemoliveira 
Sorry to see your having concerns connecting to WiFi. 
Does your hotspot have any additional login requirements such  as user name, and enterprise/ certificate. 
It seems from your description that the RPi is not connecting to your Hotspot WiFi network, have you tried another WiFi network to determine if the Hotspot is the issue? 

Regards,
Rusty

miguelriemoliveira, 2025-04-29:
Hey @RustyCPR ,

so I tried everything listed below: 

1. phone hotspot, wpa2 , using 2.4GHZ network (does not work)
2. phone hotspot, wpa2 , using 5GHZ network (does not work)
3. old router, 2.4GZH, WEP no password (does not work)
4. new router, 2.4GHZ, WPA password (does not work)
5. new router, 5GHZ, WPA password (does not work)

I tried to get the ipaddress inside the turtlebot4 (I connect through hdmi) 

ifconfig

says ifconfig not found. Install with ... (cannot install since I have no network)

Tried to ping the router using

ping 192.168.1.1

results in network unreachable.

I am out of ideas now. Any suggestions from you?

Thanks in advance.

miguelriemoliveira, 2025-05-04:
Hey @RustyCPR ,

Could I ask you the forvor of responding to my latest answer. 

My organization purchased 5 TB4 and I need to approve the purchase. 
If I can't connect to the wireless I will have to return the robots...

Thanks in advance ...

RustyCPR, 2025-05-05:
Hi @miguelriemoliveira 
Can you send us a screen shot of the Wi-Fi Setup configuration settings before you save it. (Using your phone hotspot). 
Do you have the WiFi settings setup like this: 

[image]

Once the Wi-Fi settings are applied, the Raspberry Pi will reboot and connect to your Wi-Fi network. 
DHCP will assign it a new IP address. On the TurtleBot 4, this IP address will be shown at the top of the small LCD display.

Regards,
Rusty

miguelriemoliveira, 2025-05-06:
Hey @RustyCPR ,

thanks for replying. I am not with the robot right now so I can't send a printscreen but will do asap. 
In any case the menu is exactly like that. I put my credentials, save and them apply new settings. 

The TB4 reboots and when it boots back up again it does not connect to the wifi.

I have tested several (I think all) combinations in that menu, 5G, 2.4 GHz.

I always use dhcp because that's what my hotspot provides ...

RustyCPR, 2025-05-12:
Hi @miguelriemoliveira  

Any update on this request? 
Do you still have connectivity issues? 
Thanks,
Rusty

miguelriemoliveira, 2025-05-12:
Hey Rusty,

Yes. Nothing changed. We received 5 tb4 robots and soon I will have to
decide whether or not to ship them back.

So I am in a bit of a tricky situation.  Please share any suggestions you
have.

Thanks,
Miguel

RustyCPR, 2025-05-12:
Hi @miguelriemoliveira 
We need details to help you resolve your concern. 

1. Screen shot of the Wi-Fi Setup configuration settings before you save it

Can you send the following logs after you apply  changes and the Pi attempts to connect, preferably to a 5 GHz network from a router.
2. The full system log from the robot ` journalctl -b > turtlebot_log.txt `
3. The files in `/etc/netplan`
4. Provide the outputs of `ip a` on both the offboard computer and the Pi while connected to your wifi network. 

Thank you,
Rusty

miguelriemoliveira, 2025-05-12:
Hey @RustyCPR ,

Thanks. I have classes all day tomorrow, but Wednesday the latest I will get back to you.

miguelriemoliveira, 2025-05-13:
Hey @RustyCPR ,

Here's the info you requested:

1. Since I do not have a graphical interface (only terminal) I took a picture. I am using a connection through hdmi and usb for keyboard.

[image]

2., 3 an 4. Are all in the zip file attached.

[tb4_issue_585.zip](https://github.com/user-attachments/files/20189344/tb4_issue_585.zip)

Looking forward to hear from you.

Thanks, Miguek

closed 2025-05-24 by miguelriemoliveira