turtlebot/turtlebot4 #586: Rpi no longer boots up

ValentinLeglise, 2025-04-22:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

I do not know

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

*

### Type of issue

Battery

### Expected behaviour

When the robot is turn on, the LEDs and the robot display should light up, and a green LED should be visible on the Rpi to confirm power supply.

### Actual behaviour

Hello. My customer's robot is no longer working.

From one day to the next, the robot's Raspberry Pi won't boot, so the robot's screen and LEDs won't light up. On startup, the Create3 base unit starts up fine (white circle of light and little music), but there's no green LED on the Raspberry Pi to confirm that it's been powered up.

The cables have been connected correctly, the battery has been removed from the robot for more than 15 min to reset the battery, and the adapter has also been removed and replaced. 

What can my customer do to solve or identify the problem?

### Error messages

```bash

```

### To Reproduce

1. Put the robot on his dock to boot it up

RustyCPR, 2025-04-24:
Hi there @ValentinLeglise 
Sorry to hear your customer has these issues -here are some items to verify:

1. Remove the rear door on the TB4 and ensure the power distribution board is pushed in (it can come out) and ensure the 2 connections - USB C and Battery voltage connection is connected. 

[image]

2. Verify the connections in this reply: https://github.com/turtlebot/turtlebot4/issues/537#issuecomment-2638012695
Please also verify:
-Please check the other ends of the above connections are securely connected.
-Please make sure the RPi ribbon cable is securely connected to the top PCBA.
-When booting does the Create3 play a chime?
-What is the color ring light on the Create3 showing: https://iroboteducation.github.io/create3_docs/hw/face/

Thank you,
Rusty

ValentinLeglise, 2025-04-25:
Hello Rusty,

Thank you for your response. There is the response from my customer:

"

1.     I have removed the rear door and can confirm that the power distribution board is well connected - also, the USB-C and the battery voltage connections are well connected. 

2.  All connections are as they should be, and the RPi ribbon cable is securely connected to the top PCBA.

3. When booting, the Create 3 chimes when the TB4 boot up is complete, and the colour ring on the Create 3 is solid white.

After booting up, I can still move the TB4 with the controller, but the lights and the displays are not working. I need to set up the Lidar right now, but cant SSH into the RPi because of this issue.

"

Best regards, Valentin

ValentinLeglise, 2025-04-25:
Hello,

My customer solved his problem for the moment by using SSH into the Rpi and update the package of the robot.
Now, all seems to work correctly.

Thank you for your help.
Best regards, Valentin

RustyCPR, 2025-04-25:
Thanks for confirming @ValentinLeglise - I'll close this ticket.

closed 2025-04-25 by RustyCPR