turtlebot/turtlebot4 #587: Issue with Joy Teleoperation - launch configuration 'namespace' does not exist

otavio-paz, 2025-04-23:
### Robot Model

Turtlebot4 Standard

### ROS distro

Jazzy

### Networking Configuration

Simple Discovery

### OS

Ubuntu 24.04

### Built from source or installed?

Installed

### Package version

 ros-jazzy-turtlebot4-base 2.0.1-2noble.20250406.174303

### Type of issue

Navigation (SLAM, Nav2 etc.)

### Expected behaviour

The robot’s LIDAR data is visible in RViz, and I’m only able to control the robot using keyboard commands. After connecting to the controller, and running the joy `ros2 launch turtlebot4_bringup joy_teleop.launch.py`, I should be able to move the robot.

### Actual behaviour

While the controller connects successfully, the /joy_teleop topic is not being published and running `ros2 launch turtlebot4_bringup joy_teleop.launch.py` results in an error message.

[image]

[image]

### Error messages

```bash
ubuntu@turtlebot4:~$ ros2 launch turtlebot4_bringup joy_teleop.launch.py --debug
[DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.IncludeLaunchDescription'
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2025-04-21-10-02-41-634210-turtlebot4-3830
[INFO] [launch]: Default logging verbosity is set to DEBUG
[DEBUG] [launch]: processing event: '<launch.events.include_launch_description.IncludeLaunchDescription object at 0xffff9645c8c0>'
[DEBUG] [launch]: processing event: '<launch.events.include_launch_description.IncludeLaunchDescription object at 0xffff9645c8c0>' ✓ '<launch.event_handlers.on_include_launch_description.OnIncludeLaunchDescription object at 0xffff9645c7a0>'
Executing <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /opt/ros/jazzy/lib/python3.12/site-packages/launch/launch_service.py:228> exception=SubstitutionFailure("launch configuration 'namespace' does not exist") created at /opt/ros/jazzy/lib/python3.12/site-packages/launch/launch_service.py:319> took 0.344 seconds
[DEBUG] [launch]: An exception was raised in an async action/event
[DEBUG] [launch]: Traceback (most recent call last):
  File "/opt/ros/jazzy/lib/python3.12/site-packages/launch/launch_service.py", line 343, in run_async
    raise exception_to_raise
  File "/opt/ros/jazzy/lib/python3.12/site-packages/launch/launch_service.py", line 230, in _process_one_event
    await self.__process_event(next_event)
  File "/opt/ros/jazzy/lib/python3.12/site-packages/launch/launch_service.py", line 250, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/ros/jazzy/lib/python3.12/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/ros/jazzy/lib/python3.12/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/ros/jazzy/lib/python3.12/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  [Previous line repeated 1 more time]
  File "/opt/ros/jazzy/lib/python3.12/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
                   ^^^^^^^^^^^^^^^^^^^^^
  File "/opt/ros/jazzy/lib/python3.12/site-packages/launch/action.py", line 109, in visit
    return self.execute(context)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/opt/ros/jazzy/lib/python3.12/site-packages/launch_ros/actions/node.py", line 490, in execute
    self._perform_substitutions(context)
  File "/opt/ros/jazzy/lib/python3.12/site-packages/launch_ros/actions/node.py", line 445, in _perform_substitutions
    evaluated_parameters = evaluate_parameters(context, self.__parameters)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/ros/jazzy/lib/python3.12/site-packages/launch_ros/utilities/evaluate_parameters.py", line 164, in evaluate_parameters
    output_params.append(param.evaluate(context))
                         ^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/ros/jazzy/lib/python3.12/site-packages/launch_ros/parameter_descriptions.py", line 235, in evaluate
    param_file = perform_substitutions(context, self.__param_file)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/ros/jazzy/lib/python3.12/site-packages/launch/utilities/perform_substitutions_impl.py", line 26, in perform_substitutions
    return ''.join([context.perform_substitution(sub) for sub in subs])
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/ros/jazzy/lib/python3.12/site-packages/launch/launch_context.py", line 241, in perform_substitution
    return substitution.perform(self)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/ros/jazzy/lib/python3.12/site-packages/nav2_common/launch/rewritten_yaml.py", line 100, in perform
    root_key = launch.utilities.perform_substitutions(context, self.__root_key)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/ros/jazzy/lib/python3.12/site-packages/launch/utilities/perform_substitutions_impl.py", line 26, in perform_substitutions
    return ''.join([context.perform_substitution(sub) for sub in subs])
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/ros/jazzy/lib/python3.12/site-packages/launch/launch_context.py", line 241, in perform_substitution
    return substitution.perform(self)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/ros/jazzy/lib/python3.12/site-packages/launch/substitutions/launch_configuration.py", line 98, in perform
    raise SubstitutionFailure(
launch.substitutions.substitution_failure.SubstitutionFailure: launch configuration 'namespace' does not exist

[ERROR] [launch]: Caught exception in launch (see debug for traceback): launch configuration 'namespace' does not exist
[DEBUG] [launch.launch_context]: emitting event: 'launch.events.Shutdown'
[DEBUG] [launch]: processing event: '<launch.events.shutdown.Shutdown object at 0xffff9668f950>'
[DEBUG] [launch]: processing event: '<launch.events.shutdown.Shutdown object at 0xffff9668f950>' ✓ '<launch.event_handlers.on_shutdown.OnShutdown object at 0xffff963a4770>'
[DEBUG] [launch]: processing event: '<launch.events.shutdown.Shutdown object at 0xffff9668f950>' ✓ '<launch.event_handlers.on_shutdown.OnShutdown object at 0xffff96564e00>'
```

### To Reproduce

1. turn on robot
2. hear the happy sound
3. connect controller
4. run `ros2 launch turtlebot4_bringup joy_teleop.launch.py`

### Other notes

Additionally, when I try to manually change the namespace on the Create® 3 webserver and on the robot, since one shows "/_do_not_use" and the other is empty, but if I do that, I lose visibility of the topics. I’ve confirmed that the ROS_DOMAIN_ID is set to 0 on both the robot and the connected PC. 

The robot is undocked when I run the command, all the LEDs are green, and I hear the happy tune.

smatarCPR, 2025-04-29:
Hello @otavio-paz ,
Thank you for reaching out on the Turtlebot 4 github page. I am very sorry to hear that you are encountering problems with your Turtlebot unit. Based on the information shared I have the following questions/recommendations:

- In general you should not need to launch the joy_teleop_launch.py file, this file gets automatically launched at startup in the background. If you do not launch the teleop_joy file, and attempt to control using the controller does the Turtlebot respond?
- Are the controller inputs reaching the Turtlebot? You can test this using jstest:
- Run the command: `sudo apt-get install jstest-gtk to install jstest`
- Run the command: `jstest /dev/input/(device_name)` and use the controller to verify that the correct axis and buttons are being pressed.

Looking forward to hearing from you soon.
Best Regards,
Saif

otavio-paz, 2025-05-02:
Hi @smatarCPR ,

I appreciate for the reply!

No, the robot doesn't move without running the joy_teleop_launch.py file. The controller is connected (white-blue light is continuously on).

I think you meant for me to run: `sudo apt-get install jstest-gtk` and `jstest-gtk /dev/input/<device name>'

However, it seems that we only have _mice_ in `/dev/input'. What should be the expected (device_name) and would it show up in /dev/ anywhere else?

Thank you!

Warm regards,
Otavio.

otavio-paz, 2025-05-11:
Hello again @smatarCPR,

I was able to solve this problem. As mentioned earlier, the controller seemed to be connect, but the white-blue light that I thought it was indicating the successful connection wasn't the blue light that was intended. The fact that I couldn't see the device in `/dev/input` made this clear it was a connection issue.

Looking at this related [issue](https://github.com/turtlebot/turtlebot4/issues/544), I downloaded the p4pair.zip and ran the below as soon as I turned the controller on (pairing mode):

`sudo apt install expect`
`chmod 755 p4pair`
`./p4pair <MAC of P4 Wireless Controller>`

The remote does seem to disconnect randomly and only connect if I run the script again.

I appreciate your time.

Best regards,
Otavio.

closed 2025-05-11 by otavio-paz