turtlebot/turtlebot4 #589: cannot drive turtlebot4 lite with keyboard from user PC

bobjiang, 2025-04-28:
### Robot Model

Turtlebot4 Lite

### ROS distro

Jazzy

### Networking Configuration

Simple Discovery

### OS

Ubuntu 24.04

### Built from source or installed?

Installed

### Package version

**on turtlebot4 lite**

```
ros2 topic list
/battery_state
/cmd_audio
/cmd_lightring
/cmd_vel
/cmd_vel_unstamped
/diagnostics
/diagnostics_agg
/diagnostics_toplevel_state
/dock_status
/function_calls
/hazard_detection
/imu
/interface_buttons
/ip
/joint_states
/joy
/joy/set_feedback
/mouse
/oakd/imu/data
/oakd/rgb/preview/camera_info
/oakd/rgb/preview/image_raw
/oakd/rgb/preview/image_raw/compressed
/oakd/rgb/preview/image_raw/compressedDepth
/oakd/rgb/preview/image_raw/theora
/oakd/rgb/preview/image_raw/zstd
/odom
/parameter_events
/robot_description
/rosout
/scan
/tf
/tf_static
/wheel_status
ubuntu@turtlebot4:~$ ros2 topic echo /ip
data: 10.0.0.12
---
```

**on user PC**
Ubuntu 24.04
```
ros2 topic list
/battery_state
/cmd_audio
/cmd_lightring
/cmd_vel
/cmd_vel_unstamped
/diagnostics
/diagnostics_agg
/diagnostics_toplevel_state
/dock_status
/function_calls
/hazard_detection
/imu
/interface_buttons
/ip
/joint_states
/joy
/joy/set_feedback
/mouse
/oakd/imu/data
/oakd/rgb/preview/camera_info
/oakd/rgb/preview/image_raw
/oakd/rgb/preview/image_raw/compressed
/oakd/rgb/preview/image_raw/compressedDepth
/oakd/rgb/preview/image_raw/theora
/oakd/rgb/preview/image_raw/zstd
/odom
/parameter_events
/robot_description
/rosout
/scan
/tf
/tf_static
/wheel_status
(OM1) bob@bob-desktop:~$ ros2 topic echo /ip
data: 10.0.0.12
---
data: 10.0.0.12
---
^C(OM1) bob@bob-desktop:~ros2 run teleop_twist_keyboard teleop_twist_keyboard --ros-args -p stamped:=trueue

This node takes keypresses from the keyboard and publishes them
as Twist/TwistStamped messages. It works best with a US keyboard layout.
---------------------------
Moving around:
   u    i    o
   j    k    l
   m    ,    .

For Holonomic mode (strafing), hold down the shift key:
---------------------------
   U    I    O
   J    K    L
   M    <    >

t : up (+z)
b : down (-z)

anything else : stop

q/z : increase/decrease max speeds by 10%
w/x : increase/decrease only linear speed by 10%
e/c : increase/decrease only angular speed by 10%

CTRL-C to quit

currently:	speed 0.5	turn 1.0
```

When I tried to drive my tb4 with keyboard, 
expected: turtlebot can move
actual: nothing happened.

### Type of issue

Navigation (SLAM, Nav2 etc.)

### Expected behaviour

Expected tb4 can be driven by keyboard

### Actual behaviour

but nothing happened, when pressing keyboard with following:

Moving around:
   u    i    o
   j    k    l
   m    ,    .

For Holonomic mode (strafing), hold down the shift key:
---------------------------
   U    I    O
   J    K    L
   M    <    >

### Error messages

```bash
terminal, no error msg
```

### To Reproduce

1. run command on user pc - `ros2 run teleop_twist_keyboard teleop_twist_keyboard --ros-args -p stamped:=true`
2. no response

bobjiang, 2025-04-30:
first, I changed my env as following:

ROS -> humble
create3 firmware -> H.2.6

followed the setup page -> https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html

so on robot side, all is done 
the screenshots are following:

**create3**

[image]

**turtlebot bash setup**

setup on user pc as following:

```
ROS_DOMAIN_ID=0
RMW_IMPLEMENTATION=rmw_fastrtps_cpp
```

can get ip address by command `ros2 topic echo /ip`

**but still failed to drive turtlebot by keyboard.**

no response from any commands you advised.
the log file is attached too. 

[tb4_log.txt](https://github.com/user-attachments/files/19970340/tb4_log.txt)

smatarCPR, 2025-05-06:
Hello @bobjiang 
Thank you for providing the requested details. From the details provided I have the following observations/questions:

- From the settings pictures and the results regarding the topic echos I believe we can safely say that communication between the Create3 and Raspberry Pi is working well
- Viewing the cmd_vel command, it looks like the command sent was all 0s (i.e. a stationary command) would it be possible to resend  the command, modifying the linear x component to 0.5. I have added an example block below:

`ros2 topic pub /cmd_vel geometry_msgs/msg/Twist \
"linear:
  x: 0.5
  y: 0.0
  z: 0.0
angular:
  x: 0.0
  y: 0.0
  z: 0.0"`

- When running the keyboard teleop command, are you doing so from your remote computer or when SSH'ed into the Raspberry Pi? If you have not yet, would it be possible to try running the command after you have SSH'ed into the Raspberry Pi? Please also do so for the above cmd_vel command as well.
- Should the above succeed this would indicate that communication between your offboard computer and the Raspberry Pi is not functioning. Would it be possible to run a quick test by running the talker listener test between the offboard computer and Raspberry. Do this test twice, once publishing from the offboard computer and listening from the Pi, then vice-versa. This way we will know if communication works only in one way.

Looking forward to hearing from you soon.
Best Regards,
Saif

bobjiang, 2025-06-03:
@smatarCPR 

It seems my turtlebot4 stuck, 
when I run the command

```
ros2 topic pub /cmd_vel geometry_msgs/msg/Twist "linear:
  x: 1.0
  y: 0.0
  z: 0.0
angular:
  x: 0.5
  y: 0.0
  z: 0.0"
publisher: beginning loop
publishing #1: geometry_msgs.msg.Twist(linear=geometry_msgs.msg.Vector3(x=1.0, y=0.0, z=0.0), angular=geometry_msgs.msg.Vector3(x=0.5, y=0.0, z=0.0))

publishing #2: geometry_msgs.msg.Twist(linear=geometry_msgs.msg.Vector3(x=1.0, y=0.0, z=0.0), angular=geometry_msgs.msg.Vector3(x=0.5, y=0.0, z=0.0))
```

**no movement, and the pub is in loop,** 

try the command
```
ros2 action send_goal /drive_distance irobot_create_msgs/action/DriveDistance \
"distance: 0.5
max_translation_speed: 0.3"
Waiting for an action server to become available...
```

don't know why, and how to fix this unexpected stuck.

smatarCPR, 2025-06-10:
Hello @bobjiang 
Thank you for providing these details. Based on the details shared would it be possible to:

- Rerun test of sending the cmd_vel command, ensure that you are SSH'ed into the Raspberry Pi when publisinging the command. Afterwards, please collect the update Create 3 Logs (https://iroboteducation.github.io/create3_docs/webserver/logs/) and the updated Raspberry Pi logs `journalctl -b > turtlebot4_log.txt`
- After publishing the command when SSH'ed in the Raspberry Pi, open a new terminal, SSH into the Pi in this new terminal, and see if you are able to echo the cmd_vel command.

Let me know the results of these test. Looking forward to hearing from you soon.
Best Regards,
Saif