turtlebot/turtlebot4 #590: Turtlebot4 connects to PC with all topics visible but doesn't allow driving/mapping/etc...

TastesLikeHarry, 2025-05-06:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Discovery Server

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

N/A

### Type of issue

Networking

### Expected behaviour

Same default behaviour documented in the user manual, for teleoperation, mapping, etc...

### Actual behaviour

I can listen to the all topics on the Raspberry Pi, and all the topics seem to display, I can even give the robot commands through ssh and it will follow the undock, dock, teleop through ssh. but I can get anything to happen from the PC.

### Error messages

```bash
This is the topic list on the turtlebot4:
/battery_state
/cliff_intensity
/cmd_audio
/cmd_lightring
/cmd_vel
/diagnostics
/diagnostics_agg
/diagnostics_toplevel_state
/dock_status
/events/write_split
/function_calls
/hazard_detection
/hmi/buttons
/hmi/display
/hmi/display/message
/hmi/led
/imu
/interface_buttons
/ip
/ir_intensity
/ir_opcode
/joint_states
/joy
/joy/set_feedback
/kidnap_status
/mobility_monitor/transition_event
/mouse
/oakd/rgb/preview/camera_info
/oakd/rgb/preview/image_raw
/odom
/parameter_events
/robot_description
/robot_state/transition_event
/rosout
/scan
/slip_status
/static_transform/transition_event
/stop_status
/tf
/tf_static
/wheel_status
/wheel_ticks
/wheel_vels

ros2 topic echo /ip
data: 192.168.1.127
---

and this is it on the PC:
/battery_state
/cliff_intensity
/cmd_audio
/cmd_lightring
/cmd_vel
/diagnostics
/diagnostics_agg
/diagnostics_toplevel_state
/dock_status
/events/write_split
/function_calls
/hazard_detection
/hmi/buttons
/hmi/display
/hmi/display/message
/hmi/led
/imu
/interface_buttons
/ip
/ir_intensity
/ir_opcode
/joint_states
/joy
/joy/set_feedback
/kidnap_status
/mobility_monitor/transition_event
/mouse
/oakd/rgb/preview/camera_info
/oakd/rgb/preview/image_raw
/odom
/parameter_events
/robot_description
/robot_state/transition_event
/rosout
/scan
/slip_status
/static_transform/transition_event
/stop_status
/tf
/tf_static
/wheel_status
/wheel_ticks
/wheel_vels

ros2 topic echo /ip
data: 192.168.1.127
---
```

### To Reproduce

run teleoperation on the turtlebot4 through ssh:
Robot moves as expected.
Run teleoperation on the PC, with all topics visible:
nothing happens.

### Other notes

I've followed the process to "Install latest Raspberry Pi image" several times now thinking I've made some sort of error with the setup. It should be noted that when I received the robot initially it worked out of the box, and I was able to map my space. I attempted to set it a discovery server, to allow Create3 to be isolated from other robots as development progressed but that didn't work causing me to reimage the turtlebot4.

It's an odd one the discovery server, because when I set it up this time it didnt initally work at all until I removed the:

/_do_not_use

  motion_control:
    ros__parameters:
      # safety_override options are 
      # "none" - standard safety profile, robot cannot backup more than an inch because of lack of cliff protection in rear, max speed 0.306m/s
      # "backup_only" - allow backup without cliff safety, but keep cliff safety forward and max speed at 0.306m/s
      # "full" - no cliff safety, robot will ignore cliffs and set max speed to 0.46m/s
      safety_override: "none"

from the Application ROS 2 Parameters File and the ROS2 Namespace on the Create3 webpage.

The second I took out /do_not_use namespace everything started working as normal on the turtlebot4, but I can't command the turtlebot to do anything from the PC.

I think this has to be some sort of networking issue.

Thank you for your time,

Harry

Rahul-K-A, 2025-05-06:
Hi, I just had the same issue as you did, and I managed to solve it. I would recommend that you read https://github.com/turtlebot/turtlebot4/issues/572#issuecomment-2755463014. Essentially, there is 2 steps:

1. Put the `/do_not_use` namespace back in the Webserver settings.
2. Run `turtlebot4-setup`, then go to `ROS Setup -> Bash Setup -> ROBOT_NAMESPACE`, and set any valid namespace for all the topics
3. Reboot the turtlebot

According to the link I've attached it looks like the topics only work properly if they've been namespaced (atleast for Humble).

TastesLikeHarry, 2025-05-07:
Hi yeah, so I've tried your suggestion, the robot is now not accepting any commands at all. I can't echo back /robot1/ip topic even over ssh.
The comms light and the batt light are now off, I can see the /robot1/topics but I can interact with them in either the turtlebot4 through ssh or on the PC.

TastesLikeHarry, 2025-05-07:
edit so the robot accepted this command over ssh:

ros2 action send_goal /robot1/_do_not_use/undock irobot_create_msgs/action/Undock "{}"

Is it meant to do that? I though the point of the republisher was to allow you to use everything as normal?

Also this command did not work on the PC. So we are still at square one but with a namespace now. The lights of comm and batt have turned back on now.

Rahul-K-A, 2025-05-07:
@TastesLikeHarry Glad I could help :)

RustyCPR, 2025-05-21:
Hi there @Rahul-K-A 
Here are the requested details in TB4 online manual: https://turtlebot.github.io/turtlebot4-user-manual/software/create3.html?highlight=do_not_use#create-3-republisher 

Regards,
Rusty

guenterbauch, 2025-06-08:
> Hi, I just had the same issue as you did, and I managed to solve it. I would recommend that you read [#572 (comment)](https://github.com/turtlebot/turtlebot4/issues/572#issuecomment-2755463014). Essentially, there is 3 steps:
> 
> 1. Put the `/_do_not_use` namespace back in the Webserver settings.
> 
> [image]
> 
> 2. Run `turtlebot4-setup`, then go to `ROS Setup -> Bash Setup -> ROBOT_NAMESPACE`, and set any valid namespace for all the topics
> 3. Reboot the turtlebot and wait for the create3 and the raspberry pi to fully boot up
> 
> Assuming that the discovery server is set up correctly, you should now be able to interact with the create3 topics in the Host PC. According to the link I've attached it looks like the topics only work properly if they've been namespaced (atleast for Humble).
> 
> P.S.1: The reason for the `/_do_not_use` namespace is that the create3 republisher used in the discovery server specifically expects the create3 nodes to have that namespace when bringing up the robot. Refer: https://github.com/turtlebot/turtlebot4_robot/blob/humble/turtlebot4_bringup/launch/robot.launch.py
> 
> P.S.2: As mentioned by OP, the robot might need more than 1 reboot for the above settings to work

Hi @Rahul-K-A,

I have the same issue and setup (H2.6, TB4 and a running discovery server) like @TastesLikeHarry. 

To clarify: 

- You start with a TB4 and disovery server setup without any input 
- change the Create3 namespace to "/do_not_use" on the webserver (or only, if you emptied it?)
- change the robot namespace to "/robot1"
- reboot

Question: The namespacechange with "turtlebot4-setup"  would put the same namespace in the create3 namespace? Is that okay?

Kind regards,
gb

Rahul-K-A, 2025-06-10:
> Hi there @Rahul-K-A 
> Here are the requested details in TB4 online manual: https://turtlebot.github.io/turtlebot4-user-manual/software/create3.html?highlight=do_not_use#create-3-republisher 
> 
> Regards,
> Rusty
> 

Hi, it'd be great if we could link/reference this page directly in the discovery server page. I'd be willing to open a PR

closed 2025-08-13 by RustyCPR