turtlebot/turtlebot4 #591: Navigation Demo Stopped Working on ROS2 Jazzy

frank-gallagher, 2025-05-10:
### Robot Model

Turtlebot4 Standard

### ROS distro

Jazzy

### Networking Configuration

Simple Discovery

### OS

Ubuntu 24.04

### Built from source or installed?

Installed

### Package version

ros-jazzy-turtlebot4-navigation 
Version: 2.0.1-1noble.20250430.071125

### Type of issue

Navigation (SLAM, Nav2 etc.)

### Expected behaviour

Behaviour described in the navigation tutorial

### Actual behaviour

TB4 doesn't move to 2D goal pose

### Error messages

```bash
[controller_server-1] [ERROR] [1746882382.323839447] [controller_server]: Caught exception in callback for transition 10
[controller_server-1] [ERROR] [1746882382.323941469] [controller_server]: Original error: Considering footprint in collision checking but no robot footprint provided in the costmap.
[controller_server-1] [WARN] [1746882382.323985645] [controller_server]: Error occurred while doing error handling.
[controller_server-1] [FATAL] [1746882382.323999128] [controller_server]: Lifecycle node controller_server does not have error state implemented
[lifecycle_manager-10] [ERROR] [1746882382.327852169] [lifecycle_manager_navigation]: Failed to change state for node: controller_server
[lifecycle_manager-10] [ERROR] [1746882382.328004271] [lifecycle_manager_navigation]: Failed to bring up all requested nodes. Aborting bringup.
```

### To Reproduce

1. ros2 launch turtlebot4_navigation localization.launch.py map:=office.yaml
2. ros2 launch turtlebot4_navigation nav2.launch.py
3. ros2 launch turtlebot4_viz view_navigation.launch.py

### Other notes

This worked out of the box a few weeks ago when I got the robot. Without changing anything it stopped working.

dscholles81, 2025-05-16:
Hi,
i am new with ROS, i bought a TurtleBot4 this year.
Since i updated the turtlebot4 packages with apt i run into the same error.
So i flashed a new sd card for the pi, updated it, but the error still exists.

Thank you frank-gallagher for reporting the error, thank you pradyum for fixin it :) :) :)

civerachb-cpr, 2025-05-23:
Thanks for this! It looks like this may be the result of a recent change to Nav2. Because the Turtlebot4 is circular we were using the `robot_radius` parameter to set the footprint in the local and global costmaps. It appears that's no longer an option, so the change in https://github.com/turtlebot/turtlebot4/pull/592 to change to a footprint works.

PR #592 (2025-05-23): footprint changed to 8 sided polygon
## Description

The issue is create by CostCritic of MPPI Controller when `consider_footprint` is set to `true`. In this case the `controller_server` requires a footprint polygon and does not work with radius footprint.

Fixes #591.

## Type of change

- [x] Bug fix (non-breaking change which fixes an issue)
- [ ] New feature (non-breaking change which adds functionality)

## How Has This Been Tested?

This is tested with Turtlebot 4 navigation demo steps.

```bash
# Run this command
ros2 launch turtlebot4_navigation nav2.launch.py
```

## Checklist

- [x]  [... cut]
closed 2025-05-23 by civerachb-cpr