turtlebot/turtlebot4 #594: Multiple TB4s running discovery servers

fabioscap, 2025-05-22:
### Robot Model

Turtlebot4 Lite

### ROS distro

Jazzy

### Networking Configuration

Discovery Server

### OS

Ubuntu 24.04

### Built from source or installed?

Installed

### Package version

ros-jazzy-turtlebot4-desktop 2.0.1-1noble.20250430.025917 (not relevant)

### Type of issue

Networking

### Expected behaviour

We are trying to access topics from two Turtlebot4 Lites on an offboard PC. The two robots have been namespaced and set up to run a discovery server.

We expect that the topics appear on the offboard PC, with their namespace.

### Actual behaviour

No topics appear on the offboard PC. We note that if we use just one of the two discovery server IPs in the `ROS_DISCOVERY_SERVER` envvar then the corresponding topics pop up. I have read #145 but it's not clear to me if the capability has been added or if it refers to our problem specifically. 

### Error messages

```bash

```

### To Reproduce

TB4A (running discovery server):
namespace: "/tb4a"
IP on WLAN: 129.69.223.120
```
ROS_VERSION=2
ROS_PYTHON_VERSION=3
ROS_SUPER_CLIENT=True
ROS_DOMAIN_ID=0
ROS_AUTOMATIC_DISCOVERY_RANGE=SUBNET
ROS_DISTRO=jazzy
ROS_DISCOVERY_SERVER=127.0.0.1:11811;
```

TB4B (running discovery server) :
namespace: "/tb4b"
IP on WLAN: 129.69.223.102
```
ROS_VERSION=2
ROS_PYTHON_VERSION=3
ROS_SUPER_CLIENT=True
ROS_DOMAIN_ID=0
ROS_AUTOMATIC_DISCOVERY_RANGE=SUBNET
ROS_DISTRO=jazzy
ROS_DISCOVERY_SERVER=127.0.0.1:11811;
```

Offboard PC:
```
ROS_VERSION=2
ROS_PYTHON_VERSION=3
ROS_SUPER_CLIENT=False
ROS_DOMAIN_ID=0
ROS_AUTOMATIC_DISCOVERY_RANGE=SUBNET
ROS_DISTRO=jazzy
ROS_DISCOVERY_SERVER=129.69.223.120:11811;129.69.223.102:11811
```

### Other notes

Other than trying to run a single discovery server on the offboard PC, which would not be convenient for us, do you have advice to give us to resolve this problem?

Should the two TB4s have their IPs listed in `ROS_DISCOVERY_SERVER`?  that would conflict with [https://fast-dds.docs.eprosima.com/en/latest/fastdds/ros2/discovery_server/ros2_discovery_server.html](ros2_discovery_server), and we do not want the two robots to talk to each other.

I thank you for your assistance and wish you a nice day

RustyCPR, 2025-06-16:
Hi there @fabioscap - any feedback on my comment from 3 weeks back? Are you still experiencing issues? 

Thanks,
Rusty

RustyCPR, 2025-08-13:
Hi there @fabioscap 
I hope this email finds you well. 
I will go ahead and close this ticket. Should you need further assistance, please open a new ticket. 
Best regards,
Rusty