turtlebot/turtlebot4 #595: TurtleBot4 Screen Remains Black After Power On

LightAaron2, 2025-05-27:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Simple Discovery

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

Unable to determine package version. The TurtleBot4 arrived non-functional — the Raspberry Pi does not display anything and seems to be defective. I cannot boot into the system or run any diagnostics from the robot side.

### Type of issue

Other

### Expected behaviour

When powering on the TurtleBot4, the Raspberry Pi screen should display the expected boot sequence and bring up the ROS system. I should be able to interact with the system using the touchscreen or remotely via the network.

### Actual behaviour

The Raspberry Pi screen remains completely black after powering on. There is no signal, no boot sequence, and no indication of power on the screen. I am unable to interact with the system in any way. I suspect a hardware defect or loose connection.

### Error messages

```bash
None. The device does not boot, so I am unable to access any terminal or system logs.
```

### To Reproduce

1. Power on the TurtleBot4 using the supplied battery or power cable.
2. Observe the Raspberry Pi screen.
3. The screen remains black, and no system activity is visible.

### Other notes

The TurtleBot4 was received recently and has not been operational since arrival. There is no visual output on the screen, no boot indicator, and I cannot SSH into the unit either. It appears to be a hardware-level issue (possibly power or board failure). Please advise on whether a replacement or repair process is available.

LightAaron2, 2025-06-04:
None of the above actions worked.

RustyCPR, 2025-06-09:
Hi there @LightAaron2 
Some questions to help understand the issue: 

1. When booting does the Create3 play a chime?
2. What is the color ring light on the Create3 showing after a few minutes: https://iroboteducation.github.io/create3_docs/hw/face/
3. If you lift up the PCB-a do you see any active green LEDs? How many? Which ones? 
4. Can you [manually connect to the RPi using a ethernet cable](https://turtlebot.github.io/turtlebot4-user-manual/troubleshooting/faq.html#access-the-raspberry-pi-over-ethernet), ssh in and get an share logs: `sudo journalctl -u turtlebot4 -r`

Thank you,
Rusty

RustyCPR, 2025-08-13:
Hi there @LightAaron2 
I hope this message finds you well. I wanted to follow up as we haven’t heard back from you in some time.
I just wanted to check in to see if everything is moving forward smoothly or if there are any questions or concerns we can help address.
Best regards,
Rusty